The-OpenROAD-Project/OpenROAD #2687: Warn when derating is present when calling write_timing_model

antonblanchard, 2023-01-02:
### Describe the bug

The SDC used in Openlane contains various derating parameters:

https://github.com/The-OpenROAD-Project/OpenLane/blob/master/scripts/base.sdc

This SDC is used to create liberty timing models, which means we apply derating to both the timing model and the top level integration.

### Expected Behavior

Print a warning so the user is aware that the timing model has been derated.

### Environment

```shell
Git commit: e41b1dcc4510b1a7dda30f4c7db7de4e60050579
os: Fedora Linux 36 (Workstation Edition)
cmake version 3.25.1
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 14.0.5 (Fedora 14.0.5-2.fc36)
```

### To Reproduce

N/A

antonblanchard, 2023-01-02:
Openlane also adds external capacitance to outputs via the `set_load` command. This has an impact on the liberty model. The output tables are indexed by capacitance, so it seems adding a capacitance on top of that is bad.

Should this be a warning too?

jjcherry56, 2023-01-05:
I think it is really wrong that openlane writes liberty models by default in the flow and has no mechanism to override all of the sdc that it adds to a design. I stupid-proofed write_timing_model to ignore set_input_delay, set_output_delay, set_load, set_timing_derate in commit 384b13cd3

closed 2023-01-05 by jjcherry56