The-OpenROAD-Project/OpenROAD #2693: experiments with macro placer 2 (mpl2)

andyfox-rushc, 2023-01-03:
### Describe the bug

Running test case in src/mpl2/test causes assert to be thrown.
To reproduce:
openroad -gui
source mp_test1.tcl

height_list : 439.6 -  439.6
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:

### Expected Behavior

No crash. The tcl scripts in the test directories for each tool are incredibly useful to see what the tools do from the gui. Please if possible fix the little mp_test1.tcl to allow the new macro placer to be show cased.

### Environment

```shell
andy@DESKTOP-LFCOIHA:~/openroad_latest/OpenROAD$ ./etc/Env.sh
Git commit: bb4801b7457d04f2ddf27c202803dfbe5c8f324d
kernel: Linux 5.10.16.3-microsoft-standard-WSL2
os: Ubuntu 20.04.3 LTS (Focal Fossa)
cmake version 3.14.0
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
clang version 10.0.0-4ubuntu1
```

### To Reproduce

To reproduce:
openroad -gui
source mp_test1.tcl

### Relevant log output

```shell
OpenROAD v2.0-6284-ge41b1dcc4
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[ERROR STA-0511] cannot open 'mp1_test.tcl'.
[INFO ODB-0222] Reading LEF file: ./Nangate45/Nangate45_tech.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0226] Finished LEF file:  ./Nangate45/Nangate45_tech.lef
[INFO ODB-0222] Reading LEF file: ./Nangate45/Nangate45.lef
[WARNING ODB-0217] duplicate VIARULE (Via1Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-1) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-2) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-3) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via1Array-4) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-1) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-2) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-3) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via2Array-4) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via3Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via3Array-1) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via3Array-2) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via4Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via5Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via6Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via7Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via8Array-0) ignoring...
[WARNING ODB-0217] duplicate VIARULE (Via9Array-0) ignoring...
[INFO ODB-0225]     Created 135 library cells
[INFO ODB-0226] Finished LEF file:  ./Nangate45/Nangate45.lef
[INFO ODB-0222] Reading LEF file: ./Nangate45/fake_macros.lef
[INFO ODB-0225]     Created 2 library cells
[INFO ODB-0226] Finished LEF file:  ./Nangate45/fake_macros.lef
[WARNING STA-0160] ./Nangate45/fake_macros.lib line 32, default_max_transition is 0.0.
[INFO ODB-0128] Design: mp_test1
[INFO ODB-0252]     Updated 8 pins.
[INFO ODB-0253]     Updated 10 components.
[INFO MPL-2023] Floorplan Outline: (0.0, 0.0) (450.0, 450.0),  Core Outline: (4 4 444 443)
num level: 3, max_macro: 100, min_macro: 0, max_inst:100000, min_inst:0
[INFO MPL-0402] Traversed logical hierarchy
        Number of std cell instances : 0
        Area of std cell instances : 0.0
        Number of macros : 10
        Area of macros : 160000.0
        Total area : 160000.0
        Design Utilization : 0.82832885
        Core Utilization: 0.0

remove cluster : L0, id: 1
remove cluster : L2, id: 3
remove cluster : T0, id: 4
remove cluster : T1, id: 5
remove cluster : T2, id: 6
remove cluster : R0, id: 7
remove cluster : R2, id: 9
remove cluster : B0, id: 10
remove cluster : B1, id: 11
remove cluster : B2, id: 12
Finish Creating Vertices. Num of Vertices: 40
Finish hypergraph creation
Finish Calculate Data Flow detailed CR done
[Debug][HierRTLMP::MultiLevelCluster]  root 1:  num_macro 10 num_stdcell 0
[Debug][HierRTLMP::MultiLevelCluster]  root 2:  num_macro 10 num_stdcell 0
[Debug][HierRTLMP::MultiLevelCluster]  root 3:  num_macro 10 num_stdcell 0
[Debug][HierRTLMP::BreakCluster] cluster_name : root
[INFO MPL-2022]
********************************************************************************
[INFO] cluster_name :  L1  cluster_id : 2
num_std_cell : 0  num_macro : 0
width : 0.000000  height : 150.000000  location :  ( 0.000000 , 150.000000 )

[INFO MPL-2022]
********************************************************************************
[INFO] cluster_name :  R1  cluster_id : 8
num_std_cell : 0  num_macro : 0
width : 0.000000  height : 150.000000  location :  ( 450.000000 , 150.000000 )

[INFO MPL-2025] root  (0)    num_macro :  10  num_std_cell :  0  macro_area :  160000.000000  std_cell_area : 0.000000
tot_num_macro_in_children = 0
[INFO MPL-2025] +----L1  (2)    num_macro :  0  num_std_cell :  0  macro_area :  0.000000  std_cell_area : 0.000000
tot_num_macro_in_children = 0
[INFO MPL-2025] +----R1  (8)    num_macro :  0  num_std_cell :  0  macro_area :  0.000000  std_cell_area : 0.000000
tot_num_macro_in_children = 0
[INFO MPL-2025] root  (0)    num_macro :  10  num_std_cell :  0  macro_area :  160000.000000  std_cell_area : 0.000000
tot_num_macro_in_children = 0
[INFO MPL-2025] +----L1  (2)    num_macro :  0  num_std_cell :  0  macro_area :  0.000000  std_cell_area : 0.000000
tot_num_macro_in_children = 0
[INFO MPL-2025] +----R1  (8)    num_macro :  0  num_std_cell :  0  macro_area :  0.000000  std_cell_area : 0.000000
tot_num_macro_in_children = 0
root
width_list : 440.04  -  440.04
height_list : 439.6 -  439.6
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
/usr/include/c++/9/bits/stl_vector.h:1042: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = float; _Alloc = std::allocator<float>; std::vector<_Tp, _Alloc>::reference = float&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
```

### Additional Context

Running a test tcl from within the gui for the mpl2 to see how the new macro placer fares.

tspyrou, 2023-01-04:
@abk-openroad fyi

ravi-varadarajan, 2023-01-04:
Will push the fix for this

vijayank88, 2023-06-22:
@andyfox-rushc 
With latest master, able to run successfully. 
```
boundary_h_th : 100.0, boundary_v_th : 110.0
notch_h_th : 125.0, notch_v_th : 137.5
number of updated macros : 10
number of macros in HardMacroCluster : 10
Differences found at line 342.
    - U1 HM_100x400_4x4 + FIXED ( 19880 17640 ) N ;
    - U1 HM_100x400_4x4 + FIXED ( 19880 18480 ) N ;
    
```
Please update your build and try again.

closed 2023-06-22 by vijayank88