The-OpenROAD-Project/OpenROAD #2709: OpenROAD metrics should consider absolute value of clock skew

antonblanchard, 2023-01-06:
### Describe the bug

I have a PR where the clock skew got better, but is now negative:

```
- ,"DRT::clock_skew" : "0.036251402905609166"
+ ,"DRT::clock_skew" : "-0.028153347117122647"
```

This is failing even though the metric limit is larger (positive):

```
"DRT::clock_skew" : "0.036251402905609166"
````

Furthermore, even after running save_flow_metrics_limits it still fails.

### Expected Behavior

A comparison of the absolute values of the clock_skew is done, the test passes, and save_flow_metrics_limits does the right thing.

### Environment

```shell
os: Fedora Linux 36 (Workstation Edition)
cmake version 3.25.1
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 14.0.5 (Fedora 14.0.5-2.fc36)
```

### To Reproduce

The PR exhibiting this issue is https://github.com/The-OpenROAD-Project/OpenROAD/pull/2616. I think it is cursed.

rovinski, 2023-01-06:
I am mildly confused about how clock skew can be negative at all? It should be the max difference in arrival times of clocks to endpoints, which I would expect to be an absolute value.

antonblanchard, 2023-01-07:
> I am mildly confused about how clock skew can be negative at all? It should be the max difference in arrival times of clocks to endpoints, which I would expect to be an absolute value.

It confused me too. I can understand how a single endpoint might have negative clock skew but wasn't sure if the overall clock skew metric can be negative or not. @jjcherry56 might be able to help.

jjcherry56, 2023-01-07:
report_clock_skew reports the maximum difference in clock arrival between every source and target register that have a path between them. It is ACTUAL skew for a path in the design. Using max difference in arrival of the clock at endpoints and startpoints (NOT just endpoints) is more speculative because there may not be a path between those registers. If you look at the result of report_clock_skew rather than the abstracted number in the metrics it is more obvious:

Clock clk1
Latency      CRPR       Skew
r1/CP ^
   1.71
r3/CP ^
   3.52     -0.71      -2.52

maliberty, 2023-01-10:
@antonblanchard is anything more needed here?

maliberty, 2023-01-10:
I think the simplest thing is to report the abs of the current value.  The sign isn't very interesting at the level of metrics.  This will require updating the current metrics files (@vvbandeira )

vvbandeira, 2023-01-11:
> I think the simplest thing is to report the abs of the current value. The sign isn't very interesting at the level of metrics. This will require updating the current metrics files (@vvbandeira )

We track clock skew, but it is not part of the CI rules. Thus, there's no need to update current metrics files.

I believe the only change required is to add a cast to an absolute value in [this line](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/Metrics.tcl#L51). If this is the path we decide on, I can create the PR; let me know.

maliberty, 2023-01-11:
@antonblanchard said he will take care of this update

antonblanchard, 2023-01-23:
This was fixed in https://github.com/The-OpenROAD-Project/OpenROAD/commit/399b2c9dc597627f80fbcd383158d8492001a594

closed 2023-01-23 by antonblanchard