The-OpenROAD-Project/OpenROAD #2711: [ERROR MPL-2024] Unable to complete RTL-MPL2

vijayank88, 2023-01-06:
### Describe the bug

I'm trying explore rtl-mpl2 features with OpenROAD. 
If I try to use regular `macro_placement` it works fine. 
If I use `rtl_macro_placer` it fails.

### Expected Behavior

Complete `rtl_macro_placer` with sram macro placement

### Environment

```shell
Git commit: bc67783cb6d819f39828a6a511ae771c9e2f083f
kernel: Linux 3.10.0-1160.49.1.el7.x86_64
os: CentOS Linux 7 (Core)
cmake version 3.24.2
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
./etc/Env.sh: line 45: clang: command not found
```

### To Reproduce

[RTLMP_Testcase.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10360093/RTLMP_Testcase.zip)
1. Unzip the testcase
2. cd RTLMP_Testcase
3. openroad > source user_wrapper1.tcl

### Relevant log output

```shell
[INFO MPL-2023] Floorplan Outline: (0.0, 0.0) (2880.0, 3480.0),  Core Outline: (5 10 2874 3468)
num level: 2, max_macro: 160, min_macro: 40, max_inst:20000000, min_inst:50000
[INFO MPL-0402] Traversed logical hierarchy
	Number of std cell instances : 107685
	Area of std cell instances : 1165220.1
	Number of macros : 11
	Area of macros : 3036097.25
	Total area : 4201317.5
	Design Utilization : 0.42347714
	Core Utilization: 0.16924272

[ERROR MPL-2024] Floorplan has not been initialized? Pin location error.
```

ravi-varadarajan, 2023-01-07:
This error typically means that there is some issue with pin placement. Will look at the test case and get back to you

maliberty, 2023-01-08:
The pins stick out of the block bounds:
[image]

mpl2 seems to assume they will align to the edges.  I tried fixing it but then hit a different crash.

maliberty, 2023-01-09:
@ravi-varadarajan I fixed the issue with the pin but there is a subsequent crash.  It appears to be related to having all macros in the top most verilog module

ravi-varadarajan, 2023-01-19:
@vijayank88 @maliberty   I have isolated the issue in this test case. The fundamental cause is that the synthesis was done flat all std cells and macros are at the top level of the hierarchy.  For ideal use of RTLMP there should be hierarchy in the synthesized netlist as it uses it optimize the clustering and grouping of hard macros. However this corner case should be handled without failure too.

You can either use Yosys to generate the hierarchical netlist (look at nangate45 black_parrot) in ORFS or even use a cached gate level netlist. 

The fix for the corner case will be in a PR that I will merge very shortly
thanks

ravi-varadarajan, 2023-01-20:
Fixed this issue in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2797

closed 2023-01-23 by maliberty