The-OpenROAD-Project/OpenROAD #2715: `utl::metric_float` outputs invalid JSON in case of `inf` values

donn, 2023-01-07:
### Describe the bug

Generating metrics with infinite values generates invalid JSON files.

[image]

JSON generally does not have full support for IEEE754 and infinite values are not allowed. I'm… not sure what the alternative here would be exactly but compliant JSON parsers will reject the output file.

### Expected Behavior

A parseable JSON file.

Can the values be intercepted and dumped as strings, "+Infinity" or "-Infinity", perhaps?

### Environment

```shell
Git commit: bc67783cb6d819f39828a6a511ae771c9e2f083f
kernel: Linux 5.15.0-1025-gcp
os: Ubuntu 20.04.4 LTS (Focal Fossa)
cmake version 3.22.2
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

### To Reproduce

[repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10366981/repro.tar.gz)

QuantamHD, 2023-01-07:
@gadfort I think you made this change

gadfort, 2023-01-07:
I've filed a PR to fix it since we were having the same issue. Just pending approval

gadfort, 2023-01-07:
#2712 is the PR for reference

gadfort, 2023-01-09:
Actually looking at this, I think this might be a divide by 0 issue.

maliberty, 2023-01-09:
It is which I'll fix.

maliberty, 2023-01-09:
Do you still want to proceed?  https://github.com/The-OpenROAD-Project/OpenROAD/pull/2721

gadfort, 2023-01-09:
I'm okay with #2721 if the metrics logger needs to be augmented we can revisit

PR #2721 (2023-01-09): Handle zero core area in utilization (report as -1 for no value)
Fixes #2715
commit be59c88d (2023-01-09): Handle zero core area in utilization (report as -1 for no value)

Fixes #2715
closed 2023-01-09 by maliberty