The-OpenROAD-Project/OpenROAD #2716: Segfault in report_checks with floorplan def

gadfort, 2023-01-07:
### Describe the bug

When I added `check_setup` to my flow before reading the floorplan DEF, `report_checks` causes a segfault, while if I don't include the `check_setup`, it does not segfault

### Expected Behavior

No segfault

### Environment

```shell
Git commit: bc67783cb6d819f39828a6a511ae771c9e2f083f
kernel: Linux 5.15.79.1-microsoft-standard-WSL2
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.25.0
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Ubuntu clang version 14.0.0-1ubuntu1
```

### To Reproduce

[floorplan_gcd_sky130hd_base_2023-01-07_16-21.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10367228/floorplan_gcd_sky130hd_base_2023-01-07_16-21.tar.gz)
./run-me-gcd-sky130hd-base.sh

### Relevant log output

```shell
==========================================================================
floorplan final check_setup
--------------------------------------------------------------------------
Signal 11 received
Stack trace:
 0# 0x0000558EC44933BB in /usr/local/bin/openroad
 1# 0x00007F2F9CC6E520 in /lib/x86_64-linux-gnu/libc.so.6
 2# std::_Rb_tree<sta::Vertex*, sta::Vertex*, std::_Identity<sta::Vertex*>, sta::VertexIdLess, std::allocator<sta::Vertex*> >::_M_get_insert_unique_pos(sta::Vertex* const&) in /usr/local/bin/openroad
 3# std::pair<std::_Rb_tree_iterator<sta::Vertex*>, bool> std::_Rb_tree<sta::Vertex*, sta::Vertex*, std::_Identity<sta::Vertex*>, sta::VertexIdLess, std::allocator<sta::Vertex*> >::_M_insert_unique<sta::Vertex*>(sta::Vertex*&&) in /usr/local/bin/openroad
 4# sta::Search::findInputDrvrVertices(sta::VertexSet&) in /usr/local/bin/openroad
 5# sta::Search::seedArrivals() in /usr/local/bin/openroad
 6# sta::Search::findArrivals1() in /usr/local/bin/openroad
 7# sta::Search::findArrivals(int, sta::VertexVisitor*) in /usr/local/bin/openroad
 8# sta::Search::findAllArrivals(sta::VertexVisitor*) in /usr/local/bin/openroad
 9# sta::Sta::checkTiming(bool, bool, bool, bool, bool, bool, bool) in /usr/local/bin/openroad
10# 0x0000558EC469D537 in /usr/local/bin/openroad
11# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
12# 0x00007F2FA0A85028 in /lib/x86_64-linux-gnu/libtcl8.6.so
13# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
14# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
15# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /usr/local/bin/openroad
16# ord::tclAppInit(Tcl_Interp*) in /usr/local/bin/openroad
17# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
18# main in /usr/local/bin/openroad
19# 0x00007F2F9CC55D90 in /lib/x86_64-linux-gnu/libc.so.6
20# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
21# _start in /usr/local/bin/openroad
Command terminated by signal 11
Elapsed time: 0:00.29[h:]min:sec. CPU time: user 0.26 sys 0.02 (99%). Peak memory: 106748KB.
make: *** [Makefile:420: results/sky130hd/gcd/base/2_1_floorplan.odb] Error 139
```

jjcherry56, 2023-01-09:
can't reproduce on macos or linux with the same commit or the ORD head

maliberty, 2023-01-09:
I get it as well.

maliberty, 2023-01-09:
It happens when compiled debug.  When run under gdb the problem goes away which suggests uninit memory.  Valgrind shows
```
--------------------------------------------------------------------------
==255197== Invalid read of size 2
==255197==    at 0x3D26F4C: sta::Vertex::objectIdx() const (Graph.hh:335)
==255197==    by 0x3D277CF: sta::ObjectTable<sta::Vertex>::objectId(sta::Vertex const*) (ObjectTable.hh:157)
==255197==    by 0x3D23515: sta::Graph::id(sta::Vertex const*) const (Graph.cc:363)
==255197==    by 0x3D26E79: sta::VertexIdLess::operator()(sta::Vertex const*, sta::Vertex const*) const (Graph.cc:1611)
==255197==    by 0x3BB3A40: std::_Rb_tree<sta::Vertex*, sta::Vertex*, std::_Identity<sta::Vertex*>, sta::VertexIdLess, std::allocator<sta::Vertex*> >::_M_get_insert_unique_pos(sta::Vertex* const&) (stl_tree.h:2054)
==255197==    by 0x3E90AA4: std::pair<std::_Rb_tree_iterator<sta::Vertex*>, bool> std::_Rb_tree<sta::Vertex*, sta::Vertex*, std::_Identity<sta::Vertex*>, sta::VertexIdLess, std::allocator<sta::Vertex*> >::_M_insert_unique<sta::Vertex*>(sta::Vertex*&&) (stl_tree.h:2106)
==255197==    by 0x3E8F071: std::set<sta::Vertex*, sta::VertexIdLess, std::allocator<sta::Vertex*> >::insert(sta::Vertex*&&) (stl_set.h:520)
==255197==    by 0x3E85211: sta::Search::findInputDrvrVertices(sta::VertexSet&) (Search.cc:1620)
==255197==    by 0x3E83E56: sta::Search::seedArrivals() (Search.cc:1366)
==255197==    by 0x3E82406: sta::Search::findArrivals1() (Search.cc:1002)
==255197==    by 0x3E822AC: sta::Search::findArrivals(int, sta::VertexVisitor*) (Search.cc:982)
==255197==    by 0x3E820D4: sta::Search::findAllArrivals(sta::VertexVisitor*) (Search.cc:940)
==255197==  Address 0x24 is not stack'd, malloc'd or (recently) free'd
==255197== 
Signal 11 received
```

maliberty, 2023-01-09:
@jjcherry56 is that sufficient for you to take over?

closed 2025-08-26 by maliberty