The-OpenROAD-Project/OpenROAD #2724: Enable set_io_pin_constraint to constrain ordering of pins

oharboe, 2023-01-10:


oharboe, 2023-01-10:
@eder-matheus I need to investigate more, but I'm getting an assert:

```
Error: io_placement.tcl, 13 vector::_M_range_check: __n (which is 1073743116) >= this->size() (which is 4936)
```

eder-matheus, 2023-01-10:
> @eder-matheus I need to investigate more, but I'm getting an assert:
> 
> ```
> Error: io_placement.tcl, 13 vector::_M_range_check: __n (which is 1073743116) >= this->size() (which is 4936)
> ```

Could you provide a testcase?

oharboe, 2023-01-10:
> > @eder-matheus I need to investigate more, but I'm getting an assert:
> > ```
> > Error: io_placement.tcl, 13 vector::_M_range_check: __n (which is 1073743116) >= this->size() (which is 4936)
> > ```
> 
> Could you provide a testcase?

I quickly tried this simple test-case and it doesn't fail and the pins are placed as expected.

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/764

maliberty, 2023-01-10:
Coverity reported this morning:
```
** CID 1502481:  Error handling issues  (NEGATIVE_RETURNS)
/src/ppl/src/HungarianMatching.cpp: 152 in ppl::HungarianMatching::getFinalAssignment(std::vector<ppl::IOPin, std::allocator<ppl::IOPin>> &, std::unordered_map<odb::dbBTerm *, odb::dbBTerm *, std::hash<odb::dbBTerm *>, std::equal_to<odb::dbBTerm *>, std::allocator<std::pair<odb::dbBTerm *const , odb::dbBTerm *>>> &, bool) const()

________________________________________________________________________________________________________
*** CID 1502481:  Error handling issues  (NEGATIVE_RETURNS)
/src/ppl/src/HungarianMatching.cpp: 152 in ppl::HungarianMatching::getFinalAssignment(std::vector<ppl::IOPin, std::allocator<ppl::IOPin>> &, std::unordered_map<odb::dbBTerm *, odb::dbBTerm *, std::hash<odb::dbBTerm *>, std::equal_to<odb::dbBTerm *>, std::allocator<std::pair<odb::dbBTerm *const , odb::dbBTerm *>>> &, bool) const()
146               odb::Point mirrored_pos = core_->getMirroredPosition(io_pin.getPos());
147               mirrored_pin.setPos(mirrored_pos);
148               mirrored_pin.setLayer(slots_[slot_index].layer);
149               mirrored_pin.setPlaced();
150               assigment.push_back(mirrored_pin);
151               slot_index = getSlotIdxByPosition(mirrored_pos);
>>>     CID 1502481:  Error handling issues  (NEGATIVE_RETURNS)
>>>     "slot_index" is passed to a parameter that cannot be negative. [Note: The source code implementation of the function has been overridden by a builtin model.]
152               slots_[slot_index].used = true;
153             }
154             break;
155           }
156           col++;
157         }
```

perhaps it is related

oharboe, 2023-01-10:
> Coverity reported this morning:
>
> ```
> ** CID 1502481:  Error handling issues  (NEGATIVE_RETURNS)
> /src/ppl/src/HungarianMatching.cpp: 152 in [...]

If this can be fixed first, I will retry. I failed to put together a testcase based upon my example after a few fleeting vain attempts....

eder-matheus, 2023-01-10:
@oharboe @maliberty PR with a possible fix here: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2726

oharboe, 2023-01-10:
> @oharboe @maliberty PR with a possible fix here: #2726

Still crashes:

```
[WARNING PPL-0037] message limit reached, this message will no longer print
Error: io_placement.tcl, 13 vector::_M_range_check: __n (which is 1073743116) >= this->size() (which is 4936)
```

oharboe, 2023-01-10:
@eder-matheus This looks suspicious. Since we have assigned more pins than the total, the loop below never terminates...

```
[INFO PPL-0009] Unsuccessfully assigned pins to sections (617 out of 611).
```

[image]

oharboe, 2023-01-10:
Shooting from the hip, worked on my design: https://github.com/The-OpenROAD-Project/OpenROAD/pull/2727

eder-matheus, 2023-01-10:
> Shooting from the hip, worked on my design: #2727

Thanks for the fix, @oharboe! I've approved your PR. I didn't notice the bug before, thanks for looking at it.

oharboe, 2023-01-10:
> > Shooting from the hip, worked on my design: #2727
> 
> Thanks for the fix, @oharboe! I've approved your PR. I didn't notice the bug before, thanks for looking at it.

@eder-matheus Can a testcase be deviced?

Seems like testcases should catch this?

oharboe, 2023-01-10:
> > > > Shooting from the hip, worked on my design: #2727
> > > 
> > > 
> > > Thanks for the fix, @oharboe! I've approved your PR. I didn't notice the bug before, thanks for looking at it.
> > 
> > 
> > @eder-matheus Can a testcase be deviced?
> > Seems like testcases should catch this?
> 
> Yes, I'll make a testcase. Can I push it to your PR?

You can just create a new PR based on mine and close my PR, I don't think you can push to my git repository.

eder-matheus, 2023-01-10:
> You can just create a new PR based on mine and close my PR, I don't think you can push to my git repository.

Ok. Let's merge your PR to keep in the history the update came from you and I can create a PR for the unit test.

oharboe, 2023-01-10:
> Ok. Let's merge your PR to keep in the history the update came from you and I can create a PR for the unit test.

Gotit. I saw it is in master now. Note that I'm still getting the same infinite loop error, I guess there are more ways to configure pins that trigger this sort of bug.

I suspect you'll find a few with your test-cases...

eder-matheus, 2023-01-10:
> Gotit. I saw it is in master now. Note that I'm still getting the same infinite loop error, I guess there are more ways to configure pins that trigger this sort of bug.
>
> I suspect you'll find a few [...]

Could you share the configurations that are generating the infinite loops?

oharboe, 2023-01-10:
> Could you share the configurations that are generating the infinite loops?

Unfortunately, I cant share this design.

However, tomorrow I was thinking about removing the accounting code of placed pins that is interleaved with the code and simply check if all pins are placed where the progress message was. This is again shooting for the hip, but the hope is that this will be more robust as less code has to be correct.

maliberty, 2023-01-10:
Can you use the delta debugger to make it small enough to share?

oharboe, 2023-01-10:
> Can you use the delta debugger to make it small enough to share?

I think that might be possible, but probably overkill. If I can see the pattern of a unittest or two, I fancy I can create one the fails for the same reasons it fails here.

oharboe, 2023-01-10:
While not a failing example, I have found out that part of the problem is that there are various ways to provide information about the pins.

Here I have provided all the information, but in three calls:

```
set_io_pin_constraint -region up:* -pin_names $uppins
set_io_pin_constraint -region bottom:* -pin_names $bottompins
set_io_pin_constraint -mirrored_pins $mirrored
```

A single invocation ```set_io_pin_constraint``` can also provide unambigious information about the pin-layout:

```
set_io_pin_constraint -region up:* -pin_names $uppins -mirrored_pins $mirrored
```

When I fiddle with the alternatives, I sometimes get an infinite loop, others not.

oharboe, 2023-01-11:
To reproduce the problem, use https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/764 and run the below.

```
make DESIGN_CONFIG=designs/asap7/mock-array-big/Element/config.mk
[deleted]
[INFO PPL-0010] Tentative 0 to set up sections.
[INFO PPL-0001] Number of slots          408
[INFO PPL-0002] Number of I/O            65
[INFO PPL-0003] Number of I/O w/sink     65
[INFO PPL-0004] Number of I/O w/o sink   0
[INFO PPL-0005] Slots per section        200
[INFO PPL-0006] Slots increase factor    0.01
[INFO PPL-0009] Unsuccessfully assigned pins to sections (97 out of 65).
[deleted]
[WARNING PPL-0037] Number of slots per sections is 11485420 while the maximum recommended value is 600 this may negatively affect performance.
[WARNING PPL-0037] message limit reached, this message will no longer print
Error: io_placement.tcl, 13 vector::_M_range_check: __n (which is 1073741925) >= this->size() (which is 408)
Elapsed time: 0:01.29[h:]min:sec. CPU time: user 1.24 sys 0.04 (99%). Peak memory: 177752KB.
make: *** [results/asap7/mock-array-big_Element/base/3_2_place_iop.odb] Error 1

```

oharboe, 2023-01-13:
@eder-matheus @maliberty Is the failure case above adequate to reproduce or do I need to provide anything else?

oharboe, 2023-01-13:
@eder-matheus Good error and warning messages when I am doing things wrong is enormously helpful. Almost.more than documentation, I am sure to read them 😀

eder-matheus, 2023-01-13:
@oharboe I've merged my updates into master. You should not have the infinite loop error anymore, and the warn messages I mentioned early are also added.

Also, now you should only be able to call `-mirrored_pins` without any other options of `set_io_pin_constraints`. So now this is not valid:
`set_io_pin_constraint -region up:* -pin_names $uppins -mirrored_pins $mirrored`

I've made this to avoid confusion with different command calls generating the same results.

oharboe, 2023-01-13:
@eder-matheus Makes sense. Seems to work, I will continue to do more testing and let you know if something shows up. Thanks!

tspyrou, 2023-02-07:
This is resolved after discussion.

closed 2023-02-07 by tspyrou