The-OpenROAD-Project/OpenROAD #2729: `getFirstPinLocation` API Break

donn, 2023-01-11:
### Describe the bug

In earlier versions of OpenROAD, the Python API for `BTerm::getFirstPinLocation` would simply return a tuple of three objects: a boolean representing success or failure and two integers. In the most recent version, however, the Python API seems to have a bug where it requests two values by reference. Requesting integers by reference is not possible in Python.

### Expected Behavior

The API should remain consistent with earlier versions of OR.

### Environment

```shell
Git commit: 4ca23914299f8281ec1b2bb14d5b2a18440fbc6f
kernel: Linux 5.15.0-1025-gcp
os: Ubuntu 20.04.4 LTS (Focal Fossa)
cmake version 3.22.2
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

### To Reproduce

Run `run.sh` inside the folder. [repro.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10387673/repro.tar.gz)

### Relevant log output

```shell
OpenROAD v2.0-6404-g4ca239142 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
Design name: spm
HERE <odb.dbBTerm; proxy of <Swig Object of type 'odb::dbBTerm *' at 0x7f7343378ba0> >
Traceback (most recent call last):
  File "./diodes.py", line 416, in <module>
    cli()
  File "/home/donn/.local/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/donn/.local/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/donn/.local/lib/python3.8/site-packages/click/core.py", line 1657, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/home/donn/.local/lib/python3.8/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/donn/.local/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/donn/OpenLane/repro/reader.py", line 76, in wrapper
    function(**kwargs)
  File "./diodes.py", line 374, in place
    di.execute()
  File "./diodes.py", line 284, in execute
    src_pos = self.net_source(net)
  File "./diodes.py", line 101, in net_source
    good, x, y = bt.getFirstPinLocation()
TypeError: getFirstPinLocation() missing 2 required positional arguments: 'x' and 'y'
```

maliberty, 2023-01-11:
@macd I think this relates to the odb change in .i that I hightlighted in your pdn PR

macd, 2023-01-11:
Yes, it sure does. I just fixed it and will push soon.

maliberty, 2023-01-11:
Fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/2730

closed 2023-01-11 by maliberty