The-OpenROAD-Project/OpenROAD #2738: CMake build flow support for ASAN

cdleary, 2023-01-11:
### Description

Address Sanitizer is a great way to discover memory leaks and memory corruption, and works even in optimized builds with small performance penalties. I'm not sure if there's a way today to configure the CMake based build to run with ASAN -- I don't see it mentioned in a search of the docs -- but it could be quite valuable, I think I see some instances where `Ath__array1D` objects are `new`'d and seem to fall out of scope, e.g. https://github.com/The-OpenROAD-Project/OpenROAD/blob/e41b1dcc4510b1a7dda30f4c7db7de4e60050579/src/rcx/src/extBenchDB.cpp#L407

maliberty, 2023-01-11:
There are tons of Coverity warnings in rcx that might catch many of these.   We do run a valgrind pipeline on gf12/gcd daily.  It shows a bunch of rcx leaks but the total loss is small enough that it hasn't been a priority.

That said more tools are better.

Iyury1, 2023-05-16:
I would like to work on this issue.

maliberty, 2023-05-16:
Great!

Iyury1, 2023-05-17:
To compile with address sanitizer I added the following lines to /src/CMakeLists.txt inside a conditional expression:

```
  add_compile_options(
    -fsanitize=address
    -g
  )
  add_link_options(
    -fsanitize=address
  )
  ```
  
  Now I have a couple questions. [Here](https://github.com/The-OpenROAD-Project/OpenROAD/issues/3243#issuecomment-1522538121) @antonblanchard used the flags "-fno-omit-frame-pointer" and "-O2". Are these flags necessary?
  
  Also, how do we want to tell CMake to build with ASAN from the command line?

  I noticed that the variable CMAKE_BUILD_TYPE is not being used in any CMakeLists.txt file apart from the top level file in OPENROAD_HOME. There the variable is set to RELEASE if not already set. In /docs/user/Build.md the value DEBUG is listed as a valid CMAKE_BUILD_TYPE. Since we are using the flag "-g" for address sanitizer, I think it makes sense to require CMAKE_BUILD_TYPE to be DEBUG. Should there be another option to enable address sanitizer? Perhaps -DASAN or -DSANITIZE.

Iyury1, 2023-05-17:
By the way, when compiling with GNU the ASAN library was not located without adding the link option. I will make sure it works for both GNU and Clang.

I believe the default compiler optimization for CMake in release mode is -O3. I observed the -O3 flag being used when the build type RELEASE, although I do not see any compiler optimization flag when the build type is DEBUG. The optimization level could be increased during debug so that it is closer to the release mode, but I am not sure if this should be a default setting for DEBUG.

For activating Address Sanitizer I could think of a few options.
1. Always enable ASAN with -DCMAKE_BUILD_TYPE=DEBUG
2. Enable ASAN with -DCMAKE_BUILD_TYPE=DEBUG and -DASAN=ON
3. With -DASAN=ON only, enable ASAN and set CMAKE_BUILD_TYPE to DEBUG

maliberty, 2023-05-17:
I believe DEBUG uses "-g -O0" by default.  Using any more than that risks being unable to debug or missing ASAN violations.

I prefer: 
4.  With -DASAN=ON, enable ASAN

Whether you want debug or not is orthogonal and up to the developer.  For very large runs you might not be able to use DEBUG due to runtime considerations.

rovinski, 2023-05-17:
`-DCMAKE_BUILD_TYPE=RelWithDebugInfo` enables more optimizations.

Iyury1, 2023-05-20:
I have made the change to enable ASAN with -DASAN=ON. However, I am concerned that not including the "-g" flag could cause confusion. Without debug symbols Address Sanitizer will not show file names and line numbers in the call stack. I suggest that /docs/user/Build.md be modified to list the new ASAN argument along with a note suggesting to compile with "-g" by changing CMAKE_BUILD_TYPE to DEBUG in order to display line numbers.

Before making a pull request, I would like to test running openroad with a memory leak present, since Address Sanitizer will not show any output unless a memory leak is detected. I will try running some existing test cases.

I am only able to build with Address Sanitizer manually. When I try building using /etc/Build.sh. I am unable to pass the ASAN argument. I believe this is an issue with the shell script and not CMake, so I am considering opening a separate issue for this.

```
./etc/Build.sh -cmake='DASAN=ON'
[INFO] Saving logs to build/openroad_build.log
[INFO] ./etc/Build.sh -cmake=DASAN=ON
CMake Warning:
  Ignoring extra path from command line:

   "/home/ian/OpenROAD/DASAN=ON"
```

maliberty, 2023-05-20:
I think you need a dash in:
```
./etc/Build.sh -cmake='-DASAN=ON'
```

maliberty, 2023-05-20:
I think it is reasonable to add -g whenever asan is enabled.

Iyury1, 2023-05-27:
As I mentioned last week, I would like to run some existing test cases with the goal of finding a memory leak to confirm Address Sanitizer is giving output correctly. I am a little confused about how to do this.

I have built OpenROAD in a directory "OpenROAD/build". Any output from AddressSanitizer would need to be produced from the executable created in "OpenROAD/build/src".

The tests I thought I could run are either located in "OpenROAD/test/", or in the tool test folders in "OpenROAD/src/". I noticed that some tool regression tests I can run from my shell directly, although most tests report an error.

I have been trying to run "OpenROAD/test/gcd_nangate45.tcl". If I try to run this file from OpenROAD I receive an error "[ERROR STA-0511] cannot open '../../test/gcd_nangate.tcl'." If I try to run this from tclsh, I eventually get an error for "invalid command name "utl::suppress_message"", and I cannot locate which .tcl file defines this namespace.

Are these test cases meant to be run inside OpenROAD itself or by a TCL shell? Is this the wrong approach for what I want to achieve?

maliberty, 2023-05-27:
The simplest way to run the tests is to use the regression script.

Iyury1, 2023-05-29:
I ran the regression script and was a bit surprised at the results. LeakSanitizer is throwing errors on every single regression test. I found identical errors when compiling with both gnu and clang.

I was not expecting this many leaks to be reported. I will inspect some of the reported leaks to try to determine if these are false positives being reported.

The only changes I made to the build is to include the compiler flags "-fsanatize=address,-fno-omit-frame-pointer, -g" and linker flag "-fsanatize=address" with the argument ASAN=ON. I did this using add_compile_options(), add_link_options(), target_compile_options(), and target_link_options().

maliberty, 2023-05-30:
I'm not too surprised that there are some leaks but they are pretty small.  We should clean them up when we have some time.  It is nice to have this as an option.

PR #3427 (2023-06-07): Add CMake variable for building with Address Sanitizer
Created a new variable "ASAN" in the top-level CMakeLists.txt. This variable is OFF by default. To enable building with Address Sanitizer, set the ASAN variable to ON by passing the argument to CMake "-DASAN=ON".

When ASAN is ON, the src/CMakeLists.txt file adds compile and link options for using Address Sanitizer to the directory properties and also to the target properties. These compile and link options will be applied to all tools in the src directory and to openroad itself.

Fixes #2738
closed 2023-06-07 by maliberty