The-OpenROAD-Project/OpenROAD #2742: RePlAce diverged at newStepLength -- fixed with `PL_BASIC_PLACEMENT`

xobs, 2023-01-12:
### Describe the bug

As requested in https://github.com/The-OpenROAD-Project/OpenLane/pull/1576#issuecomment-1352562946 -- I've encountered a design that fails placement unless `PL_BASIC_PLACEMENT` is 1.

### Expected Behavior

Designs should never need `PL_BASIC_PLACEMENT` apparently, so this should not cause a design to suddenly work

### OpenROAD Environment

```shell
Please describe how to run this command -- openlane appears to use docker, so there is no `etc` directory to work with.
```

### OpenLane Environment

```shell
Kernel: Linux v5.15.79.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: e570a6a5b428eddb19f8efe4f9d18d79f1050940
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 3f9bdbd857564726b731760dc2c817e84ca7d8ac)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

e570a6a 2023-01-02T15:10:43+02:00 `run_designs` fixes (#1597) - Mohamed Gaber -  (grafted, HEAD, tag: 2023.01.03)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### To Reproduce

1. Clone `caravel-minimax`: `git clone -b pl-basic-placement-bug https://github.com/xobs/caravel-minimax.git`
2. Run `. activate-caravel.sh`
3. Update the paths in `openlane/mimi/config.json` (https://github.com/The-OpenROAD-Project/OpenLane/issues/1616)
4. Run `make mimi`

The `issue_reproducible` has also been attached: [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10400232/issue_reproducible.tar.gz)

### Relevant log output

```shell
[STEP 5]
[INFO]: Running Global Placement (log: ../si/work/caravel-minimax/openlane/mimi/runs/23_01_12_17_39/logs/placement/5-global.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl
[ERROR]: Log: ../si/work/caravel-minimax/openlane/mimi/runs/23_01_12_17_39/logs/placement/5-global.log
[ERROR]: Last 10 lines:
[NesterovSolve] Iter: 2640 overflow: 0.314285 HPWL: 42734126349
[NesterovSolve] Iter: 2650 overflow: 0.316778 HPWL: 43025225816
[NesterovSolve] Iter: 2660 overflow: 0.312883 HPWL: 43706165197
[NesterovSolve] Iter: 2670 overflow: 0.312504 HPWL: 44809153194
[NesterovSolve] Iter: 2680 overflow: 0.317737 HPWL: 46104438646
[NesterovSolve] Iter: 2690 overflow: 0.3169 HPWL: 47268095746
[NesterovSolve] Iter: 2700 overflow: 0.314167 HPWL: 48250061064
[ERROR GPL-0305] RePlAce diverged at newStepLength.
Error: gpl.tcl, 69 GPL-0305
child process exited abnormally

[ERROR]: Creating issue reproducible...
```

### Additional Context

Setting `PL_BASIC_PLACEMENT: 1` lets it get further, and I was under the impression that there was interest in removing this feature because it is confusing and works around a bug. That is why I'm submitting this issue.

maliberty, 2023-01-12:
Yes to "I was under the impression that there was interest in removing this feature because it is confusing and works around a bug. That is why I'm submitting this issue."  Thanks for submitting

maliberty, 2023-01-12:
I see a mismatch in expectations here:

[INFO GPL-0019] Util(%): 65.14
[INFO GPL-0023] TargetDensity: 0.45

You are trying to place at a lower density than is possible given your utilization.   I suspect you will find that even if you could place at 65% util you will have routing congestion issues.

maliberty, 2023-01-12:
Is it intentional that your RAMs have no connections to their D, Q, or WEN pins?  Because they have identical connectivity they tend to wind up exactly on top of each other which is a problem for the placer (they tend to feel the same forces and move identically).

xobs, 2023-01-13:
Thanks for pointing that out. I'm still trying to understand how macros work.

However, I've managed to come up with another repo that has the same placement issue but only uses Verilog.

The branch at https://github.com/xobs/caravel-minimax/tree/replace-divergence contains only Verilog code and has the same error that is fixed with `PL_BASIC_PLACEMENT` in the `minimax_rf` project. If you type `make minimax_rf` you will find that it fails:

```
[ERROR]: during executing openroad script /openlane/scripts/openroad/gpl.tcl
[ERROR]: Log: ../si/work/caravel-minimax/openlane/minimax_rf/runs/23_01_13_11_03/logs/placement/7-global.log
[ERROR]: Last 10 lines:
[NesterovSolve] Iter: 450 overflow: 0.236868 HPWL: 3469863079
[NesterovSolve] Iter: 460 overflow: 0.238695 HPWL: 3656905160
[NesterovSolve] Iter: 470 overflow: 0.234744 HPWL: 3948363037
[NesterovSolve] Iter: 480 overflow: 0.225043 HPWL: 4248587630
[NesterovSolve] Iter: 490 overflow: 0.211341 HPWL: 4521576227
[NesterovSolve] Iter: 500 overflow: 0.233532 HPWL: 4802824168
[NesterovSolve] Iter: 510 overflow: 0.261653 HPWL: 5207952959
[ERROR GPL-0307] RePlAce divergence detected. Re-run with a smaller max_phi_cof value.
Error: gpl.tcl, 69 GPL-0307
child process exited abnormally
```

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10407837/issue_reproducible.tar.gz)

Setting `PL_BASIC_PLACEMENT=1` fixes the issue. Adjusting other vlaues may also fix the issue, and I'm still trying to understand what those knobs do.

maliberty, 2023-02-10:
What makes this block interesting is it high aspect ratio.  That revealed an area for improvement in the global placer.  I'm glad you submitted this as its exactly the sort of thing I don't want to see swept under the rug of basic placement.  You should get a much better placement when this fix is merged.

maliberty, 2023-02-10:
[image]

xobs, 2023-02-10:
I'm trying to route DFFRAM, and it's not converging in global routing when trying to route a 32x32 regfile. Might that be improved with this fix?

[image]

Each iteration takes about 45 minutes, and it looks like a full run (which will probably fail) takes around 7 hours for "Global Placement".

QuantamHD, 2023-02-10:
Isn't DFFRAM already placed? You shouldn't need to do any kind of global or detailed placement.

xobs, 2023-02-10:
Not for GF180MCU -- that just got merged last week, and doesn't have prebuilt memories yet: https://github.com/AUCOHL/DFFRAM/commit/b143d4d6ad8a8760639d1128b19d88e9e97b764e

Additionally, I'll need a 64x32 regfile, so I'll need to route it myself eventually anyway.

maliberty, 2023-02-10:
It might help but I think it will be hard to route a generic placement in such a high aspect ratio with so few metal layers.  In any case it would be a separate issue.

PR #2879 (2023-02-10): gpl: consider the aspect ratio of the block when building the BinGrid
Fixes #2742

The test case provided has a ~16:1 aspect ratio.  This produces bins with a similar aspect ratio which is bad for convergence.  This change considers the aspect ration when constructing the BinGrid.  The grid must be a power of two in both X & Y but it doesn't have to be the same value.
commit 6726d3ee (2023-02-10): gpl: consider the aspect ratio of the block when building the BinGrid

Fixes #2742

The test case provided has a ~16:1 aspect ratio.  This produces bins
with a similar aspect ratio which is bad for convergence.  This change
considers the aspect ration when constructing the BinGrid.  The grid
must be a power of two in both X & Y but it doesn't have to be the
same value.
closed 2023-02-10 by maliberty