The-OpenROAD-Project/OpenROAD #2746: Segfault in Global Placement

xobs, 2023-01-13:
### Describe the bug

When running global placement, openroad segfaults somehow:

```
33# 0x00007F3BA1372F1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000000CC6CC7 in openroad
child killed: SIGABRT
```

### Expected Behavior

OpenROAD should not segfault, and should print an error if it is unable to do something.

### OpenROAD Environment

```shell
N/A -- there does not appear to be a way to show the OpenROAD environment when using it from the recommended Docker file
```

### OpenLane Environment

```shell
Kernel: Linux v5.15.79.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: e570a6a5b428eddb19f8efe4f9d18d79f1050940
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 3f9bdbd857564726b731760dc2c817e84ca7d8ac)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

e570a6a 2023-01-02T15:10:43+02:00 `run_designs` fixes (#1597) - Mohamed Gaber -  (grafted, HEAD, tag: 2023.01.03)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### To Reproduce

1. Clone https://github.com/xobs/caravel-minimax/
2. Check out the `global-placement-segfault` branch
3. Run `. activate-caravel.sh`
4. Run `make setup`
5. Run `make minimax_rv`

### Relevant log output

```shell
[492248.984269] potentially unexpected fatal signal 6.
[492248.984628] CPU: 22 PID: 31944 Comm: openroad Not tainted 5.15.79.1-microsoft-standard-WSL2 #1
[492248.985042] RIP: 0033:0x7f3b9c83c387
[492248.985178] Code: 02 00 00 85 f6 75 34 b8 ba 00 00 00 0f 05 89 c1 64 89 04 25 d0 02 00 00 89 c6 48 63 d7 48 63 f6 48 63 f9 b8 ea 00 00 00 0f 05 <48> 3d 00 f0 ff ff 77 1e f3 c3 0f 1f 80 00 00 00 00 85 c9 7f db 89
[492248.986030] RSP: 002b:00007ffcc03ecbc8 EFLAGS: 00000202
[492248.986191] RAX: 0000000000000000 RBX: 0000000000000041 RCX: 00007f3b9c83c387
[492248.986430] RDX: 0000000000000006 RSI: 000000000000015b RDI: 000000000000015b
[492248.986668] RBP: 0000000008e331c0 R08: 7463657078655f6e R09: 00007f3b9c8532cd
[492248.986901] R10: 0000000000000008 R11: 0000000000000202 R12: 0000000000000040
[492248.987185] R13: 000000000000101c R14: 0000000000000040 R15: 00000000095dda90
[492248.987407] FS:  00007f3ba1866140 GS:  0000000000000000
```

vijayank88, 2023-01-13:
@xobs 
can you try once again with latest OpenLane tag and confirm?

If segfault repeats follow instruction from here: https://openlane.readthedocs.io/en/latest/for_developers/using_or_issue.html#manually and create standalone test for placement stage and attach here.

xobs, 2023-01-13:
It still segfaults with `efabless/openlane:2023.01.12`

I'm using a project based on https://github.com/efabless/caravel_user_project -- the link you sent says to run `make mount`, however that does not work:

```
export CARAVEL_ROOT=/si/work/dependencies/caravel && make -f /si/work/dependencies/caravel/Makefile mount
make[1]: Entering directory '/si/work/caravel-minimax'
make[1]: *** No rule to make target 'mount'.  Stop.
make[1]: Leaving directory '/si/work/caravel-minimax'
make: *** [Makefile:84: mount] Error 2
```

Is there any documentation on how to do this from inside a project derived from the efabless caravel_user_project template?

vijayank88, 2023-01-13:
`export OPENLANE_IMAGE_NAME = efabless/openlane:2023.01.12`
set above command and try `make mount` inside `OPENLANE_ROOT` folder and try to create test case by following instructions.

xobs, 2023-01-13:
If I change to $OPENLANE_ROOT and run `make mount`, I'm dropped into a Docker container.

However, I'm not sure what to do next.

1. How do I access my project from within the container? On my system it's located under `/si/work/caravel-minimax/` but that directory doesn't exist in the container.
2. What commands do I need to run? The document says to run `./flow.tcl` and add `-verbose`. How do I do that? All of the documentation I read says to build things using Makefiles, and I just run `make minimax_rf`

vijayank88, 2023-01-13:
Ok. Just follow below method and end of the run `issue_reproducible` will generate.

In `gpl.tcl` line 19 https://github.com/The-OpenROAD-Project/OpenLane/blob/06b26813465d8745c2cdfe6605ac3233cef89dec/scripts/openroad/gpl.tcl#L19 add `exit 1` and save the file and run `make minimax_rf`.
Once flow exit with `issue_reproducible` attach the file here and remove `exit 1` from `gpl.tcl`.

xobs, 2023-01-13:
Here's the resulting issue_reproducible: [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10412365/issue_reproducible.tar.gz)

vijayank88, 2023-01-13:
@maliberty Issue confirmed.
To run test case remove `exit 1` from `openlane/scripts/openroad/gpl.tcl` at line 19 and segfault re-produced.
```
[INFO GPL-0100] worst slack 2.86e-08
[INFO GPL-0103] Weighted 778 nets.
[NesterovSolve] Iter: 450 overflow: 0.193818 HPWL: 3634692755
[INFO GPL-0075] Routability numCall: 1 inflationIterCnt: 1 bloatIterCnt: 0
[INFO GPL-0036] TileLxLy: 0 0
[INFO GPL-0037] TileSize: 16800 16800
[INFO GPL-0038] TileCnt: 345 29
[INFO GPL-0039] numRoutingLayers: 5
[INFO GPL-0040] NumTiles: 10005
[INFO GPL-0063] TotalRouteOverflowH2: 2607.142851948738
[INFO GPL-0064] TotalRouteOverflowV2: 0.0
[INFO GPL-0065] OverflowTileCnt2: 7646
[INFO GPL-0066] 0.5%RC: 2.0553935741891665
[INFO GPL-0067] 1.0%RC: 1.9882179732175218
[INFO GPL-0068] 2.0%RC: 1.902664679937412
[INFO GPL-0069] 5.0%RC: 1.7900118602756643
[INFO GPL-0070] 0.5rcK: 1.0
[INFO GPL-0071] 1.0rcK: 1.0
[INFO GPL-0072] 2.0rcK: 0.0
[INFO GPL-0073] 5.0rcK: 0.0
[INFO GPL-0074] FinalRC: 2.0218058
[INFO GPL-0045] InflatedAreaDelta: 693292624948
[INFO GPL-0046] TargetDensity: 0.33
[INFO GPL-0049] WhiteSpaceArea: 2416458598400
[INFO GPL-0050] NesterovInstsArea: 725777024000
[INFO GPL-0051] TotalFillerArea: 71654341632
[INFO GPL-0052] TotalGCellsArea: 797431365632
[INFO GPL-0053] ExpectedTotalGCellsArea: 1490723990580
[INFO GPL-0054] NewTargetDensity: 0.6169044
[INFO GPL-0055] NewWhiteSpaceArea: 2416458598400
[INFO GPL-0056] MovableArea: 1490723930112
[INFO GPL-0057] NewNesterovInstsArea: 1419069648948
[INFO GPL-0058] NewTotalFillerArea: 71654281164
[INFO GPL-0059] NewTotalGCellsArea: 1490723930112
/opt/rh/devtoolset-8/root/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = gpl::Bin*; _Alloc = std::allocator<gpl::Bin*>; std::vector<_Tp, _Alloc>::reference = gpl::Bin*&; std::vector<_Tp, _Alloc>::size_type = long unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
Signal 6 received
Stack trace:
 0# 0x0000000000CEB187 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 1# 0x00007F190F065400 in /lib64/libc.so.6
 2# gsignal in /lib64/libc.so.6
 3# abort in /lib64/libc.so.6
 4# 0x0000000000CEB262 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 5# gpl::BinGrid::updateBinsGCellDensityArea(std::vector<gpl::GCell*, std::allocator<gpl::GCell*> > const&) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 6# gpl::NesterovPlace::doNesterovPlace(int) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 7# 0x0000000000D3D7C1 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 8# 0x00007F1913BBBEB2 in /lib64/libtcl8.5.so
 9# 0x00007F1913C0036C in /lib64/libtcl8.5.so
10# TclObjInterpProcCore in /lib64/libtcl8.5.so
11# 0x00007F1913BBBEB2 in /lib64/libtcl8.5.so
12# 0x00007F1913C0036C in /lib64/libtcl8.5.so
13# 0x00007F1913C08647 in /lib64/libtcl8.5.so
14# TclEvalObjEx in /lib64/libtcl8.5.so
```

vijayank88, 2023-01-13:
@xobs 
If I increase density `set ::env(PL_TARGET_DENSITY) {0.65}` I'm not facing any segfault.
Can you try the same?

xobs, 2023-01-13:
Yes, if I increase the `PL_TARGET_DENSITY` then it gets further, but later on complains about the density being too high, and telling me to lower it.

Ultimately the solution to the problem in my design is to increase the area and redo my hand-placed macros, but I thought I'd report the segfault in openroad while I had a reproducible testcase.

mithro, 2023-01-13:
@vijayank88 - The tools should never segfault even if given non-sensible input values.

maliberty, 2023-01-15:
Fixed with https://github.com/The-OpenROAD-Project/OpenROAD/pull/2759

closed 2023-01-15 by maliberty