The-OpenROAD-Project/OpenROAD #2758: Segfault in Global Routing Resizer Timing Optimizations

xobs, 2023-01-15:
### Describe the bug

When running OpenLane to build a project based on `caravel_user_project`, a particular configuration causes openroad to segfault during the `Running Global Routing Resizer Timing Optimizations` step.

### Expected Behavior

openroad should not segfault

### OpenROAD Environment

```shell
`./etc/Env.sh` does not work because I'm using the recommended Docker flow, which hides openroad. I have reprodued this when building with OPENLANE_TAG=2023.01.12, so I'm using whichever openroad environment that pulls in.
```

### OpenLane Environment

```shell
Kernel: Linux v5.15.79.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: e570a6a5b428eddb19f8efe4f9d18d79f1050940
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: MISMATCH
The version of open_pdks used in building the PDK does not match the version OpenLane was tested on (installed: 0059588eebfc704681dc2368bd1d33d96281d10f, tested: 3f9bdbd857564726b731760dc2c817e84ca7d8ac)
This may introduce some issues. You may want to re-install the PDK by invoking `make pdk`.
---
Git Log (Last 3 Commits)

e570a6a 2023-01-02T15:10:43+02:00 `run_designs` fixes (#1597) - Mohamed Gaber -  (grafted, HEAD, tag: 2023.01.03)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane (push)
```

### To Reproduce

1. Clone `https://github.com/xobs/caravel-minimax/`
2. Check out the `global-routing-segfault` branch
3. Run `. activate-caravel.sh`
4. Run `make setup`
5. Run `make minimax_rv`
6. Run `make mimi`

### Relevant log output

```shell
[INFO GRT-0018] Total wirelength: 853893 um
[INFO GRT-0014] Routed nets: 2801
[INFO]: Setting RC values...
[INFO GRT-0009] rerouting 2 nets.
[INFO GRT-0001] Minimum degree: 5
[INFO GRT-0002] Maximum degree: 5
[INFO GRT-0009] rerouting 4 nets.
[INFO GRT-0001] Minimum degree: 3
[INFO GRT-0002] Maximum degree: 4
[INFO GRT-0009] rerouting 2 nets.
[INFO GRT-0001] Minimum degree: 5
[INFO GRT-0002] Maximum degree: 5
Signal 11 received
Stack trace:
 0# 0x0000000000CCC4E7 in openroad
 1# 0x00007EFD15F18400 in /lib64/libc.so.6
 2# rsz::BufferedNet::BufferedNet(rsz::BufferedNetType, odb::Point, std::shared_ptr<rsz::BufferedNet>, std::shared_ptr<rsz::BufferedNet>, rsz::Resizer const*) in openroad
 3# 0x0000000000E1E632 in openroad
 4# 0x0000000000E1E560 in openroad
 5# 0x0000000000E1E560 in openroad
 6# 0x0000000000E1E560 in openroad
 7# rsz::Resizer::makeBufferedNetGroute(sta::Pin const*, sta::Corner const*) in openroad
 8# rsz::Resizer::makeBufferedNet(sta::Pin const*, sta::Corner const*) in openroad
 9# rsz::RepairSetup::rebuffer(sta::Pin const*) in openroad
10# rsz::RepairSetup::repairSetup(sta::PathRef&, float) in openroad
11# rsz::RepairSetup::repairSetup(float, int) in openroad
12# 0x0000000000E1B03E in openroad
13# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
14# 0x00007EFD1AA9236C in /lib64/libtcl8.5.so
15# TclObjInterpProcCore in /lib64/libtcl8.5.so
16# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
17# 0x00007EFD1AA9236C in /lib64/libtcl8.5.so
18# 0x00007EFD1AA9A647 in /lib64/libtcl8.5.so
19# TclEvalObjEx in /lib64/libtcl8.5.so
20# 0x00007EFD1AAD527F in /lib64/libtcl8.5.so
21# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
22# 0x00007EFD1AA9236C in /lib64/libtcl8.5.so
23# 0x00007EFD1AA9A647 in /lib64/libtcl8.5.so
24# TclEvalObjEx in /lib64/libtcl8.5.so
25# 0x00007EFD1AA551D0 in /lib64/libtcl8.5.so
26# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
27# 0x00007EFD1AA9236C in /lib64/libtcl8.5.so
28# 0x00007EFD1AA9A647 in /lib64/libtcl8.5.so
29# TclEvalObjEx in /lib64/libtcl8.5.so
30# 0x00007EFD1AA58F00 in /lib64/libtcl8.5.so
31# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
32# 0x00007EFD1AA9236C in /lib64/libtcl8.5.so
33# TclObjInterpProcCore in /lib64/libtcl8.5.so
34# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
35# 0x00007EFD1AA9236C in /lib64/libtcl8.5.so
36# TclObjInterpProcCore in /lib64/libtcl8.5.so
37# 0x00007EFD1AA4DEB2 in /lib64/libtcl8.5.so
38# 0x00007EFD1AA4EF1E in /lib64/libtcl8.5.so
39# Tcl_EvalEx in /lib64/libtcl8.5.so
40# Tcl_Eval in /lib64/libtcl8.5.so
41# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
42# ord::tclAppInit(Tcl_Interp*) in openroad
43# Tcl_Main in /lib64/libtcl8.5.so
44# main in openroad
45# __libc_start_main in /lib64/libc.so.6
46# 0x0000000000CC6CC7 in openroad
```

### Additional Context

This design uses a macro. I am probably using the macro incorrectly, because I can only get this project to get past the `PDN` step by setting `FP_PDN_ENABLE_RAILS`. Otherwise it complains in step 7 "PDN":

```
[WARNING PDN-0189] Supply pin vdd of instance minimax.regfile_execution is not connected to any net.
[WARNING PDN-0189] Supply pin vss of instance minimax.regfile_execution is not connected to any net.
...
[INFO PDN-0001] Inserting grid: macro - minimax.regfile_execution
[INFO PDN-0001] Inserting grid: macro - minimax.regfile_microcode
...
[WARNING PSM-0038] Unconnected PDN node on net vdd at location (6.720um, 1509.200um), layer: 1.
[WARNING PSM-0038] Unconnected PDN node on net vdd at location (19.600um, 1509.200um), layer: 1.
[WARNING PSM-0038] Unconnected PDN node on net vdd at location (6.720um, 1501.360um), layer: 1.
[WARNING PSM-0038] Unconnected PDN node on net vdd at location (19.600um, 1501.360um), layer: 1.
[ERROR PSM-0069] Check connectivity failed.
Error: pdn.tcl, 39 PSM-0069
```

It's unclear what this message is saying, and very unclear how to resolve it.

However, even if I am using this macro incorrectly, the global router shouldn't segfault.

rovinski, 2023-01-15:
PDN is Power Delivery Network and PSM is Power SiMulation. PDN is what inserts the power grid at the top level, and PSM verifies that the connectivity is correct.

The error is saying that the macro has a power pin (`vdd`) which is not physically connected to the top-level power grid. As a result, this macro is unpowered and not correct.

There's a few reasons why this can happen, the best bet is to just open up the GUI and look at the coordinates specified. The macro might be too small for the macro grid specified. Macros might be too close to each other or have some other placement problem. The power pins on the macro might be poorly defined or incorrect.

It's not related to the segfault but is very likely an issue with the design.

xobs, 2023-01-15:
Thanks for the input. I thought it might be due to me feeding it something invalid, which explains the segfault. I'd still like it to not segfault, but at least I know it's a failure state due to bad input.

Do you have any suggestions on how I can troubleshoot this particular design? There are no macros at those coordinates -- my macros only begin at 30um, so anything less than 30 is empty space. Opening the last-available file also shows empty space at all reported locations such as `6.720um, 1501.360um`

xobs, 2023-01-15:
Okay, perhaps there's some clamping code in there that's causing it to wrap. I moved everything down by 200 um and eventually it managed to continue even with `FP_PDN_ENABLE_RAILS` as default unset.

However, it still segfaults when it gets to Step 13 `Running Global Routing Resizer Timing Optimizations`

xobs, 2023-01-15:
Here is the `issue_reproducible` from the run where the placement succeeded and PDN should have been run, yet it still segfaults: [issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10419676/issue_reproducible.tar.gz)

maliberty, 2023-01-15:
This is no a properly packaged test.  First I get:

Error: resizer_routing_timing.tcl, 15 cannot read file pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_1v80.lib

I manually copied over that file but then I get:

Error: resizer_routing_timing.tcl, 15 cannot read file /si/work/caravel-minimax/lib/minimax_rf.lib.

Please provide a test case that is complete.

xobs, 2023-01-16:
Is the `issue_reproducible` file not enough to reproduce the issue? What should I do in order to create the test?

The following script takes about 7 minutes to run on my system, and replicates it, but if there's something in particular you'd like me to provide from this directory I can do that:

```bash
#!/bin/bash
set -ex
git clone -b global-routing-segfault --recurse-submodules https://github.com/xobs/caravel-minimax.git
cd caravel-minimax
. activate-caravel.sh
make setup
make minimax_rf
make mimi
```

maliberty, 2023-01-16:
The reproducible doesn't contain the needed files I mentioned above.

vijayank88, 2023-01-16:
@xobs 
Please raise github issue in https://github.com/The-OpenROAD-Project/OpenLane/issues for the issue reproducible problem as its creating recursive folder name instead of respective file to copy. See below, its copied directory not files.
```
issue_reproducible/pdk/
issue_reproducible/pdk/gf180mcuC/
issue_reproducible/pdk/gf180mcuC/libs.ref/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_1v80.lib /
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_1v80.lib /si/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_1v80.lib /si/work/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_1v80.lib /si/work/caravel-minimax/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_ip_sram/liberty/gf180mcu_fd_ip_sram__sram512x8m8wm1__tt_025C_1v80.lib /si/work/caravel-minimax/lib/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__ss_125C_1v62.lib
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__tt_025C_3v30.lib
issue_reproducible/pdk/gf180mcuC/libs.ref/gf180mcu_fd_sc_mcu7t5v0/liberty/gf180mcu_fd_sc_mcu7t5v0__ff_n40C_5v50.lib
```

vijayank88, 2023-01-16:
@maliberty 
based on user repo, re-created the test case and crash confirmed: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10422997/issue_reproducible.zip)

xobs, 2023-01-16:
Filed an issue at https://github.com/The-OpenROAD-Project/OpenLane/issues/1624

PR #2772 (2023-01-17): rsz: fix crash when the global routes contain a dead end
The can happen when a guide connects into a pin wrong-way and we add an extra bit of guide on the adjacent layer to allow for bridging within a single gcell.

Fixes #2758
commit 862c50b7 (2023-01-17): rsz: fix crash when the global routes contain a dead end

The can happen when a guide connects into a pin wrong-way and we add
an extra bit of guide on the adjacent layer to allow for bridging
within a single gcell.

Fixes #2758
closed 2023-01-17 by maliberty