The-OpenROAD-Project/OpenROAD #2770: grt: pin offset code doesn't mark gcells crossed as used

antonblanchard, 2023-01-17:
### Describe the bug

I have a test case that routes a bus around an obstruction:

[image]

When looking at the grt 3D grid usage, the first two cells next to the pins all have 0 usage (ie 20 available tracks):

```
20 20  1  1  1  1 11 26 30 30 30 30 30  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
20 20  1  1  1  1  1 16 20 20 20 20 20  0 
```

The issue looks to be the pin offset code, which moves the grid position for all pins 2 gcells in, but doesn't add any cost to the first 2 gcells. That seems bad, because any congestion would make grt want to route back through those 2 "empty" gcells.

I can guess at what the pin offset code is trying to achieve (avoid congestion in the gcell containing the pin), but I wonder how much tuning has been done of this (1 or 0 could easily be a better default).

### Expected Behavior

Correct grid usage on all gcells

### Environment

```shell
Git commit: 1ff101efbd3bf6ce4952e93c9c6b8db7e13c9346
os: Fedora Linux 37 (Workstation Edition)
cmake version 3.25.1
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 15.0.6 (Fedora 15.0.6-2.fc37)
```

### To Reproduce

Run test case in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2636

arthurjolo, 2023-01-17:
The test case is missing the files on the folder "sky130hd/". Or are they the same as the ones in "sky130hs"?

maliberty, 2023-01-17:
hd and hs are not the same.  Most likely you can just copy them over from ORFS.

arthurjolo, 2023-01-17:
Anton, I am having problems running the test case. 
I get FAIL: met3-5 found when running the bus_route.tcl. What can I do?

antonblanchard, 2023-01-17:
> The test case is missing the files on the folder "sky130hd/". Or are they the same as the ones in "sky130hs"?

The test case has a symlink back to test/sky130hd. What problem are you seeing?

arthurjolo, 2023-01-17:
I copied the files from test/sky130hd, but when running bus_route.tcl I get FAIL: met3-5 found

antonblanchard, 2023-01-17:
> @antonblanchard I got the files correctly now. But when running bus_route.tcl I get FAIL: met3-5 found
> 
> foreach line $lines { if {[regexp {met[3-5]} $line] != -1} { puts "FAIL: met3-5 found" exit } }

The test case is meant to fail, there are a bunch of grt fixes that are waiting to be merged. It's just managing to also hit this issue.

You need to look at the 3d grid usage. I have a patch that dumps it in ascii form, perhaps we want something like that upstream. I forgot to mention it, but the issue is on `met2`, and this is the usage for that layer:

```
20 20  1  1  1  1 11 26 30 30 30 30 30  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
 0  0  0  0  0  0 20 20 20 20 20 20 20  0 
20 20  1  1  1  1  1 16 20 20 20 20 20  0 
```

I guess we could write numbers in the gui inside each gcell via a debug option too.

maliberty, 2023-01-17:
@gadfort it might be good in the congestion map to have an option to show capacity or usage

gadfort, 2023-01-17:
Right now we could just change the data source data to do that, but it might be a little more involved be able to display 2 pieces of data together.

maliberty, 2023-01-17:
I was thinking more of a choice of what to display - congestion, usage, or capacity - from the setting dialog

antonblanchard, 2023-01-18:
This is the patch to dump the grt 3d graph values in ascii, perhaps I should clean it up and add it as a proper debug option.

[grt-ascii.patch.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10444881/grt-ascii.patch.gz)

maliberty, 2023-01-18:
This is controlled by grt's set_macro_extension.  It is mostly a hack for process that have unidirectional routing where the guides might be too close to the macro to really allow access.  We route to 'fake pins' extended away in order to ensure that the final connection is right way to the real pin.  It seems we don't account for this during the fr initialization.

antonblanchard, 2023-01-18:
> This is controlled by grt's set_macro_extension. It is mostly a hack for process that have unidirectional routing where the guides might be too close to the macro to really allow access. We route to [...]

This test case isn't using set_macro_extension. Pin offset is 2 by default: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/grt/src/GlobalRouter.cpp#L94

arthurjolo, 2023-01-18:
> This test case isn't using set_macro_extension. Pin offset is 2 by default: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/grt/src/GlobalRouter.cpp#L94

This is aplied also to pins conected to Pads

arthurjolo, 2023-01-18:
I tired running with it set to 0 and the problem goes away

antonblanchard, 2023-01-18:
@maliberty also confirming what @arthurjolo said - `set_pin_offset 0` fixes it

eder-matheus, 2023-01-18:
> Just to verify - if you set the extension to zero does the problem go away?
> 
> @eder-matheus do you recall why we defaulted it to two?

We defaulted it to two as it was the best value across the designs we had at that time that avoided DRCs near the macro pins.

Actually, the idea was not necessarily to avoid congestion near the macros, but to avoid connections to the macro pins that would cause min spacing violations (with detours near routing obstructions or overlaps with other wires).

arthurjolo, 2023-01-20:
@antonblanchard I created a pull request mentioning this issue that should correct this issue, as I mention in the PR for this test case it also needs your fix for the function computeTrackAdjustments().

closed 2023-01-30 by arthurjolo