The-OpenROAD-Project/OpenROAD #2775: Command could not found openroad

JeffreyHJF, 2023-01-17:


JeffreyHJF, 2023-01-17:
[100%] Built target openroad
[100%] Linking CXX executable trTest
cd /home/eeuser/OPENROAD/OpenROAD/build/src/drt && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/trTest.dir/link.txt --verbose=1
/usr/bin/g++ -O3 -DNDEBUG CMakeFiles/trTest.dir/test/gcTest.cpp.o CMakeFiles/trTest.dir/test/fixture.cpp.o CMakeFiles/trTest.dir/test/stubs.cpp.o CMakeFiles/trTest.dir/__/gui/src/stub.cpp.o -o trTest  drt.a ../stt/stt.a ../dst/dst.a /usr/local/lib/libboost_serialization.a /usr/local/lib/libboost_system.a /usr/local/lib/libboost_thread.a ../dbSta/src/dbSta.a ../gui/gui.a ../odb/src/cdl/libcdl.a ../odb/src/defin/libdefin.a ../odb/src/def/libdef.a ../odb/src/def/libdefzlib.a ../odb/src/lefin/liblefin.a ../odb/src/lef/liblef.a ../odb/src/lef/liblefzlib.a ../odb/src/db/libdb.a ../odb/src/defout/libdefout.a ../odb/src/lefout/liblefout.a ../odb/src/db/libdb.a ../odb/src/defout/libdefout.a ../odb/src/lefout/liblefout.a ../odb/src/upf/upf.a ../odb/src/tm/libtm.a ../odb/src/zutil/libzutil.a /usr/lib/x86_64-linux-gnu/libtcl.so /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5.15.3 /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5.15.3 /usr/lib/x86_64-linux-gnu/libQt5Core.so.5.15.3 /home/eeuser/OPENROAD/OpenROAD/src/sta/app/libOpenSTA.a ../utl/utl.a /usr/local/lib/libspdlog.a /usr/lib/gcc/x86_64-linux-gnu/11/libgomp.so /usr/lib/x86_64-linux-gnu/libpthread.a /usr/local/lib/libboost_serialization.a /usr/lib/x86_64-linux-gnu/libz.so 
gmake[2]: Leaving directory '/home/eeuser/OPENROAD/OpenROAD/build'
[100%] Built target trTest
gmake[1]: Leaving directory '/home/eeuser/OPENROAD/OpenROAD/build'
/usr/local/bin/cmake -E cmake_progress_start /home/eeuser/OPENROAD/OpenROAD/build/CMakeFiles 0

real	11m16.908s
user	79m4.245s
sys	4m29.090s

vijayank88, 2023-01-17:
As per log installation successful. Execute below command now
```
cd /home/eeuser/OPENROAD/OpenROAD/build/src
./openroad -gui
```

JeffreyHJF, 2023-01-17:
Ok! It do work right now, Thanks for your help! I do appreciate it!

vijayank88, 2023-01-17:
> Other way add following format in `.bashrc` file and execute from anywhere `export OPENROAD_HOME="<path-to>/OpenROAD/build/src"` `source ~/.bashrc` `$OPENROAD_HOME/openroad -gui`

repeat this. It can be source anywhere and launched.

closed 2023-01-17 by vijayank88