The-OpenROAD-Project/OpenROAD #2777: Regression run for new docker build gives errors in dbsta find_clks{1,2}

cdleary, 2023-01-17:
### Describe the bug

- Build from source in a new docker ubuntu container
- Run `./test/regression`

Observe:

```
find_clks1 (tcl) *ERROR* segmentation violation
find_clks2 (tcl) *ERROR* segmentation violation  
```

find_clks1-tcl.log:

```
[INFO ODB-0222] Reading LEF file: Nangate45/Nangate45.lef
[INFO ODB-0223]     Created 22 technology layers
[INFO ODB-0224]     Created 27 technology vias
[INFO ODB-0225]     Created 135 library cells
[INFO ODB-0226] Finished LEF file:  Nangate45/Nangate45.lef
[INFO ODB-0222] Reading LEF file: pad.lef
[INFO ODB-0225]     Created 1 library cells
[INFO ODB-0226] Finished LEF file:  pad.lef
[INFO ODB-0128] Design: top
[INFO ODB-0130]     Created 3 pins.
[INFO ODB-0131]     Created 4 components and 22 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 6 connections.
[INFO ODB-0133]     Created 6 nets and 10 connections.
Signal 11 received
Stack trace:
 0# 0x000055FBDCE9416B in /home/cdleary/OpenROAD/build/src/openroad
 1# 0x00007FC7C7A03520 in /lib/x86_64-linux-gnu/libc.so.6
 2# 0x00007FC7C7B5E97D in /lib/x86_64-linux-gnu/libc.so.6
 3# odb::dbNet::getName[abi:cxx11]() in /home/cdleary/OpenROAD/build/src/openroad
 4# 0x000055FBDD3B8334 in /home/cdleary/OpenROAD/build/src/openroad
 5# 0x000055FBDD06E4DE in /home/cdleary/OpenROAD/build/src/openroad
 6# TclNRRunCallbacks in /lib/x86_64-linux-gnu/libtcl8.6.so
 7# 0x00007FC7CB89C028 in /lib/x86_64-linux-gnu/libtcl8.6.so
 8# Tcl_EvalEx in /lib/x86_64-linux-gnu/libtcl8.6.so
 9# Tcl_Eval in /lib/x86_64-linux-gnu/libtcl8.6.so
10# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in /home/cdleary/OpenROAD/build/src/openroad
11# ord::tclAppInit(Tcl_Interp*) in /home/cdleary/OpenROAD/build/src/openroad
12# Tcl_MainEx in /lib/x86_64-linux-gnu/libtcl8.6.so
13# main in /home/cdleary/OpenROAD/build/src/openroad
14# 0x00007FC7C79EAD90 in /lib/x86_64-linux-gnu/libc.so.6
15# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
16# _start in /home/cdleary/OpenROAD/build/src/openroad
ERROR segmentation violation
```

Not sure why this would not show in CI? LMK if I likely did something wrong on my end.

### Expected Behavior

No errors from regression.

### Environment

```shell
$ ./etc/Env.sh 
Git commit: deed8b941265211d244d78d3e150fe9449b3b49f
kernel: Linux 5.19.11-1rodete1-amd64
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.24.2
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Ubuntu clang version 14.0.0-1ubuntu1
```
```

### To Reproduce

```
git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
sudo ./etc/DependencyInstaller.sh
mkdir build
cd build
cmake ..
make -j
cd ..
./test/regression
```

maliberty, 2023-01-17:
@cdleary do you plan to look into this or are you asking for help?

cdleary, 2023-01-17:
@maliberty Working on a different bug at the moment, if nobody else is going to be able to look soon I can put it in my queue?

maliberty, 2023-01-17:
We also ready run the unit tests in a ubuntu docker container and they pass.  See https://jenkins.openroad.tools/job/OpenROAD-Public/job/master/ "Test docker ubuntu20".  How is this different?

cdleary, 2023-01-17:
Only meaningful difference I see in the logs is there's an argument to the regression script. Let me take assignment until I have something more meaningfully detailed to report.

QuantamHD, 2023-01-17:
@cdleary Can you check the commit of the OpenROAD repo as well as the commit of the sta submodule?

src/sta sometimes goes out of sync.

cdleary, 2023-01-17:
$ git submodule status --recursive
 2a2b5cd59f5d1b596f20cbd39609c52b33299e06 src/sta (v2.2.0-524-g2a2b5cd)
 95b3543e928640dfa25f9e882e72a090a8883a9c third-party/abc (heads/master)
cdleary@645a06e7adc4:~/OpenROAD$ git show
commit deed8b941265211d244d78d3e150fe9449b3b49f (HEAD -> master, origin/master, origin/HEAD)
Merge: 7c8ef24f8 6c995db65
Author: Matt Liberty <mliberty@eng.ucsd.edu>
Date:   Tue Jan 17 10:41:38 2023 -0800

    Merge pull request #2761 from Pinata-Consulting/grt-congestion-report-tip
    
    grt: add load congestion report tip routing failure

cdleary, 2023-01-17:
Just some debugging notes -- I don't have experience debugging across TCL swig bindings yet so getting my sea legs:

```
cdleary@645a06e7adc4:~/OpenROAD/src/dbSta/test$ gdb --args ../../../build/src/openroad find_clks1.tcl 
[INFO ODB-0133]     Created 6 nets and 10 connections.
                                                                                          
Thread 1 "openroad" received signal SIGSEGV, Segmentation fault.                                                                                                                                                                                                                                                                                                           
__strlen_avx2 () at ../sysdeps/x86_64/multiarch/strlen-avx2.S:74                    
74      ../sysdeps/x86_64/multiarch/strlen-avx2.S: No such file or directory.                                                                                                                                                                                                                                                                                              
(gdb) up                
#1  0x0000560ab4929465 in std::char_traits<char>::length (__s=0x1df6da55e8d369e8 <error: Cannot access memory at address 0x1df6da55e8d369e8>) at /usr/include/c++/11/bits/char_traits.h:399
399             return __builtin_strlen(__s);     
(gdb) p *net
$9 = {<odb::_dbObject> = {<odb::dbObject> = {static max_name_length = 256}, _oid = 3720657370}, _flags = {_sig_type = 15, _wire_type = odb::dbWireType::NONE, _special = 0, _wild_connect = 1, _wire_ordered = 1, _buffered = 0, _disconnected = 1, _spef = 0, _select = 1, _mark = 0, _mark_1 = 0, _wire_altered = 0, _extracted = 0, _rc_graph = 0, _reduced = 0, 
    _set_io = 0, _io = 0, _dont_touch = 0, _fixed_bump = 0, _source = odb::dbSourceType::NONE, _rc_disconnected = 0, _block_rule = 0}, _name = 0x1df6da55e8d369e8 <error: Cannot access memory at address 0x1df6da55e8d369e8>, {_gndc_calibration_factor = 4.48415509e-44, _refCC = 4.48415509e-44}, {_cc_calibration_factor = 0, _dbCC = 0, _CcMatchRatio = 0}, 
  _next_entry = {_id = 33}, _iterms = {_id = 0}, _bterms = {_id = 3177623264}, _wire = {_id = 22026}, _global_wire = {_id = 3177568048}, _swires = {_id = 22026}, _cap_nodes = {_id = 13}, _r_segs = {_id = 0}, _non_default_rule = {_id = 49}, guides_ = {_id = 0}, 
  _groups = {<std::vector<odb::dbId<odb::_dbGroup>, std::allocator<odb::dbId<odb::_dbGroup> > >> = std::vector of length -23649946161907, capacity -23651019903731 = {{_id = 3036091340}, {_id = 22026}, {_id = 3036091402}, {_id = 22026}, {_id = 3036091450}, {_id = 22026}, {_id = 3036091516}, {_id = 22026}, {_id = 3036091662}, {_id = 22026}, {_id = 0}, {
        _id = 0}, {_id = 3130077136}, {_id = 22026}, {_id = 3036039428}, {_id = 22026}, {_id = 3036039518}, {_id = 22026}, {_id = 3036039566}, {_id = 22026}, {_id = 3036039756}, {_id = 22026}, {_id = 0}, {_id = 0}, {_id = 3130077160}, {_id = 22026}, {_id = 3036038354}, {_id = 22026}, {_id = 3036038384}, {_id = 22026}, {_id = 0}, {_id = 0}, {_id = 3130077200}, 
      {_id = 22026}, {_id = 3036038148}, {_id = 22026}, {_id = 3036038234}, {_id = 22026}, {_id = 4294967256}, {_id = 4294967295}, {_id = 3130077200}, {_id = 22026}, {_id = 3036038224}, {_id = 22026}, {_id = 3036038281}, {_id = 22026}, {_id = 0}, {_id = 0}, {_id = 3130077256}, {_id = 22026}, {_id = 3036036876}, {_id = 22026}, {_id = 3036036966}, {
        _id = 22026}, {_id = 0}, {_id = 0}, {_id = 3130077280}, {_id = 22026}, {_id = 3036036116}, {_id = 22026}, {_id = 3036036218}, {_id = 22026}, {_id = 0}, {_id = 0}, {_id = 3130077296}, {_id = 22026}, {_id = 3035952004}, {_id = 22026}, {_id = 3035952490}, {_id = 22026}, {_id = 3035966352}, {_id = 22026}, {_id = 3035966504}, {_id = 22026}, {
        _id = 3035966198}, {_id = 22026}, {_id = 3036027288}, {_id = 22026}, {_id = 3036045514}, {_id = 22026}, {_id = 3036045538}, {_id = 22026}, {_id = 3036045562}, {_id = 22026}, {_id = 3035952538}, {_id = 22026}, {_id = 3035966656}, {_id = 22026}, {_id = 3035967538}, {_id = 22026}, {_id = 3035967592}, {_id = 22026}, {_id = 3035968670}, {_id = 22026}, {
        _id = 3035970312}, {_id = 22026}, {_id = 3035968914}, {_id = 22026}, {_id = 3035968792}, {_id = 22026}, {_id = 3035969168}, {_id = 22026}, {_id = 3035969046}, {_id = 22026}, {_id = 3035969302}, {_id = 22026}, {_id = 3035969424}, {_id = 22026}, {_id = 3035969546}, {_id = 22026}, {_id = 3035971052}, {_id = 22026}, {_id = 3035971864}, {_id = 22026}, {
        _id = 3035971636}, {_id = 22026}, {_id = 3035972416}, {_id = 22026}, {_id = 3035972668}, {_id = 22026}, {_id = 3035972542}, {_id = 22026}, {_id = 3035972794}, {_id = 22026}, {_id = 3035972942}, {_id = 22026}, {_id = 3035973090}, {_id = 22026}, {_id = 3035973238}, {_id = 22026}, {_id = 3035973446}, {_id = 22026}, {_id = 3035973628}, {_id = 22026}, {
        _id = 3035973696}, {_id = 22026}, {_id = 3035973764}, {_id = 22026}, {_id = 3035973924}, {_id = 22026}, {_id = 3035973992}, {_id = 22026}, {_id = 3035974060}, {_id = 22026}, {_id = 3035974220}, {_id = 22026}, {_id = 3035974288}, {_id = 22026}, {_id = 3035974356}, {_id = 22026}, {_id = 3035974516}, {_id = 22026}, {_id = 3035974698}, {_id = 22026}, {
        _id = 3035974806}, {_id = 22026}, {_id = 3035974956}, {_id = 22026}, {_id = 3035975064}, {_id = 22026}, {_id = 3035975172}, {_id = 22026}, {_id = 3035975240}, {_id = 22026}, {_id = 3035975308}, {_id = 22026}, [... cut ...] _id = 22026}, {_id = 3035976570}, {_id = 22026}, {_id = 3035976632}, {_id = 22026}, {_id = 3035976694}, {_id = 22026}, {_id = 3035977050}, {_id = 22026}, {
        _id = 3035978074}, {_id = 22026}, {_id = 3035978118}, {_id = 22026}, {_id = 3035978162}, {_id = 22026}...}, <No data fields>}, _weight = -1117348160, _xtalk = 22026, _ccAdjustFactor = 0, _ccAdjustOrder = 0, _drivingIterm = 49}
  (gdb) p net->_name
$12 = 0x1df6da55e8d369e8 <error: Cannot access memory at address 0x1df6da55e8d369e8>
```

cdleary, 2023-01-17:
Perhaps different TCL libraries, platform vs what `./etc/DependencyInstaller.sh` gave me / cmake linked against?

```
$ dpkg --list | grep -i tcl
ii  libtcl8.6:amd64                      8.6.12+dfsg-1build1                     amd64        Tcl (the Tool Command Language) v8.6 - run-time library files
ii  tcl                                  8.6.11+1build2                          amd64        Tool Command Language (default version) - shell
ii  tcl-dev:amd64                        8.6.11+1build2                          amd64        Tool Command Language (default version) - development files
ii  tcl-tclreadline:amd64                2.3.8-1                                 amd64        GNU Readline Extension for Tcl/Tk
ii  tcl8.6                               8.6.12+dfsg-1build1                     amd64        Tcl (the Tool Command Language) v8.6 - shell
ii  tcl8.6-dev:amd64                     8.6.12+dfsg-1build1                     amd64        Tcl (the Tool Command Language) v8.6 - development files
ii  tcllib                               1.20+dfsg-1                             all          Standard Tcl Library
```

QuantamHD, 2023-01-17:
Maybe try running cmake again in the folder, and seeing what versions of libs you're using.

cdleary, 2023-01-17:
Looks like there's only one version of tcl so far as I can tell and it's the one `ldd` shows too.

Valgrind indicates we're reading from something that got destructed, looking into that:

```
==86330== Invalid read of size 8                                                          
==86330==    at 0x6BB7150: odb::dbNet::getName[abi:cxx11]() (dbNet.cpp:629)             
==86330==    by 0x41EEE50: _wrap_dbNet_getName (odbTCL_wrap.cxx:28731)                    
==86330==    by 0x3D96191: SWIG_Tcl_MethodCommand (dbStaTCL_wrap.cxx:1328)            
==86330==    by 0xA7E6D31: TclNRRunCallbacks (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)  
==86330==    by 0xA7E8027: ??? (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)             
==86330==    by 0xA7E7AA6: Tcl_EvalEx (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)      
==86330==    by 0xA7E8C99: Tcl_Eval (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)                                                                                                      
==86330==    by 0x3E434F2: sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) (StaMain.cc:95)                                                                                  
==86330==    by 0x3A868A0: tclAppInit(int&, char**, char const*, Tcl_Interp*) (Main.cc:404)                                                                                          
==86330==    by 0x3A86B63: ord::tclAppInit(Tcl_Interp*) (Main.cc:433)                                                                                                                
==86330==    by 0xA89AB18: Tcl_MainEx (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)         
==86330==    by 0x3A86205: main (Main.cc:284)                                         
==86330==  Address 0x13140018 is 8 bytes inside a block of size 16 free'd            
==86330==    at 0xA793B6F: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==86330==    by 0x3CAFA30: __gnu_cxx::new_allocator<odb::dbNet*>::deallocate(odb::dbNet**, unsigned long) (new_allocator.h:145)                                                      
==86330==    by 0x3CAA76B: std::allocator_traits<std::allocator<odb::dbNet*> >::deallocate(std::allocator<odb::dbNet*>&, odb::dbNet**, unsigned long) (alloc_traits.h:496)
==86330==    by 0x3CA6597: std::_Vector_base<odb::dbNet*, std::allocator<odb::dbNet*> >::_M_deallocate(odb::dbNet**, unsigned long) (stl_vector.h:354)                               
==86330==    by 0x3CA29B1: std::_Vector_base<odb::dbNet*, std::allocator<odb::dbNet*> >::~_Vector_base() (stl_vector.h:335)                                                          
==86330==    by 0x3CA2A06: std::vector<odb::dbNet*, std::allocator<odb::dbNet*> >::~vector() (stl_vector.h:683)                                                                      
==86330==    by 0x3E1D836: SwigValueWrapper<std::vector<odb::dbNet*, std::allocator<odb::dbNet*> > >::SwigSmartPointer::~SwigSmartPointer() (dbStaTCL_wrap.cxx:1849)                 
==86330==    by 0x3E1AD43: SwigValueWrapper<std::vector<odb::dbNet*, std::allocator<odb::dbNet*> > >::~SwigValueWrapper() (TritonCTS-pyPYTHON_wrap.cxx:3022)                         
==86330==    by 0x3E1820D: _wrap_find_all_clk_nets (dbStaTCL_wrap.cxx:39945)
==86330==    by 0xA7E6D31: TclNRRunCallbacks (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==86330==    by 0xA7E8027: ??? (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==86330==    by 0xA7E7AA6: Tcl_EvalEx (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)         
==86330==  Block was alloc'd at                   
==86330==    at 0xA791013: operator new(unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)                                                                  
==86330==    by 0x3CB7D4F: __gnu_cxx::new_allocator<odb::dbNet*>::allocate(unsigned long, void const*) (new_allocator.h:127)                                                         
==86330==    by 0x3CB44CA: std::allocator_traits<std::allocator<odb::dbNet*> >::allocate(std::allocator<odb::dbNet*>&, unsigned long) (alloc_traits.h:464)                                                                                                                                                                                                                 
==86330==    by 0x3CAFC59: std::_Vector_base<odb::dbNet*, std::allocator<odb::dbNet*> >::_M_allocate(unsigned long) (stl_vector.h:346)                                               
==86330==    by 0x3E1F1C4: void std::vector<odb::dbNet*, std::allocator<odb::dbNet*> >::_M_range_initialize<std::_Rb_tree_const_iterator<odb::dbNet*> >(std::_Rb_tree_const_iterator<odb::dbNet*>, std::_Rb_tree_const_iterator<odb::dbNet*>, std::forward_iterator_tag) (stl_vector.h:1582)                                                                               
==86330==    by 0x3E1C211: std::vector<odb::dbNet*, std::allocator<odb::dbNet*> >::vector<std::_Rb_tree_const_iterator<odb::dbNet*>, void>(std::_Rb_tree_const_iterator<odb::dbNet*>, std::_Rb_tree_const_iterator<odb::dbNet*>, std::allocator<odb::dbNet*> const&) (stl_vector.h:657)                                                                                    
==86330==    by 0x3DA4EB7: find_all_clk_nets() (dbStaTCL_wrap.cxx:7289)
==86330==    by 0x3E1814E: _wrap_find_all_clk_nets (dbStaTCL_wrap.cxx:39923)
==86330==    by 0xA7E6D31: TclNRRunCallbacks (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==86330==    by 0xA7E8027: ??? (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==86330==    by 0xA7E7AA6: Tcl_EvalEx (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
==86330==    by 0xA7E8C99: Tcl_Eval (in /usr/lib/x86_64-linux-gnu/libtcl8.6.so)
```

cdleary, 2023-01-17:
(I do think the CI is running ubuntu 20 whereas I'm running ubuntu 22. Going to just make sure I don't see it with 20 first before digging more.)

gadfort, 2023-01-18:
I'm running it on Ubuntu 22 and I do not see this segfault either, so I don't think that would be the cause.

vvbandeira, 2023-01-19:
I was able to reproduce the error on the docker image. Looks like the problem is with the version of `swig`:
`4.0.2` -> ok
`4.1.0` -> segv
CI is still at `4.0.2`.
I will investigate this further to try and fix.

cdleary, 2023-01-19:
@vvbandeira I see in the release notes it mentions better C++11 and beyond support, perhaps move semantics for the smart wrapper aren't right if those were just enabled in some fashion:

> - Improved C99, C++11, C++14, C++17 support. Start adding C++20 standard.
> -- https://github.com/swig/swig/blob/master/RELEASENOTES

cdleary, 2023-01-19:
In `build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx` switching:

`      odb::dbNet* ptr = (((std::vector< odb::dbNet * > &)result)[i]);`

To:

`      odb::dbNet* ptr = (*&result)[i];`

Causes the issue to go away, which seems to confirm the rough suspected cause. It's doing a C-style cast to an lvalue reference when that SwigSmartPointer type defined in that file has a rvalue reference cast overload, gonna take a minute to figure out what magic function gets invoked when here. :-) Notably the rvalue reference cast operator overload is guarded by `#if __cplusplus>=201103` which seems like it'd be introduced in this rev from those release notes, which is some additional confirmation.

maliberty, 2023-01-19:
What function does this code appear in?

maliberty, 2023-01-19:
If it is _wrap_find_all_clk_nets then I see the same code generated by 4.0.1.

Fwiw I built on ubuntu22 (using wsl) with 4.1.0 and got no crash.

cdleary, 2023-01-19:
Ok, think I've got a handle on it. We're C-style casing to an lvalue reference, and `SwigSmartPointer<T>operatorT&&()` is not being invoked. So it's just giving us a stack location where the struct is kept (i.e. a wild reference). If you add the following to the `#if __cplusplus >=201103L` part of the code everything passes:

```
  operator T&() const { return *pointer.ptr; }
```

So we just need to figure out how to supplement the `SwigSmartPointer` with that cast overload.

> What function does this code appear in?

@maliberty If you look in `build/src/dbSta/src/CMakeFiles/dbSta.dir/dbStaTCL_wrap.cxx` with this version of swig in use you should see the following snippets that cause the problem:

```
 1844 /* SwigValueWrapper is described in swig.swg */
 1845 template<typename T> class SwigValueWrapper {
 1846   struct SwigSmartPointer {
 1847     T *ptr;
 1848     SwigSmartPointer(T *p) : ptr(p) { }
 1849     ~SwigSmartPointer() { delete ptr; }
 1850     SwigSmartPointer& operator=(SwigSmartPointer& rhs) { T* oldptr = ptr; ptr = 0; delete oldptr; ptr = rhs.ptr; rhs.ptr = 0; return *this; }
 1851     void reset(T *p) { T* oldptr = ptr; ptr = 0; delete oldptr; ptr = p; }
 1852   } pointer;
 1853   SwigValueWrapper& operator=(const SwigValueWrapper<T>& rhs);
 1854   SwigValueWrapper(const SwigValueWrapper<T>& rhs);
 1855 public:
 1856   SwigValueWrapper() : pointer(0) { }
 1857   SwigValueWrapper& operator=(const T& t) { SwigSmartPointer tmp(new T(t)); pointer = tmp; return *this; }
 1858 #if __cplusplus >=201103L
 1859   SwigValueWrapper& operator=(T&& t) { SwigSmartPointer tmp(new T(std::move(t))); pointer = tmp; return *this; }
 1860   operator T&&() const { return std::move(*pointer.ptr); }
 1861 #else
 1862   operator T&() const { return *pointer.ptr; }
 1863 #endif
 1864   T *operator&() const { return pointer.ptr; }
 1865   static void reset(SwigValueWrapper& t, T *p) { t.pointer.reset(p); }
 1866 };
```

```
39916 SWIGINTERN int
39917 _wrap_find_all_clk_nets(ClientData clientData SWIGUNUSED, Tcl_Interp *interp, int objc, Tcl_Obj *CONST objv[]) {
39918   SwigValueWrapper< std::vector< odb::dbNet * > > result;
39919 
39920   if (SWIG_GetArgs(interp, objc, objv,":sta::find_all_clk_nets ") == TCL_ERROR) SWIG_fail;
39921   {
39922     try {
39923       result = find_all_clk_nets();
39924     }
39925     catch (std::bad_alloc &) {
39926       fprintf(stderr, "Error: out of memory.");
39927       exit(0);
39928     }
39929     // This catches std::runtime_error (utl::error) and sta::Exception.
39930     catch (std::exception &excp) {
39931       Tcl_ResetResult(interp);
39932       Tcl_AppendResult(interp, excp.what(), nullptr);
39933       return TCL_ERROR;
39934     }
39935   }
39936   {
39937     for (unsigned int i = 0; i < (&result)->size(); i++) {
39938       odb::dbNet* ptr = (((std::vector< odb::dbNet * > &)result)[i]);
39939       Tcl_ListObjAppendElement(interp, (Tcl_GetObjResult(interp)),  SWIG_NewInstanceObj(ptr, SWIGTYPE_p_odb__dbNet, 0));
39940     }
39941   }
39942   return TCL_OK;
39943 fail:
39944   return TCL_ERROR;
39945 }
```

maliberty, 2023-01-19:
I see the same code with 4.0.1 and 4.1.  It works for me both ways.  There must be some other factor that is the problem here.  What compiler are you using?

cdleary, 2023-01-19:
With the same swig used I believe we were able to observe the same thing -- have a fix coming.

Root cause was that with the change in https://github.com/swig/swig/commit/bf36bf7d8aa8d130f30d0cbb2c260966f96bc108 (noted in the PR description it was causing backwards incompatibility) the previous type maps were using a cast to lvalue reference which is not overloaded for C++ >= C++11 in the SWIG implementation as of 4.1.0. Fix is to change the type maps to use safer conversions -- previously we ended up casting the struct value homed on the stack (containing a pointer in the SwigValueWrapper) to a lvalue vector reference, which clearly made for a wild reference.

With this fix I imagine the 4.1.0 dep and onwards should be good, I'm not seeing test failures anymore -- not sure if it means CI should update its dep as well before we close this. I'll leave that to @maliberty to determine if he wants to open a separate issue for that / how to track.

closed 2023-01-20 by cdleary