The-OpenROAD-Project/OpenROAD #2781: Add a GUI option to see all buffer connected nets

maliberty, 2023-01-18:
### Description

The inspector lets you see a single net easily.  However a single logical net may have been broken into many physical nets during buffering.  There should be some way to easily visualize all the buffer connected nets.

### Suggested Solution

I'm not positive of the best way to do this.  I am thinking of a inspector property on the net like "Aggregate Net" that would be a GUI object (not db) that contains a list of nets and buffers.  The highlight could draw the individual nets in different colors.  Here's an example I did by hand of tracing out such a aggregate (the root segment is green):
[image]

maliberty, 2023-01-18:
To be able to use the 'Route Guides' and focus buttons on the complex would be great too.

gadfort, 2023-01-22:
@maliberty Maybe calling it something like logical net might be a better term?
Also would you consider inverters and buffers the same?

maliberty, 2023-01-22:
I am hunting for a good name.  I'm concerned that logical net might be confused with nets in the hierarchical Verilog (which we will be storing in odb soon I hope).  "Net Complex" came to mind but I don't love that either.

Inverters are tricky.  I'm not interested in tracing through a user inverter but if resizer put in a pair instead of a buffer then I would be.  rsz doesn't do that today so its a bit theoretical.  Perhaps SOURCE NETLIST vs TIMING could be used to distinguish.

rovinski, 2023-01-22:
"Buffered Path"?

maliberty, 2023-01-22:
> "Buffered Path"?

Kinda. Its not a path like a timing path and is more of a tree of nets.  "Buffered Net" but that is also a bit odd.  Maybe there is no perfect, short name.

rovinski, 2023-01-22:
"Buffered (net) fanout" then?

maliberty, 2023-01-23:
> "Buffered (net) fanout" then?

Depending on which net you select you could be talking about fanout or fanin or both.  Perhaps "Buffer tree"?

gadfort, 2023-01-29:
@maliberty This PR adds the type, but does not implement the route guides

gadfort, 2024-10-18:
Competed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/2824

closed 2024-10-18 by gadfort