The-OpenROAD-Project/OpenROAD #2782: Should old congestion.rpt files be left lying around?

maliberty, 2023-01-18:
### Discussed in https://github.com/The-OpenROAD-Project/OpenROAD/discussions/2780

<div type='discussions-op-text'>

<sup>Originally posted by **oharboe** January 18, 2023</sup>
Old congestion.rpt files are not deleted upon success, which can be confusing.

Is this intentional?

https://github.com/The-OpenROAD-Project/OpenROAD/blob/9d17aaf662bcc3266ff452b7be2e6936f9221e76/src/grt/src/GlobalRouter.cpp#L282</div>

gadfort, 2023-01-19:
Is the information completely saved in the database? Ie. if I close OpenROAD and reopen it, is this information viewable?
I used this during my last tapeout since it only contains the problem areas it's much easier to see and use the DRC viewer for debugging a design. @maliberty I don't know if we want to get rid of it completely.

maliberty, 2023-01-19:
I think the point here is that if one run had errors and wrote the file, then a subsequent run without errors should remove it.

gadfort, 2023-01-19:
Ahh, that makes sense. Thanks for clarifying

tspyrou, 2023-02-07:
fixed by merge listed above.

closed 2023-02-07 by tspyrou