The-OpenROAD-Project/OpenROAD #2791: Should we fork LEMON

QuantamHD, 2023-01-19:
### Description

The [LEMON](https://lemon.cs.elte.hu/trac/lemon) library is a required by OpenROAD, but the maintainers are not very responsive to bug fixes.

The last official release of LEMON was in 2014, but it has commits as recent as 2020. We've since run into some problem with the library not supporting C++20, and were required to introduce patches.

LEMON is pretty widely distributed by package managers so by requiring a custom version of this library we're potentially introducing a conflict. If we do so I think it would be best to hard fork LEMON and make sure to rename its install directory.

maliberty, 2023-01-19:
Are your patches only required to compile lemon and not to use it as a library?  If so we should try to contribute them back and see what happens.

QuantamHD, 2023-01-20:
LEMON is largely a header only lib, and the patch affects the header.

maliberty, 2023-01-20:
So the widely distributed version by package managers will never work with c++ unless they do a new release?  That seems like enough of an issue to fork over if they will not take the changes and put out a release.  Perhaps try asking on their email lists and see if there is anyone willing/able to do a release.

mithro, 2023-01-21:
FYI - Be careful, if you fork a project you are now responsible for all its maintenance going forward.

gadfort, 2023-01-22:
When I suggested forking this, it was over the prospect that OpenROAD is dependent on this project, but they have not been responsive to patches like the one required for C++20. I don't think OpenROAD project should take over the project (clearly over the last 10 years, people have been happy with no fixes, so I would imagine that this would not add much in overhead).  LEMON does not appear in Ubuntu 20's repos, so I would guess that this would drop over time as they become more and more incompatible with modern compilers.

The only maintenance this project should be responsible for, is that required for OpenROAD (in my opinion).

maliberty, 2023-01-22:
I don't feel any need to be "responsible for all its maintenance going forward"

gadfort, 2023-03-19:
I know this topic is a bit stale, but it looks like the lemon downloads are not working... So I think it's worth revisiting at least to ensure functional installs

QuantamHD, 2023-03-19:
Agreed. @vvbandeira can you create a repo called LEMON in the organization, and give myself and @gadfort  push access

gadfort, 2023-03-19:
@QuantamHD do you happen to have a copy? I don't seem to have one on any of my machines.

maliberty, 2023-03-19:
@gadfort https://github.com/mdfbaam/Lemon appears to be a copy for now

@vvbandeira looks like their site is down which is a good argument for doing this.

QuantamHD, 2023-03-19:
I have a copy of the mercurial repo at work. Internet archive should also have a copy.

QuantamHD, 2023-03-19:
Here's the copy of the [mercurial](https://www.mercurial-scm.org/) repo.  You'll need to download mercurial if you want to use SCM commands. right now the repo's head is at my C++20 changes.

[lemon.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/11012324/lemon.tar.gz)

maliberty, 2023-03-20:
The lemon website is back online

gadfort, 2023-06-22:
@maliberty it would appear this is down again

maliberty, 2023-06-22:
We should make a GH mirror in our project.

PR #3507 (2023-06-22): Use our own fork of lemon
This change is to avoid recent intermittent network issues.

Closes #2791
closed 2023-06-22 by vvbandeira