The-OpenROAD-Project/OpenROAD #2807: make commands errors out

ratul619, 2023-01-23:
### Describe the bug

Hi , 

Getting error in make command : Please help

/usr/bin/ld: odb/src/zutil/libzutil.a(util.cpp.o): in function `odb::cutRows(odb::dbBlock*, int, std::vector<odb::dbBox*, std::allocator<odb::dbBox*> > const&, int, int, utl::Logger*)':
util.cpp:(.text+0x107a): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: util.cpp:(.text+0x10ba): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: util.cpp:(.text+0x1523): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: util.cpp:(.text+0x155c): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: util.cpp:(.text+0x1dc5): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: util.cpp:(.text+0x1e05): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
/usr/bin/ld: util.cpp:(.text+0x1f34): undefined reference to `fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<char> >, char>::iterator fmt::v7::detail::vformat_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: util.cpp:(.text+0x1f6d): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/openroad.dir/build.make:335: src/openroad] Error 1
make[2]: Leaving directory '/home/ratul619/IIT_study_VM/OpenROAD/build'
make[1]: *** [CMakeFiles/Makefile2:1421: src/CMakeFiles/openroad.dir/all] Error 2
make[1]: Leaving directory '/home/ratul619/IIT_study_VM/OpenROAD/build'

### Expected Behavior

...

### Environment

```shell
I have already pulled the git path today , but still getting below message on running Env.sh

Please pull the latest changes and try again, if problem persists file a github issue with the re-producible test case.
```

### To Reproduce

...

ratul619, 2023-01-23:
git show --shortstat : 

commit c295b08a99aacb6147b9c51104627e78ac3859e3 (HEAD -> master, origin/master, origin/HEAD)
Merge: 59f179377 5dfec95a9
Author: Peter <46405338+gadfort@users.noreply.github.com>
Date:   Sat Jan 21 21:35:37 2023 -0500

    Merge pull request #2803 from gadfort/pdn-obs-filtering
    
    pdn: fix repeated creation of rtrees in Shape::cut

 5 files changed, 53 insertions(+), 25 deletions(-)

vvbandeira, 2023-01-23:
There were a couple of commits after that:
[image]

ratul619, 2023-01-23:
Yes , its running, once it completes will upload

ratul619, 2023-01-23:
[build.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/10482911/build.log)

ratul619, 2023-01-23:
@vvbandeira  - uploaded

vvbandeira, 2023-01-23:
thanks, but that looks like a clean build to me. are you able to run the binary?

ratul619, 2023-01-23:
what should be the next step ?

ratul619, 2023-01-23:
I see the binary build in below area - is this correct ?  

 ./build/src/openroad
OpenROAD v2.0-6562-gc295b08a9 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> 
openroad>

vvbandeira, 2023-01-23:
yes, that is the expected output.

ratul619, 2023-01-23:
Thanks , closing then

closed 2023-01-23 by ratul619