The-OpenROAD-Project/OpenROAD #2812: Compilation problem on Ubuntu 20

mattvenn, 2023-01-24:
### Describe the bug

I get up 97%

```[ 97%] Linking CXX executable openroad                                                                                                                                                        
cd /home/matt/work/asic-workshop/shuttle8/OpenROAD/build/src && /usr/bin/cmake -E cmake_link_script CMakeFiles/openroad.dir/link.txt --verbose=1                                              
/usr/bin/c++  -O3 -DNDEBUG  -Wl,--export-dynamic -rdynamic CMakeFiles/openroad.dir/Design.cc.o CMakeFiles/openroad.dir/Tech.cc.o CMakeFiles/openroad.dir/OpenRoad.cc.o CMakeFiles/openroad.dir/Main.cc.o  -o openroad  -Wl,-rpath,/opt/or-tools/lib:::::::::::::::::::: ifp/src/ifp.a pad/pad.a openroad_swig.a gpl/gpl.a dpl/dpl.a dpo/dpo.a fin/fin.a rsz/src/rsz.a ppl/ppl.a stt/stt.a dbSta/src/dbSta.a ../../src/sta/app/libOpenSTA.a odb/src/swig/tcl/odbtcl.a rcx/src/rcx.a rmp/src/rmp.a cts/src/cts.a grt/grt.a tap/src/tap.a gui/gui.a drt/drt.a dst/dst.a mpl/mpl.a psm/src/psm
.a ant/src/ant.a par/par.a utl/utl.a pdn/src/pdn.a ../third-party/abc/libabc.a /usr/lib/x86_64-linux-gnu/libtcl.so -pthread mpl2/mpl2.a /usr/lib/x86_64-linux-gnu/libtclreadline.so /usr/lib/x86_64-linux-gnu/libz.so _openroad_swig_py.a odb/src/swig/python/_odb_py.a ifp/src/_ifp_py.a utl/_utl_py.a ant/src/_ant_py.a grt/_grt_py.a gpl/_gpl_py.a dpl/_dpl_py.a mpl/_mpl_py.a ppl/_ppl_p
y.a tap/src/_tap_py.a cts/src/_cts_py.a drt/_drt_py.a dpo/_dp```

Then get 1000s of log lines ending in 

```_to<char>(fmt::v7::detail::buffer<char>&, fmt::v7::basic_string_view<char>, fmt::v7::basic_format_args<fmt::v7::basic_format_context<std::back_insert_iterator<fmt::v7::detail::buffer<fmt::v7::type_identity<char>::type> >, fmt::v7::type_identity<char>::type> >)'
/usr/bin/ld: util.cpp:(.text+0x1f6d): undefined reference to `spdlog::details::log_msg::log_msg(spdlog::source_loc, fmt::v7::basic_string_view<char>, spdlog::level::level_enum, fmt::v7::basic_string_view<char>)'
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/openroad.dir/build.make:325: src/openroad] Error 1
make[2]: Leaving directory '/home/matt/work/asic-workshop/shuttle8/OpenROAD/build'
make[1]: *** [CMakeFiles/Makefile2:1595: src/CMakeFiles/openroad.dir/all] Error 2
make[1]: Leaving directory '/home/matt/work/asic-workshop/shuttle8/OpenROAD/build'
make: *** [Makefile:133: all] Error 2
```

### Expected Behavior

To compile

### Environment

```shell
kernel: Linux 5.15.0-58-generic
os: Ubuntu 20.04.2 LTS (Focal Fossa)
cmake version 3.16.3
gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
clang version 10.0.0-4ubuntu1
```

### To Reproduce

n/a

vvbandeira, 2023-01-24:
@mattvenn could you run the command below and upload the build.log file?

```
./etc/Build.sh -clean 2>&1 | tee build.log
```

rovinski, 2023-01-24:
Isn't `build_openroad.log` generated by default now?

vvbandeira, 2023-01-24:
> Isn't `build_openroad.log` generated by default now?

It is for ORFS using build_openroad.sh

I have already asked @habibayassin to add this feature to `etc/Build.sh` script at OR as well.

rovinski, 2023-01-24:
Ah right I didn't notice which repo this was.

closed 2023-01-25 by mattvenn