The-OpenROAD-Project/OpenROAD #2818: Detailed router failure

oharboe, 2023-01-26:
### Describe the bug

Reproducible with this pull request:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/789

```
make DESIGN_CONFIG=designs/asap7/mock-very-large-array/config.mk
[deleted]
[INFO DRT-0195] Start 0th optimization iteration.
Failed to find a path between pin ces_5_0/io_outs_2[3] and source aps:
( 382 529 2 ) (Idx) / ( 18081 147420 ) (coords)
extBBox (xDim, yDim, zDim) = (602, 711, 10)
[ERROR DRT-0255] Maze Route cannot find path of net ces_5_0_io_outs_2\[3\] in worker of routeBox ( 15120 143640 ) ( 18900 147420 ).
Failed to find a path between pin  and source aps:
( 195 187 4 ) (Idx) / ( 91236 143650 ) (coords)
extBBox (xDim, yDim, zDim) = (602, 711, 10)
[ERROR DRT-0255] Maze Route cannot find path of net ces_5_3_io_ins_2\[6\] in worker of routeBox ( 90720 143640 ) ( 94500 147420 ).
Failed to find a path between pin  and source aps:
( 279 187 4 ) (Idx) / ( 114996 143650 ) (coords)
extBBox (xDim, yDim, zDim) = (602, 711, 10)
[ERROR DRT-0255] Maze Route cannot find path of net ces_5_4_io_ins_2\[6\] in worker of routeBox ( 113400 143640 ) ( 117180 147420 ).
Error: detail_route.tcl, 47 DRT-0255
Elapsed time: 0:38.18[h:]min:sec. CPU time: user 404.92 sys 71.07 (1246%). Peak memory: 13332436KB.
make: *** [results/asap7/mock-very-large-array/base/5_2_route.odb] Error 1
```

### Expected Behavior

It should work, or if there is something wrong with the design, a more explicit error message should be shown.

### Environment

```shell
Git commit: 53c6e6dfd60130d056eeda896d1d75b0a632f739
kernel: Linux 5.15.0-58-generic
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.24.2
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
Ubuntu clang version 14.0.0-1ubuntu1
```

### To Reproduce

Reproducible with this pull request:

https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/pull/789

```
make DESIGN_CONFIG=designs/asap7/mock-very-large-array/config.mk
```

### Relevant log output

```shell
make DESIGN_CONFIG=designs/asap7/mock-very-large-array/config.mk
[deleted]
[INFO DRT-0195] Start 0th optimization iteration.
Failed to find a path between pin ces_5_0/io_outs_2[3] and source aps:
( 382 529 2 ) (Idx) / ( 18081 147420 ) (coords)
extBBox (xDim, yDim, zDim) = (602, 711, 10)
[ERROR DRT-0255] Maze Route cannot find path of net ces_5_0_io_outs_2\[3\] in worker of routeBox ( 15120 143640 ) ( 18900 147420 ).
Failed to find a path between pin  and source aps:
( 195 187 4 ) (Idx) / ( 91236 143650 ) (coords)
extBBox (xDim, yDim, zDim) = (602, 711, 10)
[ERROR DRT-0255] Maze Route cannot find path of net ces_5_3_io_ins_2\[6\] in worker of routeBox ( 90720 143640 ) ( 94500 147420 ).
Failed to find a path between pin  and source aps:
( 279 187 4 ) (Idx) / ( 114996 143650 ) (coords)
extBBox (xDim, yDim, zDim) = (602, 711, 10)
[ERROR DRT-0255] Maze Route cannot find path of net ces_5_4_io_ins_2\[6\] in worker of routeBox ( 113400 143640 ) ( 117180 147420 ).
Error: detail_route.tcl, 47 DRT-0255
Elapsed time: 0:38.18[h:]min:sec. CPU time: user 404.92 sys 71.07 (1246%). Peak memory: 13332436KB.
make: *** [results/asap7/mock-very-large-array/base/5_2_route.odb] Error 1
```
```

maliberty, 2023-01-26:
Does this occur after the fix for https://github.com/The-OpenROAD-Project/OpenROAD/issues/2805 ?

oharboe, 2023-01-26:
> Does this occur after the fix for #2805 ?

Yes.

oharboe, 2023-01-27:
@maliberty Mea culpa. When triple checking with latest master, no longer happens.

closed 2023-01-27 by oharboe