The-OpenROAD-Project/OpenROAD #2827: drt: GC overestimates minimum cut space on V2

antonblanchard, 2023-01-31:
### Describe the bug

When running any of the ORFS ASAP7 designs, I notice quite a number of cut spacing violations. These eventually get resolved, but I went looking as to why we see so many.

I think GC is overestimating the minimum cut space distance for V2. It should be 18nm:

```
LAYER V2
  TYPE CUT ;
 SPACING 0.018 ;
 WIDTH 0.018 ;
END V2
```

But if I look at `FlexGCWorker::Impl::checkLef58CutSpacingTbl_helper`, it appears to be enforcing a minimum cut space
distance of 34nm, which might be the constraint for V3 (not V2). 

### Expected Behavior

GC should enforce the correct minimum cut spacing

### Environment

```shell
Git commit: 78b4fef8f509bb91d6a52cb7551cb6486e7c41b4
os: Fedora Linux 37 (Workstation Edition)
cmake version 3.25.2
gcc (GCC) 12.2.1 20221121 (Red Hat 12.2.1-4)
clang version 15.0.7 (Fedora 15.0.7-1.fc37)
```

### To Reproduce

Run asap7/gcd in ORFS

antonblanchard, 2023-01-31:
`FlexGCWorker::Impl::checkCutSpacing_main` looks to apply the cut constraints from the via layer above. Is that correct?

```
  if (layer->getLayerNum() + 2 < TOP_ROUTING_LAYER
      && layer->getLayerNum() + 2 < getTech()->getLayers().size()) {
    auto aboveLayer = getTech()->getLayer(layer->getLayerNum() + 2);
    if (aboveLayer->hasLef58SameMetalCutSpcTblConstraint()) {
      checkLef58CutSpacingTbl(
          rect, aboveLayer->getLef58SameMetalCutSpcTblConstraint());
    }
    if (aboveLayer->hasLef58SameNetCutSpcTblConstraint()) {
      checkLef58CutSpacingTbl(rect,
                              aboveLayer->getLef58SameNetCutSpcTblConstraint());
    }
    if (aboveLayer->hasLef58DiffNetCutSpcTblConstraint()) {
      checkLef58CutSpacingTbl(rect,
                              aboveLayer->getLef58DiffNetCutSpcTblConstraint());
    }
    if (aboveLayer->hasLef58SameNetInterCutSpcTblConstraint()) {
      checkLef58CutSpacingTbl(
          rect, aboveLayer->getLef58SameNetInterCutSpcTblConstraint());
    }
    if (aboveLayer->hasLef58SameMetalInterCutSpcTblConstraint()) {
      checkLef58CutSpacingTbl(
          rect, aboveLayer->getLef58SameMetalInterCutSpcTblConstraint());
    }
    if (aboveLayer->hasLef58DefaultInterCutSpcTblConstraint()) {
      checkLef58CutSpacingTbl(
          rect, aboveLayer->getLef58DefaultInterCutSpcTblConstraint());
    }
```

antonblanchard, 2023-01-31:
Was added in commit b5f323514d1b3429934ff4f4fe9abca2e3a9ca50 ("drt: fix InterCutSpcTbl gc check")

maliberty, 2023-01-31:
You can have cut spacing constraints between adjacent cut layers.  Are you looking at same layer or diff layer spacing?

antonblanchard, 2023-01-31:
> You can have cut spacing constraints between adjacent cut layers. Are you looking at same layer or diff layer spacing?

In that snippet above, we are applying rules with and without `Inter`, so it makes me think we are applying same layer and different layer spacing.

osamahammad21, 2023-02-07:
That does look weird. I remember adding this to solve an issue in intel16 having many calibre DRCs. I am gonna remove the checks for same layer.
@maliberty Is there a way to run calibre on a branch other than master?

maliberty, 2023-02-07:
I don't think so but @vvbandeira can confirm/help.  You could run it yourself at UMich if you still have access.

osamahammad21, 2023-02-07:
@maliberty I have access to calibre but not intel16

maliberty, 2023-02-07:
I can do some testing for you if needed

osamahammad21, 2023-02-07:
@maliberty Thanks it would be very helpful to test/confirm before merging. You could run openroad from [my PR](https://github.com/The-OpenROAD-Project-private/OpenROAD/pull/1590) on intel16 designs and check if the calibre drcs change.

maliberty, 2023-02-10:
Osama is adding support for another rule in order to get this working properly (fixing the current overestimates revealed the issue).  It will take a bit to get that fully tested and in.

antonblanchard, 2023-02-10:
Thanks @maliberty

antonblanchard, 2023-04-13:
This is fixed in https://github.com/The-OpenROAD-Project/OpenROAD/pull/3168. Thanks @osamahammad21.

closed 2023-04-13 by antonblanchard