The-OpenROAD-Project/OpenROAD #2831: Installing openroad on Linux embedded 5.15.0-58-generic #64-Ubuntu

ripudamank2, 2023-01-31:
### Describe the bug

/home/embedded/Documents/OpenROAD/build/src/sta/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx: In function ‘int _wrap_read_spef_cmd(ClientData, Tcl_Interp*, int, Tcl_Obj* const*)’:
/home/embedded/Documents/OpenROAD/build/src/sta/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx:38297:120: error: ‘SWIG_STD_MOVE’ was not declared in this scope
38297 |       result = (bool)read_spef_cmd((char const *)arg1,arg2,(Corner const *)arg3,(MinMaxAll const *)arg4,arg5,arg6,arg7,SWIG_STD_MOVE(arg8),arg9,arg10);
      |                                                                                                                        ^~~~~~~~~~~~~
gmake[2]: *** [src/sta/CMakeFiles/sta_swig.dir/build.make:79: src/sta/CMakeFiles/sta_swig.dir/CMakeFiles/sta_swig.dir/StaAppTCL_wrap.cxx.o] Error 1
**gmake[2]: Leaving directory '/home/embedded/Documents/OpenROAD/build'
gmake[1]: *** [CMakeFiles/Makefile2:2451: src/sta/CMakeFiles/sta_swig.dir/all] Error 2
gmake[2]: Leaving directory '/home/embedded/Documents/OpenROAD/build'
gmake[1]: *** [CMakeFiles/Makefile2:4137: src/utl/CMakeFiles/utl.dir/all] Error 2
gmake[1]: Leaving directory '/home/embedded/Documents/OpenROAD/build'
gmake: *** [Makefile:139: all] Error 2**

real	0m2,966s
user	0m11,493s
sys	0m0,997s

i am getting this error , can you please tell how to fix it ?

### Expected Behavior

it should had install the openroad

### Environment

```shell
Ubunut
```

### To Reproduce

git clone --recursive https://github.com/The-OpenROAD-Project/OpenROAD.git
cd OpenROAD
./etc/DependencyInstaller.sh
./etc/Build.sh

vvbandeira, 2023-01-31:
Please provide the output of `etc/Env.sh`. Also, could you try `etc/Build.sh -clean`?

ripudamank2, 2023-01-31:
yes i tried the command /etc/build.sh -clean and it worked . 

Can you please tell how to open GUI ?

vvbandeira, 2023-01-31:
`build/src/openroad -gui`

Closing since the initial issue has been resolved.

closed 2023-01-31 by ripudamank2