The-OpenROAD-Project/OpenROAD #2832: Add "Goto Position" to the GUI

maliberty, 2023-01-31:
### Description

Have an equivalent to KLayout's "Goto Position"
[image]

When the dialog appears the default entry appears to be based on the current view.

From a discussion w/ @antonblanchard

maliberty, 2023-01-31:
I think the KLayout dialog should not be directly copied as these should be three distinct entry widgets.  DBU or microns should be supported as in the rest of the GUI.

gadfort, 2023-01-31:
@maliberty It should be relatively easy to implement. But not 100% on the timeline.

antonblanchard, 2023-01-31:
> I think the KLayout dialog should not be directly copied as these should be three distinct entry widgets. DBU or microns should be supported as in the rest of the GUI.

Sounds good. There are a few other things that annoy me about the klayout goto, so we don't have to copy it completely.

eg if you just put in x,y without a window size it doesn't zoom and just places crosshairs on the point. That requires you to manually zoom to the point. If, on the other hand, you reopen the goto dialog to add the window size, the x,y settings have been reset.

gadfort, 2023-02-01:
@antonblanchard are you suggesting that the x,y be prepopulated with the current center?

maliberty, 2023-02-01:
It seems like the dialog is pre populated with the current x, y, s

maliberty, 2023-08-22:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2974

closed 2023-08-22 by maliberty