The-OpenROAD-Project/OpenROAD #2833: Extremely large `ar` call often crashes

donn, 2023-01-31:
### Describe the bug

I've run into this far too many times on my computer: the libabc build has an outrageously long invocation in its build that just straight up crashes in most environments for me now- even more so with multiple threads. There's also a `ranlib` right after that could be the culprit.

```sh
ar qc libabc.a CMakeFiles/libabc.dir/src/base/abc/abcAig.c.o CMakeFiles/libabc.dir/src/base/abc/abcBarBuf.c.o CMakeFiles/libabc.dir/src/base/abc/abcBlifMv.c.o CMakeFiles/libabc.dir/src/base/abc/abcCheck.c.o CMakeFiles/libabc.dir/src/base/abc/abcDfs.c.o CMakeFiles/libabc.dir/src/base/abc/abcFanio.c.o CMakeFiles/libabc.dir/src/base/abc/abcFanOrder.c.o CMakeFiles/libabc.dir/src/base/abc/abcFunc.c.o CMakeFiles/libabc.dir/src/base/abc/abcHie.c.o CMakeFiles/libabc.dir/src/base/abc/abcHieCec.c.o CMakeFiles/libabc.dir/src/base/abc/abcHieGia.c.o CMakeFiles/libabc.dir/src/base/abc/abcHieNew.c.o CMakeFiles/libabc.dir/src/base/abc/abcLatch.c.o CMakeFiles/libabc.dir/src/base/abc/abcLib.c.o CMakeFiles/libabc.dir/src/base/abc/abcMinBase.c.o CMakeFiles/libabc.dir/src/base/abc/abcNames.c.o CMakeFiles/libabc.dir/src/base/abc/abcNetlist.c.o CMakeFiles/libabc.dir/src/base/abc/abcNtk.c.o CMakeFiles/libabc.dir/src/base/abc/abcObj.c.o CMakeFiles/libabc.dir/src/base/abc/abcRefs.c.o CMakeFiles/libabc.dir/src/base/abc/abcShow.c.o CMakeFiles/libabc.dir/src/base/abc/abcSop.c.o CMakeFiles/libabc.dir/src/base/abc/abcUtil.c.o CMakeFiles/libabc.dir/src/base/abci/abc.c.o CMakeFiles/libabc.dir/src/base/abci/abcAttach.c.o CMakeFiles/libabc.dir/src/base/abci/abcAuto.c.o CMakeFiles/libabc.dir/src/base/abci/abcBalance.c.o CMakeFiles/libabc.dir/src/base/abci/abcBidec.c.o CMakeFiles/libabc.dir/src/base/abci/abcBm.c.o CMakeFiles/libabc.dir/src/base/abci/abcBmc.c.o CMakeFiles/libabc.dir/src/base/abci/abcCas.c.o CMakeFiles/libabc.dir/src/base/abci/abcCascade.c.o CMakeFiles/libabc.dir/src/base/abci/abcCollapse.c.o CMakeFiles/libabc.dir/src/base/abci/abcCut.c.o CMakeFiles/libabc.dir/src/base/abci/abcDar.c.o CMakeFiles/libabc.dir/src/base/abci/abcDebug.c.o CMakeFiles/libabc.dir/src/base/abci/abcDec.c.o CMakeFiles/libabc.dir/src/base/abci/abcDetect.c.o CMakeFiles/libabc.dir/src/base/abci/abcDress.c.o CMakeFiles/libabc.dir/src/base/abci/abcDress2.c.o CMakeFiles/libabc.dir/src/base/abci/abcDress3.c.o CMakeFiles/libabc.dir/src/base/abci/abcDsd.c.o CMakeFiles/libabc.dir/src/base/abci/abcEco.c.o CMakeFiles/libabc.dir/src/base/abci/abcExact.c.o CMakeFiles/libabc.dir/src/base/abci/abcExtract.c.o CMakeFiles/libabc.dir/src/base/abci/abcFraig.c.o CMakeFiles/libabc.dir/src/base/abci/abcFx.c.o CMakeFiles/libabc.dir/src/base/abci/abcFxu.c.o CMakeFiles/libabc.dir/src/base/abci/abcGen.c.o CMakeFiles/libabc.dir/src/base/abci/abcHaig.c.o CMakeFiles/libabc.dir/src/base/abci/abcIf.c.o CMakeFiles/libabc.dir/src/base/abci/abcIfif.c.o CMakeFiles/libabc.dir/src/base/abci/abcIfMux.c.o CMakeFiles/libabc.dir/src/base/abci/abcIvy.c.o CMakeFiles/libabc.dir/src/base/abci/abcLog.c.o CMakeFiles/libabc.dir/src/base/abci/abcLut.c.o CMakeFiles/libabc.dir/src/base/abci/abcLutmin.c.o CMakeFiles/libabc.dir/src/base/abci/abcMap.c.o CMakeFiles/libabc.dir/src/base/abci/abcMerge.c.o CMakeFiles/libabc.dir/src/base/abci/abcMfs.c.o CMakeFiles/libabc.dir/src/base/abci/abcMini.c.o CMakeFiles/libabc.dir/src/base/abci/abcMiter.c.o CMakeFiles/libabc.dir/src/base/abci/abcMulti.c.o CMakeFiles/libabc.dir/src/base/abci/abcNtbdd.c.o CMakeFiles/libabc.dir/src/base/abci/abcNpn.c.o CMakeFiles/libabc.dir/src/base/abci/abcNpnSave.c.o CMakeFiles/libabc.dir/src/base/abci/abcOdc.c.o CMakeFiles/libabc.dir/src/base/abci/abcOrder.c.o CMakeFiles/libabc.dir/src/base/abci/abcPart.c.o CMakeFiles/libabc.dir/src/base/abci/abcPrint.c.o CMakeFiles/libabc.dir/src/base/abci/abcProve.c.o CMakeFiles/libabc.dir/src/base/abci/abcQbf.c.o CMakeFiles/libabc.dir/src/base/abci/abcQuant.c.o CMakeFiles/libabc.dir/src/base/abci/abcRec3.c.o CMakeFiles/libabc.dir/src/base/abci/abcReconv.c.o CMakeFiles/libabc.dir/src/base/abci/abcReach.c.o CMakeFiles/libabc.dir/src/base/abci/abcRefactor.c.o CMakeFiles/libabc.dir/src/base/abci/abcRenode.c.o CMakeFiles/libabc.dir/src/base/abci/abcReorder.c.o CMakeFiles/libabc.dir/src/base/abci/abcRestruct.c.o CMakeFiles/libabc.dir/src/base/abci/abcResub.c.o CMakeFiles/libabc.dir/src/base/abci/abcRewrite.c.o CMakeFiles/libabc.dir/src/base/abci/abcRpo.c.o CMakeFiles/libabc.dir/src/base/abci/abcRr.c.o CMakeFiles/libabc.dir/src/base/abci/abcRunGen.c.o CMakeFiles/libabc.dir/src/base/abci/abcSat.c.o CMakeFiles/libabc.dir/src/base/abci/abcSaucy.c.o CMakeFiles/libabc.dir/src/base/abci/abcScorr.c.o CMakeFiles/libabc.dir/src/base/abci/abcSense.c.o CMakeFiles/libabc.dir/src/base/abci/abcSpeedup.c.o CMakeFiles/libabc.dir/src/base/abci/abcStrash.c.o CMakeFiles/libabc.dir/src/base/abci/abcSweep.c.o CMakeFiles/libabc.dir/src/base/abci/abcSymm.c.o CMakeFiles/libabc.dir/src/base/abci/abcTim.c.o CMakeFiles/libabc.dir/src/base/abci/abcTiming.c.o CMakeFiles/libabc.dir/src/base/abci/abcUnate.c.o CMakeFiles/libabc.dir/src/base/abci/abcUnreach.c.o CMakeFiles/libabc.dir/src/base/abci/abcVerify.c.o CMakeFiles/libabc.dir/src/base/abci/abcXsim.c.o CMakeFiles/libabc.dir/src/base/cmd/cmd.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdAlias.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdApi.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdAuto.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdFlag.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdHist.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdLoad.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdPlugin.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdStarter.c.o CMakeFiles/libabc.dir/src/base/cmd/cmdUtils.c.o CMakeFiles/libabc.dir/src/base/io/io.c.o CMakeFiles/libabc.dir/src/base/io/ioJson.c.o CMakeFiles/libabc.dir/src/base/io/ioReadAiger.c.o CMakeFiles/libabc.dir/src/base/io/ioReadBaf.c.o CMakeFiles/libabc.dir/src/base/io/ioReadBblif.c.o CMakeFiles/libabc.dir/src/base/io/ioReadBench.c.o CMakeFiles/libabc.dir/src/base/io/ioReadBlif.c.o CMakeFiles/libabc.dir/src/base/io/ioReadBlifAig.c.o CMakeFiles/libabc.dir/src/base/io/ioReadBlifMv.c.o CMakeFiles/libabc.dir/src/base/io/ioReadDsd.c.o CMakeFiles/libabc.dir/src/base/io/ioReadEdif.c.o CMakeFiles/libabc.dir/src/base/io/ioReadEqn.c.o CMakeFiles/libabc.dir/src/base/io/ioReadPla.c.o CMakeFiles/libabc.dir/src/base/io/ioReadPlaMo.c.o CMakeFiles/libabc.dir/src/base/io/ioReadVerilog.c.o CMakeFiles/libabc.dir/src/base/io/ioUtil.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteAiger.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteBaf.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteBblif.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteBench.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteBlif.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteBlifMv.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteBook.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteCnf.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteDot.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteEqn.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteGml.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteList.c.o CMakeFiles/libabc.dir/src/base/io/ioWritePla.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteVerilog.c.o CMakeFiles/libabc.dir/src/base/io/ioWriteSmv.c.o CMakeFiles/libabc.dir/src/base/main/mainFrame.c.o CMakeFiles/libabc.dir/src/base/main/mainInit.c.o CMakeFiles/libabc.dir/src/base/main/mainLib.c.o CMakeFiles/libabc.dir/src/base/main/mainReal.c.o CMakeFiles/libabc.dir/src/base/main/libSupport.c.o CMakeFiles/libabc.dir/src/base/main/mainUtils.c.o CMakeFiles/libabc.dir/src/base/exor/exor.c.o CMakeFiles/libabc.dir/src/base/exor/exorBits.c.o CMakeFiles/libabc.dir/src/base/exor/exorCubes.c.o CMakeFiles/libabc.dir/src/base/exor/exorLink.c.o CMakeFiles/libabc.dir/src/base/exor/exorList.c.o CMakeFiles/libabc.dir/src/base/exor/exorUtil.c.o CMakeFiles/libabc.dir/src/base/ver/verCore.c.o CMakeFiles/libabc.dir/src/base/ver/verFormula.c.o CMakeFiles/libabc.dir/src/base/ver/verParse.c.o CMakeFiles/libabc.dir/src/base/ver/verStream.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcAbs.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcAbs2.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcAbc.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcPth.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcBlast.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcCom.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcGraft.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcJson.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcMem.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcNdr.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcNtk.c.o CMakeFiles/libabc.dir/src/base/wlc/wlcReadSmt.c.o CMakeFiles/libabc.dir/src/base/wlc/wlc [... cut ...] llb4Image.c.o CMakeFiles/libabc.dir/src/bdd/llb/llb4Nonlin.c.o CMakeFiles/libabc.dir/src/bdd/llb/llb4Sweep.c.o
```

### Expected Behavior

Not a crash. I don't know how this could be alleviated: could the build be more... hierarchical?

### Environment

```shell
Git commit: a19ae00780d477f8b1b59979277f375277f17ada
kernel: Linux 5.15.89
os: NixOS 22.11 (Raccoon)
cmake version 3.24.3
gcc (GCC) 10.4.0
clang version 11.1.0
```

### To Reproduce

Just try compiling OpenROAD with ABC with 16+ threads.

QuantamHD, 2023-01-31:
I run with 100 threads on my machine and haven't seen a crash related to this. Is it maybe a low memory issue?

donn, 2023-01-31:
I have ~20 GB free, so I don't think that's it either.

maliberty, 2023-01-31:
I've never seen this in daily -j45 builds.  Perhaps low tmp or swap space?

donn, 2023-01-31:
Nope, I have plenty of both too...

Curiously, using Clang and exactly 8 threads works fine. What version of GCC do you recommend?

maliberty, 2023-01-31:
I'm using "gcc version 8.3.1"

QuantamHD, 2023-01-31:
@donn Could you try to collect a core dump for the failing program.

https://jvns.ca/blog/2018/04/28/debugging-a-segfault-on-linux/

QuantamHD, 2023-02-02:
Since the problem exists outside our tree I think it's technically out of scope for the project. The fix would need to come from upstream abc.

donn, 2023-02-02:
OpenROAD uses a fork of ABC, for which the build can be made a bit more hierarchical.

maliberty, 2023-03-05:
@rovinski have you run into this problem with abc or just other crashes in general?  

Right now I don't have any way to reproduce this and no OR developer has ever reported seeing it.  Its never happened in the OR CI.

@donn how can I reproduce this issue before trying to fix it?

rovinski, 2023-03-05:
Not sure; I haven't paid attention to when it happens specifically. It's a random occurrence for me.

Maybe try running the build script with `threads` > `$(nproc)`?

closed 2023-08-24 by donn