The-OpenROAD-Project/OpenROAD #2838: feature request: open odb file from the command line

xobs, 2023-02-02:
### Description

The addition of the "Open Database" feature has been very useful, because it makes it much easier to discover the ability to open `.odb` files.

It would be very nice to also add the ability to open a database from the command line. This makes it easy to open "the most recent good file" from an openlane flow to inspect what might have gone wrong.

### Suggested Solution

Either add a command line argument `--open`, or add some magic to detect whether the specified file is a tcl script or an odb file.

Alternately, adding an `--eval` command line argument would also be acceptable.

### Additional Context

Currently I have this implemented using a FIFO in a script called `show-openroad.sh`:

```
#!/bin/sh
if [ ! -e "$1" ]
then
        echo "File $1 doesn't exist!"
        exit 1
fi
rm -f /tmp/pipe
mkfifo /tmp/pipe
(echo "read_db $1" > /tmp/pipe &)
openroad -gui /tmp/pipe
rm -f /tmp/pipe
```

But it would be nice if this wasn't required

maliberty, 2023-02-10:
@proppy do you have time for this?

proppy, 2023-02-10:
Related https://github.com/The-OpenROAD-Project/OpenROAD/pull/2762

proppy, 2023-02-10:
@xobs curious if the following would work.

```
openroad -gui <(echo read_db path/to/layout)
```

xobs, 2023-02-10:
That does work, yes! I had trouble remembering the bash-ism for "pipe the output of this subshell into stdin of a new shell".

maliberty, 2023-02-10:
Still I think it would be nice to have a flag for this common use case

maliberty, 2025-09-22:
I think `--db <odbfile>` is clear and also allows for a tcl file to be specified as usual.

PR #8428 (2025-10-02): Console --db option
## Summary
This PR adds a new CLI option `-db <file_name.odb>` that directly loads an design database at program startup.

## Motivation
Requested in #2838 to simplify testing workflows by avoiding the need to run `read_db` manually.

## Changes
- Added `-db <file.odb>` command-line option.
- In Terminal mode: executes `read_db` immediately.
- In GUI mode: queues `read_db` with `addRestoreStateCommand`.
- Updated usage message in `Main.cc`.

## Testing
- Verified `./openroad -gui -db design.odb` loads the design properly.
- Added `open_db.tcl` and `open_db_invalid.tcl` regr [... cut]