The-OpenROAD-Project/OpenROAD #2856: Write markers for pdn issues from psm

maliberty, 2023-02-06:
### Description

When psm fails on power grid connectivity issues with

[WARNING PSM-0039] Unconnected instance XX at location (X.XXXum, X.XXXum) layer: X.

it would be good to write an rpt file with markers that can be loaded into the drc viewer.

@VidyaChhabria do you have time for this?

VidyaChhabria, 2023-02-06:
@maliberty probably not in the near future (as it would go at the bottom of my list). Is it urgent?

maliberty, 2023-02-06:
@VidyaChhabria I figured that would be the case but thought it worth trying.  I can take care of it.

maliberty, 2025-08-26:
@gadfort I think this is resolved in your psm rewritting, correct?

gadfort, 2025-08-26:
@maliberty yes