The-OpenROAD-Project/OpenROAD #2861: Add clock trunk/leaf flag to odb

antonblanchard, 2023-02-06:
### Description

There are a few places where we want to know if a clock net is a trunk or leaf (grt, drt and gui). We should add it to odb.

Right now `setSigType(odbdbSigType::CLOCK)` is set in two places (cts and grt), so we could just do it in cts, together with the code to detect trunk vs leaf nets. That would break the def flow, so should we populate the clock fields when we read in a def too?

maliberty, 2023-02-06:
I suspect it is done in grt for test cases where we start the flow at grt rather than run a full flow (eg DEFs from contests or other tools).  Presumably it is required because the DEF itself didn't have the clock nets already marked, otherwise it would be a no-op.  Drt would likely need the same as would any other tool that cares.

LEF/DEF alone doesn't tell you enough to figure out what is a leaf/trunk as you need to know about the cell types from Liberty (eg is this a buffer/inverter/ff/etc).    Even to know what is a clock is best done by sta as gated/generated clocks can get complicated.  I think we will have to have a utility to compute this info that is called at various points in the flow.

Note that some tools use a thre-way split: leaf, trunk, and top.  Top is any clock net with transitive leaf fanout exceeding some limit.

rovinski, 2023-02-06:
I've also heard of "trunk, branch, leaf" 🙂

antonblanchard, 2023-02-09:
> LEF/DEF alone doesn't tell you enough to figure out what is a leaf/trunk as you need to know about the cell types from Liberty (eg is this a buffer/inverter/ff/etc). Even to know what is a clock is [...]

Does it make sense for that to go in dbNetwork?

At the moment drt doesn't need to access liberty files (or dbSta and dbNetwork), so this would be a change in behaviour.

maliberty, 2023-02-21:
dbSta could have a command/API to do this.  That seems as good a place as any.