The-OpenROAD-Project/OpenROAD #2864: Write out .rpt files that can be loaded into DRC viewer when logging warnings

oharboe, 2023-02-07:
### Description

It would be helpful to have a .rpt file that I can load to view the problems in the warnings below:

Tasks:

- [ ] hunt other warning messages that refer to a location in the design and add tasks to this github issue to address them
- [ ] write out a .rpt file for the specific warning below

```
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0018] Y direction bump pitch is not specified, defaulting to 140um.
[WARNING PSM-0065] VSRC location not specified, using default checkerboard pattern with one VDD every size bumps in x-direction and one in two bumps in the y-direction
[INFO PSM-0076] Setting metal node density to be standard cell height times 5.
[WARNING PSM-0030] VSRC location at (680.567um, 370.300um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (691.854um, 375.327um).
[WARNING PSM-0030] VSRC location at (400.567um, 650.300um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (394.854um, 639.327um).
[WARNING PSM-0030] VSRC location at (680.567um, 790.300um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (679.974um, 771.327um).
[WARNING PSM-0030] VSRC location at (960.567um, 930.300um) and size 10.000um, is not located on an existing power stripe node. Moving to closest node at (988.794um, 927.327um).
[INFO PSM-0031] Number of PDN nodes on net VDD = 1448763.
[INFO PSM-0064] Number of voltage sources = 10.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (988.854um, 951.183um), layer: 5.
[WARNING PSM-0039] Unconnected instance r3c3 at location (988.854um, 951.183um) layer: 5.
[WARNING PSM-0038] Unconnected PDN node on net VDD at location (988.854um, 963.471um), layer: 5.
[WARNING PSM-0039] Unconnected instance r3c3 at location (988.854um, 963.471um) layer: 5.
[ERROR PSM-0078] IR drop setup failed.  Analysis can't proceed.
Error: final_report.tcl, 58 PSM-0078
```

### Suggested Solution

Write out a .rpt file when there are warnings in the log

QuantamHD, 2023-02-07:
That's a very cool idea. Having metadata associated with a log line is a really interesting concept

oharboe, 2023-02-07:
> That's a very cool idea. Having metadata associated with a log line is a really interesting concept

A log viewer in the GUI w/clicking on lines to see what they relate to in the design?

maliberty, 2023-02-07:
This is somewhat of a duplicate of https://github.com/The-OpenROAD-Project/OpenROAD/issues/2856 as it relates to psm.

In general many warning have nothing to show in the layout, e.g.
```
[WARNING PSM-0017] X direction bump pitch is not specified, defaulting to 140um.
```

I think we could just have a warnings.rpt that is written in addition to the log where there is something to show.  Parsing the log file seems fraught with trouble.  

Even this will be problematic as the layout might be changed after the warning is issued and make the marker useless/misleading.  The warnings from psm work because it is an analysis tool and doesn't modify the layout.

oharboe, 2023-02-07:
@maliberty Trying to navigate manually to a warning is straining on the eyes... A warning.rpt might have multiple sections, like a DRC file? That would make it clear what stage some warnings belong to and there's a better chance of looking at the relevant .odb file + warnings.

maliberty, 2023-03-23:
The OR piece of this https://github.com/The-OpenROAD-Project-private/OpenROAD/pull/1641

oharboe, 2023-03-24:
> The OR piece of this https://github.com/The-OpenROAD-Project-private/OpenROAD/pull/1641

FYI, I dont have access to that private repository, which is fine. Looking forward to trying the feature when it is done.

maliberty, 2023-03-24:
> > The OR piece of this [The-OpenROAD-Project-private/OpenROAD#1641](https://github.com/The-OpenROAD-Project-private/OpenROAD/pull/1641)
> 
> FYI, I dont have access to that private repository, which is fine. Looking forward to trying the feature when it is done.

Same as https://github.com/The-OpenROAD-Project/OpenROAD/pull/3074

gadfort, 2024-03-26:
The error file is already in the rpt format and can be loaded into the DRC viewer, so this should be complete.

closed 2024-03-26 by maliberty