The-OpenROAD-Project/OpenROAD #2872: read_spef returns many "net <net-name> not found" errors

nayiri-k, 2023-02-08:
### Describe the bug

After running `extract_parasitics` on my design, I run `write_spef` to create a SPEF file for the design. But subsequently reading the same SPEF file with `read_spef` results in many "net <net-name> not found" errors. 

I then run `write_verilog`, and am able to find all of the "not found" nets in the Verilog netlist.

One observation: all of the nets listed as "not found" in the tool output have a leading "\\" character, that is not found in the SPEF file but is found in the Verilog netlist (this is consistent with how Innovus writes out both files). However, the `read_spef` error message limit was reached before all "not found" nets were printed, so maybe the remaining nets without the leading "\\" character were simply not printed.

### Expected Behavior

There should be no error messages after the `read_spef` command.

### Environment

```shell
git commit: unknown, this is not a git repository.
(Please make sure that you have the latest code changes and add the commit hash in the description)
kernel: Linux 3.10.0-1160.80.1.el7.x86_64
os: Red Hat Enterprise Linux Server 7.9 (Maipo)
cmake version 3.22.1
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
clang version 15.0.6 (https://github.com/conda-forge/clangdev-feedstock b6cf7124da893bcb57ab1f38c768ea825fabbdb1)

**NOTE: this OpenROAD install is located on a server where I don't have sudo permissions, so the git version may not have been located due to this
Running 'git log' returns:
commit 5144431cb9ad3a71208cf0a810a0674cfeced15b
Merge: f86efab 23abb26
Author: Peter <46405338+arlpetergadfort@users.noreply.github.com>
Date:   Fri Oct 28 10:56:28 2022 -0400

    Merge pull request #2430 from The-OpenROAD-Project-staging/pdn-clang
    
    pdn: clang format
```

### To Reproduce

Download tar.gz archive from here: [
spef-nets-not-found.tar.gz](https://drive.google.com/file/d/1Q51AMnaMGt8D77ovyubmAosrj4WD3j4L/view?usp=sharing)
(file size is very large because I added the pre-extraction database to speed things up, otherwise run would take several hours)

tar -xf spef-nets-not-found.tar.gz
cd spef-nets-not-found
./runme.sh

### Relevant log output

```shell
(hammer) write_spef ChipTop.par.spef
[INFO RCX-0016] Writing SPEF ...
[INFO RCX-0042] 100000 nets finished
[INFO RCX-0443] 148462 nets finished
[INFO RCX-0017] Finished writing SPEF ...

(hammer) read_spef -corner setup ChipTop.par.spef
[WARNING STA-0174] ChipTop.par.spef line 1096973, net system/prci_ctrl_domain/tileResetSetter/_out_out_bits_data_T_4 not found.
....
[WARNING STA-0174] message limit reached, this message will no longer print

(hammer) write_verilog -include_pwr_gnd ChipTop.lvs.v

Within ChipTop.lvs.v, we can see this net:
line 8012:  wire \system/prci_ctrl_domain/tileResetSetter/_out_out_bits_data_T_4 ;
```

rovinski, 2023-02-09:
The first recommendation is definitely to try and check with a more recent version of OpenROAD.
This issue may have been resolved by #2628.

nayiri-k, 2023-02-09:
No longer an issue with a new version of OpenROAD, closing this issue!

closed 2023-02-09 by nayiri-k