The-OpenROAD-Project/OpenROAD #2881: Segfault in STA

xobs, 2023-02-10:
### Describe the bug

When attempting to route DFFRAM for GF180MCU, openroad segfaults during "Multi-Corner Static Timing Analysis at the min process corner"

### Expected Behavior

openroad should never segfault

### OpenROAD Environment

```shell
[WARNING] Your current OpenROAD version is outdated.
It is recommened to pull the latest changes.
If problem persists, file a github issue with the re-producible test case.
kernel: Linux 5.15.79.1-microsoft-standard-WSL2
os: Ubuntu 22.04.1 LTS (Jammy Jellyfish)
cmake version 3.24.2
CMake Error: The source directory "/opt/Si/src/OpenROAD/etc" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```

### OpenLane Environment

```shell
Traceback (most recent call last):
  File "./share/openlane/env.py", line 265, in <module>
    main()
  File "./share/openlane/env.py", line 261, in main
    commands[args[0]]()
  File "./share/openlane/env.py", line 168, in issue_survey
    % get_tag()
  File "/si/conda/share/openlane/dependencies/get_tag.py", line 67, in get_tag
    raise e
  File "/si/conda/share/openlane/dependencies/get_tag.py", line 48, in get_tag
    f"Cannot determine git branch. Please specify OPENLANE_IMAGE_NAME manually.\nFull output: {branch_name_data.stderr.decode('utf8').strip()}"
dependencies.get_tag.NoGitException: Cannot determine git branch. Please specify OPENLANE_IMAGE_NAME manually.
Full output: fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
```

### To Reproduce

1. Download the latest `mixed-signal-0` for GF180MCU from https://github.com/proppy/conda-eda/releases/latest
2. Install it by running the program
3. Activate it by running `. ${install_path}/bin/activate`
4. Download the latest DFFRAM
5. Run:

```
./dffram.py \
    --using-local-openlane /si/conda/share/openlane \
    --pdk-root=$PDK_ROOT \
    --building-blocks=$PDK:$SCL:$NAME \
    --routing-threads 20 \
    --halo 100 \
    -s 32x32 \
    -v 2R1W \
```

### Relevant log output

```shell
[STEP 6]
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: build/32x32_2R1W/openlane/runs/RUN_2023.02.10_10.54.18/logs/signoff/6-rcx_mcsta.min.log)...
[ERROR]: during executing openroad script /si/conda/share/openlane/scripts/openroad/sta_multi_corner.tcl
[ERROR]: Log: build/32x32_2R1W/openlane/runs/RUN_2023.02.10_10.54.18/logs/signoff/6-rcx_mcsta.min.log
[ERROR]: Last 10 lines:
11# Tcl_EvalEx in /si/conda/bin/../lib/libtcl8.6.so
12# Tcl_Eval in /si/conda/bin/../lib/libtcl8.6.so
13# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
14# ord::tclAppInit(Tcl_Interp*) in openroad
15# Tcl_MainEx in /si/conda/bin/../lib/libtcl8.6.so
16# main in openroad
17# 0x00007F2BC245ED90 in /lib/x86_64-linux-gnu/libc.so.6
18# __libc_start_main in /lib/x86_64-linux-gnu/libc.so.6
19# 0x0000000000B0CC69 in openroad
child killed: segmentation violation
```

### Additional Context

Here is the issue_reproducible from the run.

[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10703302/issue_reproducible.tar.gz)

xobs, 2023-02-10:
This is using openroad 21da7da7cce402d6f0984583cf89f611870041cc

xobs, 2023-02-10:
As suggested by @proppy , I was able to successfully lay out DFFRAM by commenting out the `run_parasitics_sta` step in `scripts/openlane/interactive.tcl`

maliberty, 2023-02-10:
I can reproduce the error with the specific commit given:
```
0# 0x0000000000CCF487 in openroad
 1# 0x00007FE4911C5400 in /lib64/libc.so.6
 2# sta::MakeTimingModel::makeGateModelTable(sta::Pin const*, float, sta::RiseFall*) in openroad
 3# sta::MakeTimingModel::makeInputOutputTimingArcs(sta::Pin const*, std::map<sta::Pin const*, sta::OutputDelays, std::less<sta::Pin const*>, std::allocator<std::pair<sta::Pin const* const, sta::OutputDelays> > >&) in openroad
 4# sta::MakeTimingModel::findTimingFromInputs() in openroad
 5# sta::MakeTimingModel::makeTimingModel(char const*, char const*, char const*) in openroad
 6# sta::Sta::writeTimingModel(char const*, char const*, char const*, sta::Corner const*) in openroad
```

jjcherry56, 2023-02-14:
fixed in opensta commit c94ee13 (but integration into openroad will take a while because the version is is using is so bar behind)

QuantamHD, 2023-03-19:
@xobs I believe this is fixed in the latest version

maliberty, 2023-03-19:
Please reopen if the problem persists

closed 2023-03-19 by maliberty