The-OpenROAD-Project/OpenROAD #2892: Drc spacing violation with macro obstruction

kareefardi, 2023-02-13:
### Describe the bug

Met2 spacing violation at approx (1180, 996). Observed spacing is `0.266`. According to spacing table provided in tech lef spacing should be larger than 0.28.

### Expected Behavior

No drc violations.

### OpenROAD Environment

```shell
Git commit: 79e6a99bcbcf15a18c95cd064ef202660f73cb89
kernel: Linux 5.14.0-1057-oem
os: Ubuntu 20.04.3 LTS (Focal Fossa)
cmake version 3.14.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- OpenROAD version: v2.0-6818-g79e6a99bc
-- System name: Linux
-- Compiler: GNU 9.4.0
-- Build type: RELEASE
-- Install prefix: /usr/local
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test C_COMPILER_SUPPORTS__-Wall
-- Performing Test C_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall
-- Performing Test CXX_COMPILER_SUPPORTS__-Wall - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test C_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-array-bounds - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test C_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-nonnull - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test C_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-maybe-uninitialized - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test C_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-format-overflow - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-variable - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test C_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-unused-function - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test C_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings
-- Performing Test CXX_COMPILER_SUPPORTS__-Wno-write-strings - Success
-- Performing Test C_COMPILER_SUPPORTS__-Wno-sign-compare
[... 112 lines cut ...]

ce916510 2023-02-07T19:41:12+02:00 Added link to tutorials for Spanish learners (#1653) - Erick Carvajal -  (HEAD -> master, tag: 2023.02.08, origin/master, origin/HEAD)
bf207eda 2023-02-07T19:23:39+02:00 KLayout GDSII updates (#1608) - Kareem Farid -  ()
1ff228d4 2023-02-07T14:00:41+02:00 [BOT] Update PDK (#1655) - Openlane Bot -  ()
---
Git Remotes

donn	git@github.com:donn/openlane.git (fetch)
donn	git@github.com:donn/openlane.git (push)
fork	git@github.com:kareefardi/OpenLane.git (fetch)
fork	git@github.com:kareefardi/OpenLane.git (push)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (fetch)
origin	git@github.com:The-OpenROAD-Project/OpenLane.git (push)
```

### To Reproduce

[issue_reproducible.1.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10726201/issue_reproducible.1.zip)
[issue_reproducible.2.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10726199/issue_reproducible.2.zip)

`cat issue_reproducible.1.zip issue_reproducible.2.zip > issue_reproducible.zip` and the proceed as expected. 

### Screenshots

[image]

antonblanchard, 2023-02-14:
This is pretty bizarre. I started to look at `io_in[37]` via `detailed_route_debug -dr -net {io_in[37]}` and the macro when placed is huge:

[image]

And the fixed costs mirror that huge macro (eg the pins on met2 in green):

[image]

After routing the macro is correct sized:

[image]

antonblanchard, 2023-02-14:
Ignore that, I was looking at a worker box of course.

antonblanchard, 2023-02-14:
The fixed costs for the macro don't don't extend to the off grid, wrong direction track outside the macro. I suspect we should at least have a via blockage on this track:

[image]

antonblanchard, 2023-02-14:
At a guess, in `FlexGCWorker::Impl::checkMetalSpacing_main` the width calculation of the met2 obstruction looks to just use the 370nm chunk at the edge. We therefore pick the small minspacing value and not the large one:

[image]

That explains why we don't flag a DRC. We aren't also setting the fixed route costs, a separate issue.

maliberty, 2023-02-14:
@osamahammad21 any comments?

donn, 2023-02-16:
Sorry to bother you, but any updates?

osamahammad21, 2023-02-16:
Sorry for the delay, I will start investigating this issue right away.

osamahammad21, 2023-02-19:
Ok so after debugging, it turns out that the upper part of the obstruction is read separate from the the lower part of the obstruction. For the drc checker, they do not represent the same blockage so each one is treated separately. So when it checks for spacing violations, it finds the upper obstruction on its own legal.
I am currently discussing a solution with Matt on how to approach this. For now, a work around would be extending the upper obstructions down to have the required width that would trigger the spacing violation.

kareefardi, 2023-04-04:
Hello. Any updates on this?

maliberty, 2023-04-04:
@osamahammad21 please update

kareefardi, 2023-04-04:
I don't see that same issue. Just confirming if this was fixed.

vijayank88, 2023-06-22:
@kareefardi With latest master the met2 spacing respected with `0.281`. Can you please check once?
[image]

kareefardi, 2023-08-01:
@vijayank88 Sorry for missing this. Using OpenROAD version `41a51eaf4ca2171c92ff38afb91eb37bbd3f36da`, the same issue is there in the top left side left side of the macro at the bottom.

closed 2023-09-19 by maliberty