The-OpenROAD-Project/OpenROAD #2896: Drop or clarify `or-tools` requirement on Mac

donn, 2023-02-14:
### Description

`mpl2` does not appear to be currently supported on Darwin, yet, `or-tools` is listed as a requirement. As far as I can tell,  nothing else uses `or-tools`.

`or-tools` is a pain to package and build on Mac, quite frankly (there are scenarios where the dependency installer is not an option,) so is there another reason the requirement continues to exist or are you just keeping the requirement there just in case?

maliberty, 2023-02-14:
We plan to make mpl2 work on a mac.  Even if we removed or-tools now it will just come back in the near future.

stefanottili, 2023-02-20:
openroad build and ran a couple of weeks ago just fine on a M1 MacBook Air.
Something in the meantime broke something related to shared lib path building the executable, it compiled just fine.
Is using the CMAKE_PREFIX_PATH to point to the homebrew or-tools is not the way to do it any more ?

env | grep or-tools
CMAKE_PREFIX_PATH=/opt/homebrew/opt/or-tools
./build_openroad.sh --local --latest
...
-- Install configuration: "RELEASE"
-- Installing: /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
-- Installing: /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/sta
-- Installing: /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD/lib/libOpenSTA.a
...
openroad -gui
dyld[6376]: Library not loaded: @rpath/libortools.9.dylib
  Referenced from: <080E4EA9-5A62-32D6-B99E-39742491795C> /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD/bin/openroad
  Reason: tried: '/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS@rpath/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib64/libortools.9.dylib' (no such file),
  '/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/System/Volumes/Preboot/Cryptexes/OS/opt/or-tools/lib/libortools.9.dylib' (no such file),
  '/usr/local/lib/libortools.9.dylib' (no such file),
  '/usr/lib/libortools.9.dylib' (no such file, not in dyld cache)
zsh: abort      openroad -gui

vvbandeira, 2023-02-21:
> CMAKE_PREFIX_PATH=/opt/homebrew/opt/or-tools

Note that the latest installer script is installing or-tools in the `brew --prefix or-tools`: https://github.com/The-OpenROAD-Project/OpenROAD/blob/ed167734d8eab758a1846bf31ca7abc7c6b485fb/etc/DependencyInstaller.sh#L141-L143

@donn / @rovinski in my mac, I can build and run OpenROAD (and gcd through the flow) with:
```
export PATH="$(brew --prefix bison)/bin:$PATH"
export PATH="$(brew --prefix flex)/bin:$PATH"
export PATH="$(brew --prefix tcl-tk)/bin:$PATH"
export PKG_CONFIG_PATH="$(brew --prefix tcl-tk)/lib/pkgconfig"
export CMAKE_PREFIX_PATH="$(brew --prefix or-tools)"
export PATH=/Applications/KLayout/klayout.app/Contents/MacOS/:$PATH
alias awk=gawk
```

Also, I disable any env manager like conda (e.g., `conda deactivate`).

rovinski, 2023-02-22:
Haven't checked - I use WSL Ubuntu or RHEL7 for my daily use. MacOS is only when I'm testing builds 🙂.

stefanottili, 2023-02-24:
Feb 24, MacBook Air M1, latest DependencyInstaller.sh upgraded or-tools
==> Upgrading or-tools
  9.5_1 -> 9.5_2 

CMake Error at src/mpl2/CMakeLists.txt:42 (find_package):
  Found package configuration file:
    /opt/homebrew/opt/or-tools/lib/cmake/ortools/ortoolsConfig.cmake
  but it set ortools_FOUND to FALSE so package "ortools" is considered to be
  NOT FOUND.  Reason given by package:
  The following imported targets are referenced, but are missing:
  Coin::CbcSolver Coin::OsiCbc Coin::ClpSolver Coin::OsiClp

stefanottili, 2023-02-26:
After using ./build_openroad.sh --clean ./build_openroad.sh --local --latest I get past the cmake issue.
But then the build fails to link the openroad executable with something unrelated to or-tools.

vvbandeira, 2023-02-27:
> After using ./build_openroad.sh --clean ./build_openroad.sh --local --latest I get past the cmake issue. But then the build fails to link the openroad executable with something unrelated to or-tools.

Can you share your `build_openroad.log`? Is the link error something about "symbol not found for arch"?

stefanottili, 2023-02-27:
if you want the full, rather large build_openroad.log:
should I attach it or can I drop it somewhere were you can access it ?

[100%] Linking CXX executable openroad
..
ld: warning: directory not found for option '-F/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/System/Library/Frameworks'
Undefined symbols for architecture arm64:
  "operations_research::sat::CpModelBuilder::AddExactlyOne(absl::lts_20230125::Span<operations_research::sat::BoolVar const>)", referenced from:
      par::TPpartitioner::OptimalPartCplexWarmStart(std::__1::shared_ptr<par::TPHypergraph>, std::__1::vector<std::__1::vector<float, std::__1::allocator<float> >, std::__1::allocator<std::__1::vector<float, std::__1::allocator<float> > > > const&, std::__1::vector<int, std::__1::allocator<int> >&) in par.a(TPPartitioner.cpp.o)
  "absl::lts_20230125::container_internal::kEmptyGroup", referenced from:
      par::TPpartitioner::OptimalPartCplexWarmStart(std::__1::shared_ptr<par::TPHypergraph>, std::__1::vector<std::__1::vector<float, std::__1::allocator<float> >, std::__1::allocator<std::__1::vector<float, std::__1::allocator<float> > > > const&, std::__1::vector<int, std::__1::allocator<int> >&) in par.a(TPPartitioner.cpp.o)
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
gmake[2]: *** [src/CMakeFiles/openroad.dir/build.make:335: src/openroad] Error 1

vvbandeira, 2023-02-27:
@stefanottili, just the top of the file where the cmake prints the version and path of the dependencies is enough -- ~150 first lines up to "-- Build files have been written to:". I think I have seen something similar.

stefanottili, 2023-02-27:
```
./build_openroad.sh --local --latest
[INFO FLW-0027] Saving logs to build_openroad.log
[INFO FLW-0028] ./build_openroad.sh --local --latest
[INFO FLW-0002] Updating git submodules.
Submodule path 'tools/OpenROAD': checked out '5c85b36f8d50975d4e57af6678a0bbfe62046f0e'
[INFO FLW-0004] Updating OpenROAD app to the HEAD  of origin/master.
Previous HEAD position was 5c85b36f8 Merge pull request #2908 from The-OpenROAD-Project-staging/par-coverity
HEAD is now at c9d8d8889 Merge pull request #2947 from The-OpenROAD-Project-staging/odb-dead-code
From https://github.com/The-OpenROAD-Project/OpenROAD
 * branch                master     -> FETCH_HEAD
Already up to date.
[INFO FLW-0001] Using local build method. This will create binaries at 'tools/install' unless overwritten.
[INFO FLW-0017] Compiling Yosys.
$BREW_PREFIX is [/opt/homebrew/opt]
mkdir -p /Users/user/OpenROAD-flow-scripts/tools/install/yosys/bin
cp yosys yosys-config yosys-abc yosys-filterlib yosys-smtbmc /Users/user/OpenROAD-flow-scripts/tools/install/yosys/bin
strip -S /Users/stefanthiede/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys
strip /Users/user/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-abc
strip /Users/user/OpenROAD-flow-scripts/tools/install/yosys/bin/yosys-filterlib
mkdir -p /Users/user/OpenROAD-flow-scripts/tools/install/yosys/share/yosys
cp -r share/. /Users/user/OpenROAD-flow-scripts/tools/install/yosys/share/yosys/.
[INFO FLW-0018] Compiling OpenROAD.
-- OpenROAD version: v2.0-7029-gc9d8d8889
-- System name: Darwin
-- Compiler: AppleClang 14.0.0.14000029
-- Build type: RELEASE
-- Install prefix: /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD
-- C++ Standard: 17
-- C++ Standard Required: ON
-- C++ Extensions: OFF
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- Using SWIG >= 4.1.1 -flatstaticmethod flag for python
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0")  
-- boost: 1.81.0
-- spdlog: 1.9.2
CMake Warning at src/CMakeLists.txt:225 (message):
  spdlog: SPDLOG_FMT_EXTERNAL=ON

-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- STA version: 2.3.3
-- STA git sha: 2a2b5cd59f5d1b596f20cbd39609c52b33299e06
-- System name: Darwin
-- Compiler: AppleClang 14.0.0.14000029
-- Build type: RELEASE
-- Build CXX_FLAGS: -O3 -DNDEBUG
-- Install prefix: /Users/user/OpenROAD-flow-scripts/tools/install/OpenROAD
-- TCL library: /opt/homebrew/opt/tcl-tk/lib/libtcl8.6.dylib
-- TCL header: TCL_HEADER-NOTFOUND
-- SSTA: 0
-- STA executable: /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/src/sta/app/sta
-- GPU is not enabled
-- GUI is enabled
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization 
-- Could NOT find VTune (missing: VTune_LIBRARIES VTune_INCLUDE_DIRS) 
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found suitable version "1.81.0", minimum required is "1.78")  
-- Found Boost: /opt/homebrew/lib/cmake/Boost-1.81.0/BoostConfig.cmake (found version "1.81.0") found components: serialization system thread 
-- Removing MPL2 to avoid run time fatal error.
-- TCL readline disabled
-- Tcl Extended disabled
-- Python3 enabled
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/user/OpenROAD-flow-scripts/tools/OpenROAD/build
```

stefanottili, 2023-03-02:
remove par and par_py until they work on apple silicon. #2967

https://github.com/The-OpenROAD-Project/OpenROAD/pull/2967/commits/15883c5c548a5e736a736517d88bed9a47fa78b9

./build_openroad.sh --local --latest

run's thru and builds openroad on M1

donn, 2023-03-07:
Seems to be fine on at least x86-64 now.

closed 2023-03-07 by donn