The-OpenROAD-Project/OpenROAD #2904: Add respect ordering constraint to set_io_pin_constraint

oharboe, 2023-02-15:
### Description

Add option to set_pin_constraint so that it is possible to constrain to an edge as well as deciding the ordering of the pins.

### Suggested Solution

Add ```-order``` option that orders the pins in increasing x/z ordering as the names were specified to the ```-pin_names``` argument

```
set_io_pin_constraint -order
```

https://github.com/The-OpenROAD-Project/OpenROAD/blob/42cea5fc6fc5244bbaf93d591fa7c3ab162cbc59/src/ppl/src/IOPlacer.tcl#L121

eder-matheus, 2023-02-24:
@oharboe Do you want the pins adjacent to each other? Or can we have other pins between two pins of the list?

oharboe, 2023-02-24:
> @oharboe Do you want the pins adjacent to each other? Or can we have other pins between two pins of the list?

I thought there was a "group" constraint that could express this?

I can easily imagine someone wanting control over this, such as when having a bus and other unconstrained pins. In this case thet would want the bus pins not intermingled with other pins.

eder-matheus, 2023-02-24:
@oharboe Yes, we have the -group option where you can define a set of pins to be placed together, following the ordering of the names. In this case, they will be placed following the min distance between each pin. I'm updating this option to have the pins ordered in ascending x/y position.

You can see an example of how to use the `-group_pins` option here: https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/src/ppl/test/group_pins6.tcl.
 
You can constrain the whole group to an edge by only constraining one pin of the group, so your script doesn't need to be as verbose as this example.

oharboe, 2023-02-24:
> @oharboe Yes, we have the -group option where you can define a set of pins to be placed together, following the ordering of the names. In this case, they will be placed following the min distance [...]

@maliberty Makes sense to me. The only concern I have is that this breaks the API, something that as OpenROAD becomes more mature should never do without careful consideration. E.g. if this has an.impact on QoR, good or bad, an upgrade can become less trivial.

Lets say this change makes routing at top level easier and macro level harder. That is a big change.

Therefore I would recommend a non change to existing API, a new -order option only considered for -group_pins, unless after due consideration a change in the API is the right thing.

eder-matheus, 2023-02-24:
@oharboe I think adding the `-order` option to the `-group_pins`, which already is an option of the `place_pins` command, can be confusing.

I suggest adding the `-group` option to the `set_io_pin_constraint` command, where you can decide using or not the `-order` option. IMO, this is a better place for defining pin groups than in the `place_pins` command, but will not change our current API. We will have redundant options for a while, but I don't see a big problem on this.

oharboe, 2023-02-24:
> @oharboe I think adding the `-order` option to the `-group_pins`, which already is an option of the `place_pins` command, can be confusing.
> 
> I suggest adding the `-group` option to the `set_io_pin_constraint` command, where you can decide using or not the `-order` option. IMO, this is a better place for defining pin groups than in the `place_pins` command, but it will not change our current API. We will have redundant options for a while, but I don't see a big problem with this.

I see. Very good. Makes sense. I would generally expect those who group pins to want them at minimum distance and in order, us included. However, this is just a software engineers expectation, I dont have any ASIC experience.

eder-matheus, 2023-02-24:
> I see. Very good. Makes sense. I would generally expect those who group pins to want them at minimum distance and in order, us included. However, this is just a software engineers expectation, I dont have any ASIC experience.

Min distance is already respected with the current `-group_pins` option. The only thing missing is that in the top and left edges, the pins are placed in reversed order, due to the way PPL computes the valid positions for the pins. My update will just make sure the pins will be placed in ascending order, independent of what edge they are placed on.

closed 2023-02-28 by eder-matheus