The-OpenROAD-Project/OpenROAD #2910: deltaDebug.py leaves files with a few insts and nets that is megabytes

oharboe, 2023-02-17:
### Description

One great thing about deltaDebug.py is that it can reduce confidential examples to the point where they can be shared as bug reports.

However, when the resulting .odb file with a few insts and nets is  megabytes, that doesn't install confidence that no confidential information is divulged.

### Suggested Solution

The .odb files should be small and then converted to a textual representation that can be inspected for confidential information.

maliberty, 2023-02-17:
You can write_lef & write_def to get text.  If you want to share a confidential one with me in private I can try to analyze the size.  Its possible the space represents deleted objects as we can only compress an array down to the last allocated element.

QuantamHD, 2023-02-17:
I mean the standard cells and other library elements represent a non-zero amount of size in ODB. Do we delete dbMasters that aren't used?

maliberty, 2023-02-17:
No the lef elements remain unchanged which is a likely cause.  We could add a final pass to remove unused cells.

maliberty, 2023-03-21:
The db doesn't have a way to delete dbMaster as we don't normally do so.  That will require an enhancement

NairiBaveyann, 2024-04-26:
Hello @oharboe @maliberty @QuantamHD I've started working on this issue, please feel free to assign it to me.

maliberty, 2024-04-28:
Thanks for looking at it.

NairiBaveyann, 2024-05-08:
Hello, currently I am working on the implementation of the new pass (which will delete unused cells) and I need a test case to reproduce the issue and check the implementation correctness. Could you please attach an appropriate experimental design and expected results?

NairiBaveyann, 2024-05-29:
Hi, I've added a new function that removes unused masters in the .odb file and reduces its size.
 I am attaching the resulting odb files (original and with removed masters) for your review. 
Note, in the latest version of OpenRoad, when I run test-delta-debug.sh it gives "No error found in original input file" instead of cutting via "Iter: 100" and reducing design, so I've tested a new function with the cfe92e17a7d986ea9bd82031a61d47120e30dc7a  version.
[ODB_files.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/15486671/ODB_files.zip)

oharboe, 2024-05-29:
[image]

maliberty, 2024-05-29:
The file size looks good.  It would be easier to review a PR with the code changes.

PR #5183 (2024-06-03): Added a new function to remove unused dbMasters.
This PR closes #2910. C++ and Python new functions have been added:

- `int dbDatabase::removeUnusedMasters()` - removes unused masters
- `def remove_unused_masters(self)` - to prepare database and call above C++ function
closed 2024-06-03 by eder-matheus