The-OpenROAD-Project/OpenROAD #2919: Correct the DB Generated names in dbReadVerilog.cc

andyfox-rushc, 2023-02-19:
### Description

The odb names generated in dbReadVerilog.cc are verilog names sometimes referring to intermediate points in the hierarchy. Please can they be replaced with structural names.  This would then allow the automatic construction of a hierarchical view from the output of the flow.

Here is an example (taken for the aes test case, run in hierarchical mode):

 NAND4_X1 \u0/u0/_1273_  (.A1(\u0/u0/_0062_ ),
    .A2(\u0/u0/_0492_ ),
    .A3(\u0/u0/_0512_ ),
    .A4(\u0/u0/_0517_ ),
    .ZN(\u0/subword[28] ))

The ZN output is connected to u0/subword[28] but this is outside the scope of the instance U0/U0/_1273.  This means that it is tricky to construct the hierarchy from the verilog alone. If however the output is connected to u0/u0/subword[28] then the hierarchical connection is self evident.

### Suggested Solution

The name is generated as a path name in the dbReadVerilog.cc code here:

void
Verilog2db::makeDbNets(const Instance* inst)
{
  bool is_top = (inst == network_->topInstance());
  NetIterator* net_iter = network_->netIterator(inst);
  while (net_iter->hasNext()) {
    Net* net = net_iter->next();
    const char* net_name = network_->pathName(net);

Rather than use the net path name, construct the "structural "net name using the instance path name using some variant like:

const char* instance_name = network_ -> pathName(inst);
  
    //get the path name to the net
    std::string leaf_net_name = network_ -> name(net);
    std::string sep("/");
    std::string i_name = std::string(instance_name);
    std::string structural_net_name =  i_name + sep + leaf_net_name;

### Additional Context

The context here is to allow the construction of a hierarchical netlist for the OpenRoad flow. If the above naming convention is used (fully scoped hierarchical names) then the following (trivial) algorithm can be used to construct a hierarchical netlist (H)  from the input design (synth_1.v) and the final resultant flat netlist from open road (F).

Algorithm : Copy And Wire Algorithm.

Steps:
1. Record the invariant points (same in H and L) reachable from ports in modules of H. Annotate and store these. (this is so that we can recover port names)
2. Remove all liberty instances from hierarchical network H. Preserve hierarchy modules M. 
3. Copy in all the liberty instances from the flat network F. Use instance name to assure correct place in hierarchy. Create output nets for each liberty gate. Axiom: require output net to be scoped in same module as driver – ok for sinks to be in other modules).
4. Wire up all instance inputs and outputs. Shared ports where possible.  Stash names as properties. Add ports to go up and down the hierarchy to resolve cross module connections.
5. Traverse ports on merged hierarchical network. Find invariant points. Recover port names. Example: walk from input ports in modules in merged network until 1st invariant point hit, recover port name stashed in step 1.

maliberty, 2023-02-20:
This looks to be based on the idea that you want to reconstruct the hierarchy.  Our approach is to carry the hierarchy in the db and not to reconstruct it later.  The instance tree is already there and work is underway to keep the full connectivity.   I think that is a better solution.

It would be helpful to have a minimal example.  It is 'normal' to me to refer to the highest net in the collection not the one at the driver.  How do you handle multi-driver nets in your approach?

maliberty, 2023-02-20:
This would also affect the user experience.  If someone has a top level net foo and tries to find it by name they will fail if we named it by driver (eg a/b/c/d/driver instead).  It would also be incompatible with the behavior of every other EDA tool afaik.

The example of how to create the name would produce a net name that doesn't even exist in the input verilog

andyfox-rushc, 2023-02-20:
Thank you for the clear explanation. Sounds like it is best to keep this exactly as is right now and then to carry forwards the hierarchy as you are doing. Thanks for the helpful commentary. I agree with you that carrying forwards the hierarchy is the best solution and I note you are still working on that (adding pins and connectivity).

andyfox-rushc, 2023-02-20:
Not needed because openroad will be forward propagating the hierarchy (and presumably updating it as the design changes) which is a much preferred solution to inferring the hierarchy at the end.

closed 2023-02-20 by andyfox-rushc