The-OpenROAD-Project/OpenROAD #2920: Make the write_db and read_db work before and after link step in flow.

andyfox-rushc, 2023-02-20:
### Description

Entry to the OpenRoad flow is by load.tcl. This reads the verilog, libraries etc and does a link. The link generates a flattened view of the design which is passed to the other tools. The frontend (dbReadVerilog.cc does indeed nicely generate a verilog hierarchy via the add dbModule data structures).

It would be nice if it was possible to do a write_db before the link step and then read the db back and do a subsequent link. It is as if that it is only after the link that the db is kosher and after link we seem to lose the hierarchy information (all the dbModule class information).

To reproduce modify the load.tcl script to write out the db after reading the verilog.

	read_verilog $::env(RESULTS_DIR)/$design_file
	write_db "before_link.db"
	link_design $::env(DESIGN_NAME)
	write_db "after_link.db"

OpenROAD v2.0-6842-g1cf0b3144 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
openroad> read_db before_link.db
openroad> link_design aes_cipher_top
db linking design
[ERROR STA-0008] cell type aes_cipher_top can not be linked.
STA-0008

### Suggested Solution

The linker is essentially flattening the design -- it seems to traverse to the leaf cells and then wire them up. An alternative approach would be to segregate the linking step into two commands: link_flat -- which generates a flattened view (as now) and link_hier -- which links all the leaf cells but preserves the hierarchy. 

With the exception of RTLMP most of the downstream tools seem to work on the flattened odb view. But perhaps in the future a link_hier might allow for tool use with a truly hierarchical odb (block in block) view. Link_hier could then be used to signal to the downstream tools that the hierarchical database is being generated and propagated.

### Additional Context

From a flow context it is also useful to be able save the design information during load.tcl and then once everything loaded up in a separate file do the link. It seems weird not to be able to read design information in, save it, and then write out the db and resume later.

maliberty, 2023-02-20:
I don't understand your description.  There is nothing in the db before the link step (Verilog2db is run by dbLinkDesign).

maliberty, 2023-02-20:
There is also nothing lost during linking as that is when the dbModules are created

andyfox-rushc, 2023-02-20:
Thank you for the helpful explanation. Sounds like calling write_db before link is meaningless, so this is an irrelevant point ! Thanks again for the comments.

andyfox-rushc, 2023-02-20:
The link step generates the db, so calling write_db before link is meaningless.

closed 2023-02-20 by andyfox-rushc