The-OpenROAD-Project/OpenROAD #2934: unable to check congestion heatmap during GRT

proppy, 2023-02-23:
### Describe the bug

During:
```
[STEP 12]
[INFO]: Running Global Routing
```
whe GRT fails with:
```
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
```
It does seems to dump the current odb database in the `runs` folder.

This make it no possible to follow the error message advise, because `or_gui` will `read_db`  for the previous step `11-top.resized.odb` can complain about missing data when trying to display the heat map:
```
[WARNING GUI-0066] Heat map "Routing Congestion" has not been populated with data.
```

### Expected Behavior

Before falling the flow dump the current state in a db file run `runs/<TAG>/tmp`.

### OpenROAD Environment

```shell
openroad                  2.0_6924_g037e4e3a4 20230222_073528    litex-hub
```
```

### OpenLane Environment

```shell
openlane                  2023.02.19_0_g95580b00 20230222_073528    litex-hub
```
```

### To Reproduce

[congestion-heatmap.zip](https://github.com/The-OpenROAD-Project/OpenROAD/files/10810156/congestion-heatmap.zip)

### Relevant log output

```shell
[INFO]: Starting routing process...
[STEP 12]
[INFO]: Running Global Routing Resizer Timing Optimizations (log: digital/runs/congestion/logs/routing/12-resizer.log)...
[INFO]: Executing openroad with Tcl script '../../../../silicon-gf180mcuC-env/share/openlane/scripts/openroad/resizer_routing_timing.tcl'...
OpenROAD 037e4e3a46ec91e8020445672d86afc9ff675cd6 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO]: Reading ODB at '/home/proppy/src/github.com/spnadig/GF180_GAFE_1/digital/runs/congestion/tmp/cts/11-top.resized.odb'...
[WARNING STA-0357] virtual clock __VIRTUAL_CLK__ can not be propagated.
[INFO]: Setting signal min routing layer to: Metal2 and clock min routing layer to Metal2. 
[INFO]: Setting signal max routing layer to: Metal5 and clock max routing layer to Metal5. 
-congestion_iterations 50 -verbose
[INFO GRT-0020] Min routing layer: Metal2
[INFO GRT-0021] Max routing layer: Metal5
[INFO GRT-0022] Global adjustment: 20%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer Metal1  Track-Pitch = 0.5600  line-2-Via Pitch: 0.5450
[INFO GRT-0088] Layer Metal2  Track-Pitch = 0.5600  line-2-Via Pitch: 0.5800
[INFO GRT-0088] Layer Metal3  Track-Pitch = 0.5600  line-2-Via Pitch: 0.5800
[INFO GRT-0088] Layer Metal4  Track-Pitch = 0.5600  line-2-Via Pitch: 0.5800
[INFO GRT-0088] Layer Metal5  Track-Pitch = 0.9000  line-2-Via Pitch: 0.9600
[INFO GRT-0019] Found 0 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 2
[INFO GRT-0003] Macros: 1
[INFO GRT-0004] Blockages: 4

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
Metal1     Horizontal          0             0          0.00%
Metal2     Vertical      1784286       1150303          35.53%
Metal3     Horizontal    1784286       1122364          37.10%
Metal4     Vertical      1784286       1102497          38.21%
Metal5     Horizontal    1019592        598478          41.30%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[INFO GRT-0197] Via related to pin nodes: 27
[INFO GRT-0198] Via related Steiner nodes: 0
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 50
[INFO GRT-0112] Final usage 3D: 1565
[ERROR GRT-0118] Routing congestion too high. Check the congestion heatmap in the GUI.
Error: resizer_routing_timing.tcl, 39 GRT-0118
[ERROR]: during executing openroad script /home/proppy/silicon-gf180mcuC-env/share/openlane/scripts/openroad/resizer_routing_timing.tcl
[ERROR]: Log: digital/runs/congestion/logs/routing/12-resizer.log
[ERROR]: Last 10 lines:
child process exited abnormally
```
```

### Screenshots

![image](https://user-images.githubusercontent.com/5268/220818840-e446c786-5d49-4869-b854-950d4aad0388.png)

proppy, 2023-02-23:
a workaround is to set `GRT_ALLOW_CONGESTION` to `true` to be able to open `runs/<TAG>/tmp/12-top.odb`:
[image]

maliberty, 2023-02-23:
This sounds like an OL bug not an OR one - is this in the right repo?

proppy, 2023-02-23:
@maliberty I'm still debating this :)

The error message does come from OpenROAD so GRT seems to imply that a db is available by the time it outputs this error message:
https://github.com/The-OpenROAD-Project/OpenROAD/blob/1af9a6cdefb5c627bb5ebd73148d7f217fc6d308/src/grt/src/GlobalRouter.cpp#L300-L305

Does OpenLane has a way to dump the db after GRT exists?

maliberty, 2023-02-23:
The message says nothing about a db being available.  You might already be in the gui or plan to start it from the current session.

maliberty, 2023-02-23:
The OL question belongs to the OL team....

proppy, 2023-02-23:
> The message says nothing about a db being available. You might already be in the gui or plan to start it from the current session.

Ah that make total sense, I didn't realize that could also been displayed during interactive usage of the GUI (and that's probably the primary use case).

So the issue is more about how OpenLane can surface an actionable message out of routing congestion error. Thanks!

closed 2023-02-23 by proppy