The-OpenROAD-Project/OpenROAD #2953: OpenROAD Singleton Removal

QuantamHD, 2023-02-27:
### Description

@grotival, @fgaray and myself have all been trying to use OpenROAD in a more library-like fashion in threaded environments.

We’ve uncovered that this will require the removal of static across the OpenROAD codebase.

In particular we would like to remove usage of OpenRoad::openroad() across the codebase, and instead use injected OpenRoad objects.

### Suggested Solution

See the following before vs after in the Design class:

## Before

```c++
Design::Design(Tech* tech) : tech_(tech)
{
}

void Design::readVerilog(const std::string& file_name)
{
  auto chip = tech_->getDB()->getChip();
  if (chip && chip->getBlock()) {
    getLogger()->error(utl::ORD, 36, "A block already exists in the db");
  }

  auto app = OpenRoad::openRoad();
  app->readVerilog(file_name.c_str());
}
```

## After

```c++
Design::Design(Tech* tech, std::unique_ptr<OpenRoad> openroad) : tech_(tech), openroad_(openroad)
{
}

void Design::readVerilog(const std::string& file_name)
{
  auto chip = tech_->getDB()->getChip();
  if (chip && chip->getBlock()) {
    getLogger()->error(utl::ORD, 36, "A block already exists in the db");
  }

  openroad_->readVerilog(file_name.c_str());
}
```

We plan to start the transition in our spare cycles with the end goal being a thread safe OpenRoad library.

QuantamHD, 2023-02-27:
This would be from the C++ API, and the benefits of threading are simpler execution/communication model, and the general preference of Google binaries.

QuantamHD, 2023-02-27:
It can construct one by default if not provided.

maliberty, 2023-02-27:
So
```
Design::Design(Tech* tech, std::unique_ptr<OpenRoad> openroad = std::unique_ptr<OpenRoad>())
```
in that case wouldn't it be better to add it to Tech() since Design takes it as an arg?

QuantamHD, 2023-02-28:
I'm not so sure since I logically think of Tech as a generic tech library and not the design itself. OpenRoad* holds the database so to me it logically makes sense to be owned by Design.

maliberty, 2023-02-28:
Openroad owns the db implies it own the Tech as well.  There is no tech outside of an OpenRoad

QuantamHD, 2023-02-28:
We should probably separate them. Tech should have it's own db that it copies into new instances of Design.

maliberty, 2023-02-28:
What is the benefit?  You'll always wind up with two copies of your LEF in memory that way.  Also there is no code to do that copying today.

QuantamHD, 2023-02-28:
There's a clone function that we'll enhance to work for this purpose.

The benefit is that you end up with a pure Tech that can be used in multiple places without having to read a bunch of setup files.

maliberty, 2023-02-28:
In the ordinary use case (one tech, one design) you will double your LEF memory usage which isn't great.

QuantamHD, 2023-02-28:
I think we've observed with @oharboe's comments on Delta debugger that that amounts to a few MB at most.

maliberty, 2023-02-28:
Its also error prone is the tech is ever modifiable.

maliberty, 2023-02-28:
Are we designing a system that can't have a large tech?  What is we want to store the gds in the tech someday?

QuantamHD, 2023-02-28:
The tech is modifiable like any object.

I think it's a trivial matter to make this work efficiently. We could accept a unique_ptr to transfer ownership, or a copy mechanism depending on the users preference. It seems easy enough to make the implementation efficient, and IMO we should prioritize user experience over implementation complexity.

maliberty, 2025-01-29:
@QuantamHD still relevant?