The-OpenROAD-Project/OpenROAD #2957: `src/rmp/src/Restructure.cpp` breaks when using system ABC

donn, 2023-02-28:
### Describe the bug

I build OpenROAD ABC separately from OpenROAD to better cache builds, as ABC represents around 50% of OpenROAD's compile time from my experience and is changed very infrequently + mitigates https://github.com/The-OpenROAD-Project/OpenROAD/issues/2833, where the most frequent offender is the `libabc.a` creation.

The file `src/rmp/src/Restructure.cpp` has two imports:

```c++
#include "base/abc/abc.h"
#include "base/main/abcapis.h"
```

These two imports assume the headers are not being used as a system library.

This is the simplest workaround:
```sh
sed -i 's@#include "base/abc/abc.h"@#include <base/abc/abc.h>@' src/rmp/src/Restructure.cpp
sed -i 's@#include "base/main/abcapis.h"@#include <base/main/abcapis.h>@' src/rmp/src/Restructure.cpp
```

### Expected Behavior

The `#include` should work regardless of the mode ABC is being used in.

### Environment

```shell
Git commit: efe3d3e2e9b7e34052965890a8c5fb74358b2412
kernel: Darwin 22.3.0
os: macOS 13.2.1
cmake version 3.24.3

---

clang version 11.1.0
Target: x86_64-apple-darwin
Thread model: posix
InstalledDir: /nix/store/f22xbkm1ml2wvimnhz97ikbaniv4695v-clang-11.1.0/bin
```

### To Reproduce

N/A

QuantamHD, 2023-02-28:
Can you add a -I flag to include the system header when you generate the cmake project.

donn, 2023-02-28:
Already do: didn't work

```sh
-DCMAKE_CXX_FLAGS=-I$ABC_PREFIX/include
```

donn, 2023-02-28:
I should note: the angled brackets did not work without the `-I` flag either iirc.

QuantamHD, 2023-02-28:
Ah, that's probably because we don't propagate compile flags to the abc subproject. I think there's some cmake function that will let you do that. 

However, the only tier 1 supported version of ABC is the version bundled with OpenROAD so I think it would require an enhancement from you or another interested party to make this work more seamlessly in your flow.

donn, 2023-02-28:
To be clear, I am building the one in https://github.com/the-openroad-project/abc, but I getcha.

I'll see if I can come up with something that works for everybody.

maliberty, 2025-08-26:
no progress