The-OpenROAD-Project/OpenROAD #2962: Dumping report from openroad tcl shell in a seperate file

khaledhasanzami, 2023-03-01:
### Describe the bug

I was trying to dump a report using below command from shell in a text file.

```
report_check_types -max_delay > test.txt
```
It creates the text file but does not generate the report in the file. Generates an empty file.

The above command generates report in gui mode but does not generate export in any seperate file.

I have also tried saving the report manually by below code:
```
set f [open test.txt w]; foreach a [exec echo [report_checks -path_group reg2reg -path_delay max]] { puts $f "$a" }; close $f
```
It also generates an empty file.

Summary:

- Goal: Saving report in a seperate file from openroad tcl shell
- Issue: Nothing is exported from openroad tcl shell

### Expected Behavior

Saving report in a seperate file from openroad tcl shell

### Environment

```shell
Any env
```

### To Reproduce

- load any design (odb)
- try to dump the file using 'report_check_types -max_delay > test.txt' or 'set f [open mostak.txt w]; foreach a [exec echo [report_checks -path_group reg2reg -path_delay max]] { puts $f "$a" }; close $f'

vijayank88, 2023-03-01:
AFAIK, no option to dump logs for a command directly. You need put the commands in tcl/shell script and execute by following command will copy logs into the file.
`./run.sh | tee test.txt`

@vvbandeira can you confirm it?

maliberty, 2023-03-01:
standalone sta supports '>' but it is not supported in OR as all output goes through the Logger.

QuantamHD, 2023-03-01:
@khaledhasanzami what we do in practice is create custom report commands that output json. If you look at the implementation of the command you're interested in it's usually easy enough to steal it's implementation, and do whatever you want.

maliberty, 2023-03-01:
Timing reports have a lot of logic.  Some tools offer the ability to return the timing report as a tcl list of lists.  Probably that would be the better solution for a timing report.  Other cases are probably easier.

khaledhasanzami, 2023-03-02:
> Timing reports have a lot of logic. Some tools offer the ability to return the timing report as a tcl list of lists. Probably that would be the better solution for a timing report. Other cases are probably easier.

Could you pklease elaborate?

maliberty, 2023-03-02:
I was responding to "it's usually easy enough to copy it's implementation".  I don't think that is the case for report_checks.  I'm not sure it is a good idea in general but particular not in this case.

QuantamHD, 2023-03-02:
@khaledhasanzami 

For example `report_check_type -max_delay` basically just calls this

```tcl
    set path_ends [find_path_ends "NULL" {} "NULL" 0 \
		     $corner $path_min_max $group_count 1 0 \
		     $slack_min $slack_max \
		     0 {} \
		     $setup $hold \
		     $recovery $removal \
		     $clk_gating_setup $clk_gating_hold]
    report_path_ends $path_ends

```

kareefardi, 2023-04-04:
> @khaledhasanzami what Google does in practice is create custom report commands that output json strings. If you look at the implementation of the command you're interested in it's usually easy enough to copy it's implementation, and do whatever you want.

@QuantamHD is Google's re-implementation public?

maliberty, 2023-04-04:
https://github.com/The-OpenROAD-Project/OpenROAD/discussions/3029 might be helpful

kareefardi, 2023-04-04:
It is. Thanks!