The-OpenROAD-Project/OpenROAD #2968: How to install the dependencies correctly?

b224hisl, 2023-03-02:
### Describe the bug

Hello, I'm trying to build openroad, so I intall the dependencies first, but I got this kind of problem, I don't know how to handle it.

### Expected Behavior

 install the dependencies correctly

### Environment

```shell
Git commit: 0264023b6c2a8ae803b8d440478d657387277d93
kernel: Linux 4.18.0-240.1.1.el8_3.x86_64
os: CentOS Linux 8
cmake version 3.20.3
CMake Error at /usr/local/cmake-3.20.3-linux-x86_64/share/cmake-3.20/Modules/CMakeTestCXXCompiler.cmake:59 (message):
  The C++ compiler

    "/usr/local/bin/c++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /tmp/tmp.ut2MlU26q6/CMakeFiles/CMakeTmp
    
    Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_202f5/fast && /usr/bin/gmake  -f CMakeFiles/cmTC_202f5.dir/build.make CMakeFiles/cmTC_202f5.dir/build
    gmake[1]: Entering directory '/tmp/tmp.ut2MlU26q6/CMakeFiles/CMakeTmp'
    Building CXX object CMakeFiles/cmTC_202f5.dir/testCXXCompiler.cxx.o
    /usr/local/bin/c++    -o CMakeFiles/cmTC_202f5.dir/testCXXCompiler.cxx.o -c /tmp/tmp.ut2MlU26q6/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
    Linking CXX executable cmTC_202f5
    /usr/local/cmake-3.20.3-linux-x86_64/bin/cmake -E cmake_link_script CMakeFiles/cmTC_202f5.dir/link.txt --verbose=1
    /usr/local/bin/c++ CMakeFiles/cmTC_202f5.dir/testCXXCompiler.cxx.o -o cmTC_202f5 
    /usr/bin/ld: cannot find crtbegin.o: No such file or directory
    /usr/bin/ld: cannot find -lstdc++
    collect2: error: ld returned 1 exit status
    gmake[1]: *** [CMakeFiles/cmTC_202f5.dir/build.make:99: cmTC_202f5] Error 1
    gmake[1]: Leaving directory '/tmp/tmp.ut2MlU26q6/CMakeFiles/CMakeTmp'
    gmake: *** [Makefile:127: cmTC_202f5/fast] Error 2
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:84 (project)

-- The CXX compiler identification is GNU 7.5.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - failed
-- Check for working CXX compiler: /usr/local/bin/c++
-- Check for working CXX compiler: /usr/local/bin/c++ - broken
-- Configuring incomplete, errors occurred!
```

### To Reproduce

run `sudo ./etc/DependencyInstaller.sh` in the OpenROAD root folder

### Relevant log output

```shell
(base) [yzhu@rios-cad1 OpenROAD]$sudo ./etc/DependencyInstaller.sh 
Updating Subscription Management repositories.

This system is registered to Red Hat Subscription Management, but is not receiving updates. You can use subscription-manager to assign subscriptions.

No match for argument: ius-release
No match for argument: epel-release
Dependencies resolved.
================================================================================================================
 Package                Architecture             Version                  Repository                       Size
================================================================================================================
Removing:
 lcov                   noarch                   1.14-1                   @@commandline                   369 k

Transaction Summary
================================================================================================================
Remove  1 Package

Freed space: 369 k
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                        1/1 
  Erasing          : lcov-1.14-1.noarch                                                                     1/1 
  Running scriptlet: lcov-1.14-1.noarch                                                                     1/1 
  Verifying        : lcov-1.14-1.noarch                                                                     1/1 
Installed products updated.

Removed:
  lcov-1.14-1.noarch                                                                                            

Complete!
Updating Subscription Management repositories.

This system is registered to Red Hat Subscription Management, but is not receiving updates. You can use subscription-manager to assign subscriptions.

Last metadata expiration check: 2:43:59 ago on Thu 02 Mar 2023 04:34:50 PM CST.
lcov-1.14-1.noarch.rpm                                                           31 kB/s |  99 kB     00:03    
Dependencies resolved.
================================================================================================================
 Package                Architecture             Version                   Repository                      Size
================================================================================================================
Installing:
 lcov                   noarch                   1.14-1                    @commandline                    99 k

Transaction Summary
================================================================================================================
Install  1 Package

Total size: 99 k
Installed size: 369 k
Downloading Packages:
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                        1/1 
  Installing       : lcov-1.14-1.noarch                                                                     1/1 
  Running scriptlet: lcov-1.14-1.noarch                                                                     1/1 
  Verifying        : lcov-1.14-1.noarch                                                                     1/1 
Installed products updated.

Installed:
  lcov-1.14-1.noarch                                                                                            

Complete!
Updating Subscription Management repositories.

This system is registered to Red Hat Subscription Management, but is not receiving updates. You can use subscription-manager to assign subscriptions.

Last metadata expiration check: 2:44:05 ago on Thu 02 Mar 2023 04:34:50 PM CST.
ius-release-el7.rpm                                                             8.3 kB/s | 8.2 kB     00:00    
Error: 
 Problem: conflicting requests
  - nothing provides epel-release = 7 needed by ius-release-2-1.el7.ius.noarch
(try to add '--skip-broken' to skip uninstallable packages or '--nobest' to use not only best candidate packages)
```

### Additional Context

here is my linux platform
[image]

vijayank88, 2023-03-02:
Please run `sudo yum update` and check cmake version. Minimum required version is 3.24.3.
You're using 3.20.3 version.

vijayank88, 2023-03-02:
[image]
As per user update from slack channel, CentOS 8 unable to pull packages from existing path with `sudo yum update`. Suggested to try this fix: https://www.tecmint.com/error-failed-to-download-metadata-for-repo-appstream/

vijayank88, 2023-03-09:
@b224hisl Are you able to resolve it?