The-OpenROAD-Project/OpenROAD #2988: Make command in ibex processor crashes after 70%

Eyantra698Sumanto, 2023-03-08:
### Describe the bug

OpenROAD crashed in between 70% with a lot of violations.

### Expected Behavior

OpenROAD should not crash and the no. of violations should be less.

### Environment

```shell
![image](https://user-images.githubusercontent.com/58599984/223680113-a79eeba6-c069-417c-9b70-6f65d67eb2bc.png)
```

### To Reproduce

Run `make "DESIGN_CONFIG=./designs/asap7/ibex/config.mk"` in the OpenROAD flow scripts.

### Screenshots

[image]

maliberty, 2023-03-08:
The design passes in the nightly CI so it is likely a local problem.  This report is too vague to be actionable.  What commit id?  What platform?  Are you running out of memory?

vvbandeira, 2023-03-08:
@Eyantra698Sumanto 

Please follow the Issue template closely. There are instructions that must be followed to help us understand the problem:

[image]

vijayank88, 2023-03-09:
As per slack channel discussion with user, in VM they had only 4GB RAM. So once it reaches peak it got exited.

Increase the swap memory for your system and try again.

closed 2023-03-09 by vijayank88