The-OpenROAD-Project/OpenROAD #2991: recipe for target 'src/openroad' failed

smosanu, 2023-03-08:
### Describe the bug

With a "build by hand", we ran into an issue at _make_:

```mkdir build
cd build
cmake ..
**make**```

```libpython3.6m.so 
/opt/or-tools/lib/libortools.so.9.5.2237: undefined reference to `std::to_chars(char*, char*, double)@GLIBCXX_3.4.29'
/opt/or-tools/lib/libortools.so.9.5.2237: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()@GLIBCXX_3.4.26'
/opt/or-tools/lib/libortools.so.9.5.2237: undefined reference to `std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::reserve()@GLIBCXX_3.4.29'
/opt/or-tools/lib/libortools.so.9.5.2237: undefined reference to `std::__throw_bad_array_new_length()@GLIBCXX_3.4.29'
collect2: error: ld returned 1 exit status```

This was on Ubuntu 18 with gcc and g++ version 8.4.0.
[build.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/10925442/build.log)

### Expected Behavior

We expected the build to complete without error

### Environment

```shell
Ubuntu 18 LTS with gcc version 8.4
```

### To Reproduce

see above

### Screenshots

see attached log file

vijayank88, 2023-03-09:
@smosanu 
Have you ran after clone `./etc/DependencyInstaller.sh` inside `OpenROAD` directory?

smosanu, 2023-03-10:
Hi @vijayank88,

Yes, I did run that, and just did it again (see log attached) and the issue with `make` still persists. Is it possible that not all dependencies got installed?

One other thing I did is, I upgraded the gcc and g++ version to 8.4.0, though I did this after installing the dependencies. Perhaps I should re-install some of the dependencies?

Thank you!

[ORdep.log](https://github.com/The-OpenROAD-Project/OpenROAD/files/10943526/ORdep.log)

maliberty, 2023-03-10:
libpython3.8 is already the newest version (3.8.16-1+bionic1).
python3-dev is already the newest version (3.6.7-1~18.04).

look to be out of sync

maliberty, 2023-03-10:
Fwiw I've built on Ubuntu 22.04 LTS

smosanu, 2023-03-14:
Thank you @vijayank88 @maliberty for your suggestions. It kept me motivated to keep digging, even replicated the same issues on another machine with Ubuntu 18.04.

The error message suggests that it has something  to do with **or-tools**, and I was able to fix it by bumping the gcc & g++ compiler to version 11 as per [https://groups.google.com/g/or-tools-discuss/c/d8W3urEp0wk](https://groups.google.com/g/or-tools-discuss/c/d8W3urEp0wk)

```
add-apt-repository -y ppa:ubuntu-toolchain-r/test
apt install -yq gcc-11 g++-11
```

Hope this is useful to others stuck with Ubuntu 18.04.

closed 2023-03-14 by smosanu