The-OpenROAD-Project/OpenROAD #2999: Automated build and publish of debian packages to a Personal Package Archive

nekronos, 2023-03-10:
### Description

@vvbandeira these are the tasks I can identify for setting up automated build and publish of debian packages.

- [ ] Make the [build deb package github action](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/.github/workflows/github-actions-build-deb-package.yml#L3-L6) trigger on changes to the [changelog file](https://github.com/The-OpenROAD-Project/OpenROAD/blob/master/debian/changelog). An update to the changelog + a new version number is needed when publishing an updated package.
- [ ] Remove the upload step of `.deb` packages
- [ ] Change the build commands to build a source package. A source package must be built, then launchpad will build deb packages from it. Docs: https://help.launchpad.net/Packaging/PPA/BuildingASourcePackage
- [ ] Create a user on https://help.launchpad.net and ask me to transfer ownership of https://launchpad.net/openroad to you.
- [ ] Upload the source package to the [openroad ppa](https://launchpad.net/openroad). Docs: https://help.launchpad.net/Packaging/PPA/Uploading
- [ ] Add readme entry for adding the ppa:
    ```
    sudo add-apt-repository ppa:user/ppa-name
    sudo apt-get update
    ```

### Additional Context

It has to be investigated/tested how this works when yosys is installed separately, does the openroad yosys fork need its own deb package or be included in this one? Maybe https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts should be packaged as a deb package

oharboe, 2023-03-11:
@vvbandeira Some thoughts, I have not had a chance to test yet:

- does yosys installed by OR work side by side with official Ubuntu yosys package?
- ORFS needs to be installed too
- The PDK, which is part of ORFS, is big, but rarely changes. How can the nightly PPA build size of ORFS be minimized?

maliberty, 2025-08-28:
@vvbandeira still relevant?