The-OpenROAD-Project/OpenROAD #3008: Lef parsing core dump

msingh9, 2023-03-13:
### Describe the bug

I am reading a tech lef file (LEF syntax is version 5.8). It core dumps and stack trace is attached. I also see warnings like 

[WARNING ODB-0279] parse mismatch in layer property LEF58_CUTCLASS for layer V1 : "
    CUTCLASS Vx WIDTH 0.020 LENGTH 0.020 CUTS 1 ;
    CUTCLASS VxPWR0 WIDTH 0.020 LENGTH 0.026 CUTS 1 ORIENT VERTICAL ;
    CUTCLASS VxBAR WIDTH 0.020 LENGTH 0.040 CUTS 2 ;
    CUTCLASS VxBARPWR0 WIDTH 0.020 LENGTH 0.042 CUTS 2 ORIENT HORIZONTAL ;
    CUTCLASS VxLRG WIDTH 0.040 LENGTH 0.040 CUTS 4 ; "

[t1.txt](https://github.com/The-OpenROAD-Project/OpenROAD/files/10954175/t1.txt)

### Expected Behavior

I don't want it to crash, rather but I do expect that it may encounter un-supported constructs.

### Environment

```shell
Linux RH8
```

### To Reproduce

LEF it proprietary so I can't share.

antonblanchard, 2023-03-13:
I created a test case to reproduce what I think is the issue. The parser doesn't understand `ORIENT VERTICAL` and `ORIENT HORIZONTAL`. This isn't a huge problem in itself, because those vias aren't going to be used by detailed routing. Perhaps it is an issue for pdn?

Later on we refer to the unknown `CUTCLASS` in a `LEF58_ENCLOSURE` statement, which SEGVs. We should fix our `LEF58_ENCLOSURE` parsing to not crash on unknown `CUTCLASS`

```
VERSION 5.8 ;
BUSBITCHARS "[]" ;
DIVIDERCHAR "/" ;
USEMINSPACING OBS OFF ;

UNITS
  DATABASE MICRONS 1000 ;
END UNITS

MANUFACTURINGGRID 0.001 ;

PROPERTYDEFINITIONS
  LAYER LEF58_CUTCLASS STRING ;
  LAYER LEF58_ENCLOSURE STRING ;
END PROPERTYDEFINITIONS

LAYER V0
  TYPE CUT ;

  PROPERTY LEF58_CUTCLASS "
    CUTCLASS Vx WIDTH 0.001 LENGTH 0.001 CUTS 1 ORIENT VERTICAL ;
    CUTCLASS VxLRG WIDTH 0.001 LENGTH 0.001 CUTS 1 ORIENT HORIZONTAL ; " ;

  PROPERTY LEF58_ENCLOSURE "
    ENCLOSURE CUTCLASS Vx BELOW 0.001 0.0 ;
    ENCLOSURE CUTCLASS VxLRG BELOW 0.001 0.0 ; " ;

END V0

END LIBRARY
```

maliberty, 2023-03-14:
Thanks @antonblanchard that made it much easier to fix

commit abc42768 (2023-03-14): odb: don't crash on invalid CUTCLASS in lefin

Fixes #3008
closed 2023-03-14 by maliberty