The-OpenROAD-Project/OpenROAD #3020: grt creates guides that go through blockages

antonblanchard, 2023-03-15:
### Describe the bug

The attached test case shows guides that go through a blockage. M7 is completely blocked, but some guides go right through it from M5 through to M8:

[image]

### Expected Behavior

Guides should not cross through obstructions

### Environment

```shell
Git commit: edcd14ffec5d232d98e8b0c34315b1ce233b1af6
os: Fedora Linux 37 (Workstation Edition)
cmake version 3.26.0-rc5
```

### To Reproduce

Test case:  [grt-through-blockage.tar.gz](https://github.com/The-OpenROAD-Project/OpenROAD/files/10978590/grt-through-blockage.tar.gz)

eder-matheus, 2023-03-17:
This is a good case to start considering via resources. The current grt core code doesn't check the resources when adding vias. Also, the 2D resources will consider the M8 tracks, even if they are inaccessible from lower layers.

The first step could be to detect inaccessible layers and "block" them from the actual resources. The via-resource model is a more complex work that can be implemented next.

maliberty, 2023-03-18:
@antonblanchard would you give some more context as to what you are seeking to have happen by blocking m7?

antonblanchard, 2023-03-19:
> @antonblanchard would you give some more context as to what you are seeking to have happen by blocking m7?

Trying to build simple test cases to exercise specific parts of drt. In this particular case I could obstruct all layers other than the two I'm interested in.

maliberty, 2023-03-19:
What does it mean to block all but two non-adjacent layers?

antonblanchard, 2023-04-13:
> What does it mean to block all but two non-adjacent layers?

To force specific vias to be used in test cases. I'm not sure how much of an issue this is outside of test cases, and I can see how grt would struggle since it starts with a 2D view of the capacity.