The-OpenROAD-Project/OpenROAD #3028: drt: pa continually calls FlexGCWorker::Impl::initDesign()

antonblanchard, 2023-03-17:
### Describe the bug

I was investigating why it takes so long to start ispd19_test10. Even after https://github.com/The-OpenROAD-Project/OpenROAD/pull/3027 we still spend a significant amount of time in `initDesign_skipObj`. 

It looks like we call it multiple times for every access point, and multiple times for every inst (both via initPA0)

### Expected Behavior

Reduce the time it takes to start ispd_test10

### Environment

```shell
Git commit: edcd14ffec5d232d98e8b0c34315b1ce233b1af6
os: Fedora Linux 37 (Workstation Edition)
cmake version 3.26.0-rc5
```

### To Reproduce

Run ispd19_test10

maliberty, 2023-03-17:
@osamahammad21 any comments?

osamahammad21, 2023-03-20:
I am not sure exactly if I understand what's the issue here but let me clarify how PA currently works. For each access point in the initial points generation, it tries to access that point from all directions (E,W,N,S,U). In every attempted access, it creates a new FlexGCWorker and initializes it. That gives you 5 calls to initDesign for each access point of each pin of each instance. 
Now I am not saying that it is optimal, but this is how it is now. If you are attempting to speedup this, I would recommend reducing the gcworkers to one per instance. However, it could backfire if you are generating access points to huge macro that would have a huge gcworker box with many objects to initialize.

antonblanchard, 2023-03-26:
A large part of the problem is https://github.com/The-OpenROAD-Project/OpenROAD/pull/3087

antonblanchard, 2023-04-13:
I opened https://github.com/The-OpenROAD-Project/OpenROAD/issues/3171 to track the rest of this.

closed 2023-04-13 by antonblanchard