The-OpenROAD-Project/OpenROAD #3039: Issues with timing driven global routing

antonblanchard, 2023-03-19:
### Describe the bug

Attached is a test case from https://github.com/The-OpenROAD-Project/OpenROAD/pull/2975. Some critical nets are getting badly routed (eg `soc0._331_`).

I had hoped `-critical_nets_percentage` would help, but it doesn't appear to. A few things:

- We don't do anything if the worst slack is positive:

```
  // Ensure the slack threshold is negative
  if (slack_th >= 0) {
    for (float slack : slacks) {
      if (slack >= 0)
        break;
      slack_th = slack;
    }
  }

  if (slack_th >= 0) {
    return;
  }
```

But it might be only slightly positive, and bad global routing it could make it much worse, driving the design into negative slack. I think we still want to prioritise the most critical nets.

- I worked around this (by removing the above code, and also setting a huge value for the default slack in `computeNetSlacks` so non critical nets get sorted last. Even with this, `-critical_nets_percentage` does a bad job of routing `soc0._331_`.

### Expected Behavior

Better routing of critical nets

### Environment

```shell
Git commit: f13f5828b53eb45ea040f46c2b6d9b41f2163f9b
os: Fedora Linux 37 (Workstation Edition)
cmake version 3.26.0-rc5
```

### To Reproduce

Test case is at https://ozlabs.org/~anton/junk/microwatt-grt-issue.tar.gz

antonblanchard, 2023-03-19:
While trying to fix this, I somehow managed to make it even worse. It makes me wonder if we should have a metric or a warning when global routes deviate a lot from their original Steiner tree.

[image]

maliberty, 2023-03-19:
We could have an ST wire length and a grt wirelength.  Some amount of deviation on a net is expected but this look pretty extreme.

antonblanchard, 2023-03-19:
> We could have an ST wire length and a grt wirelength. Some amount of deviation on a net is expected but this look pretty extreme.

Sounds good. `report_wire_length` gives us grt and drt wire length, so we could add an -st option

maliberty, 2023-03-21:
I think we need to update the net slacks during routing and not only take them at the start of routing.  @eder-matheus perhaps you could work with @arthurjolo on this?

closed 2023-07-13 by maliberty