The-OpenROAD-Project/OpenROAD #3050: (propose a new feature) Special Router to connect std cells to power rings

Jerry-lqx, 2023-03-20:
### Description

As described in #2209, in the design that involves multiple voltage domains, there is no routes from standard cell to another voltage domain that is generated by pdngen.

### Suggested Solution

Wentian and I from the openfasoc team created a PR #3043 to address this issue. Currently, it is able to connect the standard cells to power rings. However, it requires additional operations after the final cdl for lvs check, because we created new nets and pins that were not specified in the verilog.

Special Router with 1 PGpin on power ring VIN
[image]

Special Router with 4 PGpins on power ring VIN
[image]

Future Work:
1. Update the sroute so that no additional operations are needed after a complete flow script.
2. set_routing_layers [-power minLayerName maxLayerName]  for power connection.
3. Additional ndr rules [-multicut via]. May need further modification in detail router.

maliberty, 2023-03-20:
I'm interested to understand how you got these results from that PR?  The router doesn't know anything about pgpin objects today.  Is there more associated code you haven't put in the PR?

Jerry-lqx, 2023-03-20:
We'll check our repo and update PR accordingly.

Wen-Tian-Pineapple, 2023-03-22:
> I'm interested to understand how you got these results from that PR? The router doesn't know anything about pgpin objects today. Is there more associated code you haven't put in the PR?

So the router doesn't need to see the pgpin object. The create pgpin command basically create the bterm and bpin on the power ring so that the router can use those bpins to do connection. I uploaded a test with ok file to show the pin location.

maliberty, 2023-03-22:
> So the router doesn't need to see the pgpin object. The create pgpin command basically create the bterm and bpin on the power ring so that the router can use those bpins to do connection. I uploaded a test with ok file to show the pin location.

Why do we need the pgpin in that case?  Why not directly create the bterm/bpin?

Wen-Tian-Pineapple, 2023-03-22:
> Why do we need the pgpin in that case? Why not directly create the bterm/bpin?

We don't actually create pgpin object. The reason I create a class is just to put all the command related to sroute together. Also we need to find the power ring before create the bterm/bpin. I guess it's also feasible to just put the function somewhereelse.

maliberty, 2023-03-23:
> We don't actually create pgpin object. The reason I create a class is just to put all the command related to sroute together. Also we need to find the power ring before create the bterm/bpin. I guess it's also feasible to just put the function somewhereelse.

This is a mistake.  You shouldn't create a database class just for this purpose.  Probably it is better to have this conversation in the PR.

maliberty, 2025-08-26:
sroute code from UMich was accepted a while ago

closed 2025-08-26 by maliberty